LORD-MicroStrain/microstrain_inertial #147: About 3DMGX5-AHRS Euler angle

zyx-99, 2022-04-22:
I want to know whether the Euler angle of this IMU corresponds to the figure below, that is, the X axis corresponds to pitch, the Y axis corresponds to roll, and the Z axis corresponds to yaw. As shown below.
[image]
And I want to know if I can modify the drive so that the axis of IMU can correspond to the axis of lidar.
Thanks!!!!!

zyx-99, 2022-04-22:
> 

thanks for your reply! but the parameter setting still confusing me.
as you say.it would be a yaw of 180, a roll of 180, and a pitch of 0. 
if the parameter should be set as [3.14,0.0,3.14](these are in roll, pitch, yaw order.) 
rather than
[0, 3.14, 3.14] (these are in roll, pitch, yaw order.)

zyx-99, 2022-04-22:
> Hi zyx,
> 
> The rotations about the IMU axis are not as described in the image above. A roll would be around the x axis, a pitch about the y, and a positive rotation on the z axis is in the opposite direction shown in the image.
> 
> To answer your question, we provide a sensor2vehicle transformation in the driver that will allow for the axis to change, but not the + rotation direction. For your purposes, it would be a yaw of 180, a roll of 180, and a pitch of 0. So you could use the Euler angle definition and put in [0, 3.14, 3.14] (these are in roll, pitch, yaw order.)
> 
> filter_sensor2vehicle_frame_selector : 1 filter_sensor2vehicle_frame_transformation_euler : [0.0, 3.14, 3.14]
> 
> Hope that helps!
> 
> Nathan

thanks for your reply! but the parameter setting still confusing me.
as you say.it would be a yaw of 180, a roll of 180, and a pitch of 0.
if the parameter should be set as [3.14,0.0,3.14](these are in roll, pitch, yaw order.)
rather than
[0, 3.14, 3.14] (these are in roll, pitch, yaw order.)

nathanmillerparker, 2022-04-22:
Yep, you're right. [3.14, 0, 3.14]

zyx-99, 2022-04-22:
> Yep, you're right. [3.14, 0, 3.14]
Thanks a lot.

zyx-99, 2022-04-27:
> Hi zyx,
>
> The rotations about the IMU axis are not as described in the image above. A roll would be around the x axis, a pitch about the y, and a positive rotation on the z axis is in the opposite [...]

so sorry to bother you again. but could you tell me if I set "use_enu_frame" to TRUE in ROS driver. what should the coordinate looks like.
[image]
as shown in the image above. the device only describes the NED coordinate system.

nathanmillerparker, 2022-04-27:
Hi zyx,

All of our devices are NED natively... the ENU coordinate system is provided in the ROS driver as a convenience to customers.  If you select ENU, the coordinates on the unit will be as follows:

[image]

closed 2022-06-05 by github-actions[bot]