LORD-MicroStrain/microstrain_inertial #152: Data rate is not as expected

ozangerger, 2022-04-28:
**Describe the bug**
IMU data rate is set to 200hz, 
rosbag recording shows that the datarate is variable but around 50-60Hz.
rosbag hz topic still shows 200hz. node spin rate is 400hz.

**To Reproduce**
Steps to reproduce the behavior:
launch the ros driver of microstrain imu

**Expected behavior**
I would expect to see datarate in rqt_bag around 200hz.

**Environment (please complete the following information):**
 - OS: Ubuntu 16.04
 - Architecture: arm64
 - ROS Version: kinetic
 - Version: [e.g. 2.0.0]
 - Sensor(s): 3DM-GQ7

**Modifications**
Only the launch parameters are modified.

**Launch Parameters**
attached you can find launch params

**Additional context**
sensor config file fetched from sensorconnect, rosout.txt and rosbag recording file is provided as well.
[sensor logs and configs.zip](https://github.com/LORD-MicroStrain/microstrain_inertial/files/8581097/sensor.logs.and.configs.zip)

robbiefish, 2022-05-09:
Hi @ozangerger, apologies for not getting to this earlier.

Thank you for sending the config files and rosbag. I loaded up a GQ7 with that configuration, and see that the data rate is around 200 hz. Also, when playing the bag you provided with `rosbag play`, I see that the data rate for `/imu/data` is also around 200hz.

> rosbag recording shows that the datarate is variable but around 50-60Hz.

What steps do I need to take to reproduce this behavior? You mentioned `rqt_bag` in your comment. Is that how you are recording the bag, and is that where you are seeing this issue? If so, how are you determining the data rate?

ozangerger, 2022-05-12:
Hi Nathan, 

The current setup is as follows:

A serial to usb converter cable is connected to USB port of nvidia PX2 Drive unit, using 460800 baud rate. So the serial interface is converted into usb interface. If you think that's the case, I will try to connect using UART interface directly and check the data again.

nathanmillerparker, 2022-05-12:
I would suggest giving that a go... USB is known to cause issues like this with high rate data, where you get every packet, but they come in bunches.  This is especially true if the host computer is under significant load due to other processing.  Also, I would recommend switching to 921600, if you have that available on this system, just to rule out any bandwidth issues now or in the future (for instance, if you need to enable support channels to send us a raw binary, that requires more bandwidth.)   If you are utilizing the IMU data in a control loop, you really want to use an actual serial connection because the time of arrival for the data is far more deterministic.  

Nathan

ozangerger, 2022-05-27:
Hi Nathan,

First of all, thanks for the answer. I've changed the baudrate to 921600 but it doesn't seem to have any effect on data. I understand what you mean by getting deterministic data and it is a valid point. However, in my case, the actual data rate is far away from the desired rate and that is not related to determinism, that looks like an interface problem which causes a bottleneck. Could you please specify a serial cable or adapter which you might have been using so far and I can test with nvidia xavier jetson or px2 drive? 

Best regards,
Ozan

nathanmillerparker, 2022-05-31:
Hi Ozan,

In a previous message you said you would try using the UART directly and I suggested using 921600 with that interface if it is available on the Jetson... is this the test you ran?  If you ran the USB to serial converter at 921600, I don't believe that will make a difference.  When using a serial connection, running at the highest baudrate minimizes transmission latency and removes most concerns about throughput.   

If you contact me directly (nathan.miller [at] parker.com), I can get you in touch with a sales person to get you a flying lead cable that you can use to connect to the Jetson's pins.

As far as recommending a USB-to-serial converter, that is a bit harder as the USB "bunching issue" could have contributions from: the CPU load, the operating system, and the serial/USB converter hardware/driver.  I don't know what to recommend for the Jetson... most customers who need deterministic IMU data from our devices chose the serial interface.  

Hope that helps,

Nathan

ozangerger, 2022-06-23:
Hi all,

I've fixed my issue with a new serial cable. After trying different serial cables and ftdi chips, I found "StarTech.com 1 Port USB to Serial RS232" one gives the best performance.

Regards,
Ozan

closed 2022-06-23 by ozangerger