LORD-MicroStrain/microstrain_inertial #160: NMEA

binoy-meridian, 2022-05-26:
We are connecting a lidar with the lord, the lidar need PPS input and NMEA(GPGGA, GPZDA) MESSAGE FROM GPS to  do the timestamping in the pointcloud.
So that we can get the accurate time with respect to the location captured..
Usually we give NMEA output from gps to lidar sensor.
As the NMEA is not present here in lord is there any other way to timestamp the lidar data? Though ros of lord or some other ways.?
Thanks
Meridian Geomatics

nathanmillerparker, 2022-05-31:
Hello,

What device are you using?  If it is a GQ7, you can enable RTK dongle handshaking which will output a GGA message on the AUX port. Note: this will be interlaced with a MIP message, so either the lidar will need to be able to reject the MIP message, or you could act as the middle man and strip out the NMEA and send it yourself.  Also note, NMEA output on the main port is on our list for the next GQ7 firmware release, which I don't have a hard date on, but should be around the end of summer.  

If you are using a different device, we don't have a solution currently.

Hope that helps,

Nathan

robbiefish, 2022-05-31:
Hi @binoy-meridian, to add some more information to what @nathanmillerparker just said

If using the GQ7, you can configure the device to output a PPS signal via a GPIO line. Please note that you will need to have a wire with GPIO leads. The following config shows an example of outputting a PPS signal on GPIO line 1 which is on the main port of the GQ7. Change "gpio1" in the following config to a different number between 1-4 to use a different GPIO line.

```yaml
gpio1_feature  : 2
gpio1_behavior : 2
gpio_config : True

filter_pps_source : 1  # Ensure that you have at least GNSS antenna 1 connected to the GQ7. If you have GNSS antenna 2 connected, change this value to 2
```

In terms of the NMEA sentences, the GQ7 outputs GPGGA NMEA sentences on the aux port. If you are able to use only GPGGA sentences in conjunction with the PPS source and can connect the aux port, you can use the following config to publish the NMEA sentences to the ROS network:

```yaml
aux_port : /dev/ttyACM1  # Change this to wherever the aux port is located
rtk_dongle_enable : True
publish_nmea : True
```

However, if you need both GPGGA and GPZDA, you will need to form those messages yourself. If you configure the device to output its device timestamp, you could form NMEA sentences with the timestamp of any of the filter or GNSS messages

```yaml
use_device_timestamp : True
```

binoy-meridian, 2022-07-17:
We have a connection from aux port, we require just NMEA message to be provided from a single output pin to our Lidar, we are not using rtk dongle from lord as we are using our own base station and internet to receive rtk correction..
Can you tell me if we do the following configuration in ros which pin will be outputting the NMEA message?

nathanmillerparker, 2022-07-18:
Hello, 

Please refer to the following documentation for pinouts and electrical specifications:

https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/user_manual_content/specifications/Main_Aux.htm
https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/user_manual_content/specifications/Electrical.htm

On a side note... our next firmware release (approximately 2 months out) will have configurable NMEA on the main port as well.  It will have many of the commonly used NMEA messages including the time/date.

Thanks,

Nathan

meridiangeomatics, 2022-08-10:
> Hello,
>
> Please refer to the following documentation for pinouts and electrical specifications: [...]

Hi
any updates on the new firmware?

nathanmillerparker, 2022-08-10:
Hello,

Yes, we have entered the testing phase for this firmware and will hopefully release it somewhere between the end of the month and mid next month.  

Nathan

meridiangeomatics, 2022-08-28:
Could you provide beta firmware for testing

meridiangeomatics, 2022-08-28:
Could you provide bete firmware for testing?

meridiangeomatics, 2022-09-29:
any updates on newfirmware?

meridiangeomatics, 2022-10-18:
> Hello,
>
> Yes, we are currently testing the firmware and are a couple of weeks
> behind schedule for release. We hope to release it in the next 2 weeks.
>
> Nathan

Hi
Any updates?

nathanmillerparker, 2022-10-20:
Hello,

We are getting closer to release, I apologize for the delay, but the many additions in this version have taken longer to test than initially scoped. 

Thank you for your patience,

Nathan

binoy-meridian, 2022-10-20:
Thank you for your response

robbiefish, 2024-03-25:
For any future users looking to solve this problem, NMEA streaming can be configured in [this section](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/3c7cc7ebf5fdd3db411a439de9c569cae15f214e/config/params.yml#L478) of the params.yml

closed 2024-03-25 by robbiefish