LORD-MicroStrain/microstrain_inertial #162: gnss1_antenna_offset not being set properly on gq7

mdelat, 2022-06-16:
We've been having trouble setting the antenna offset of gnss1. It appears to get set to a default value.

In our yaml file, we set _gnss1_antenna_offset_ to [.17, -.39, .19]. However when we launch it, the driver spits out a message stating that its "Setting GNSS1 antenna offset to [3.0, 0.0, 0.0]." Additionally when we connect to it via SensorConnect afterwards, the value is set to [3.0, 0.0, 0.0]. 

The rosparam itself seems to get set properly. When we do a "rosparam get" on gnss1_antenna_offset, it is set to our value of [.17, -.39, .19]. 

We have no issues with gnss2. This is just happening with gnss1.

We appreciate any help we can receive! Thanks

robbiefish, 2022-06-29:
Hi @mdelat,

Apologies for the delay. A couple questions:
1. What version of the driver are you using?
2. Could you provide your `params.yml` file?
3. What device are you using?
4. Could you share the logs you see from the driver when it fails to configure `gnss2_antenna_offset`?

mdelat, 2022-06-29:
Hello @robbiefish,

I appreciate the response. 

1. We are using the latest version 1.2.4
2. Here's our params file:
[microstrain_gq7.txt](https://github.com/LORD-MicroStrain/microstrain_inertial/files/9014103/microstrain_gq7.txt)
3. We are using 3DMGQ&-GNSS/INS.
4. Here's our logs:
process[gq7/microstrain_inertial_driver-2]: started with pid [11814]
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configure]: Using MSCL Version: 64.2.2
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::connectDevice]: Attempting to open serial port </dev/microstrain_0000_6284.145861> at <115200>
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::connectDevice]: Setting to Idle: Stopping data streams and/or waking from sleep
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::connectDevice]: 
      #######################
      Model Name:    3DM-GQ7
      Serial Number: 6284.145861
      #######################
    
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureIMU]: Setting Declination Source
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureGNSS]: Setting GNSS1 antenna offset to [3.000000, 0.000000, 0.000000]
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureGNSS]: Setting GNSS2 antenna offset to [0.170000, 0.390000, 0.190000]
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureRTK]: Setting RTK dongle enable to 1
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureFilter]: Note: The device does not support the vehicle dynamics mode command.
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureFilter]: Setting PPS source to 0X01
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureFilter]: Note: The device does not support the heading source command.
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureFilter]: Note: The device does not support the filter autoinitialization command.
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureFilter]: Setting autoadaptive options to: level = 0, time_limit = 15000
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureFilterAidingMeasurement]: Filter aiding gnss pos/vel = 1
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureFilterAidingMeasurement]: Filter aiding gnss heading = 1
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureFilterAidingMeasurement]: Filter aiding altimeter = 1
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureFilterAidingMeasurement]: Filter aiding odometer = 0
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureFilterAidingMeasurement]: Filter aiding magnetometer = 1
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureFilterAidingMeasurement]: Filter aiding external heading = 0
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureFilter]: Setting reference position to: [0.000000, 0.000000, 0.010000], ref frame = 1
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureFilter]: Setting wheeled vehicle contraint enable to 1
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureFilter]: Setting vertical gyro contraint enable to 0
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureFilter]: Setting GNSS antenna calibration to: enable = 1, max_offset = 0.100000
[ WARN] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::getSupportedMipChannels]: Attempted to stream MIP field with descriptor 0x8214 at a rate of 100 hz, but the device reported that it does not support it
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::configureSensor2vehicle]: Setting sensor2vehicle frame transformation with euler angles [0.000000, 0.000000, 0.000000]
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::setupDevice]: Saving the launch file configuration settings to the device
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainConfig::setupDevice]: Resetting the filter after the configuration is complete.
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainPublishers::configure]: Publishing raw IMU data.
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainPublishers::configure]: Publishing Magnetometer data.
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainPublishers::configure]: Publishing GNSS1 NavSatFix data.
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainPublishers::configure]: Publishing GNSS1 odom data.
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainPublishers::configure]: Publishing GNSS1 time reference data.
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainPublishers::configure]: Publishing GNSS1 fix info data.
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainPublishers::configure]: Publishing GNSS1 aiding status
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainPublishers::configure]: Publishing GNSS2 NavSatFix data.
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainPublishers::configure]: Publishing GNSS2 odom data.
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainPublishers::configure]: Publishing GNSS2 time reference data.
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainPublishers::configure]: Publishing GNSS2 fix info data.
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainPublishers::configure]: Publishing GNSS2 aiding status
[... cut ...]
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainPublishers::configure]: Publising transform from imu_gq7_filter to imu_gq7_link
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainPublishers::configure]: Publishing Dual Antenna Status message
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainNodeBase::configure]: Setting spin rate to <200.000000> hz
[ INFO] [/gq7/microstrain_inertial_driver] [MicrostrainNodeBase::activate]: Resuming the device data streams
[ INFO] [/gq7/microstrain_inertial_driver] [Microstrain::run]: Starting Data Parsing
^C[gq7/microstrain_inertial_driver-2] killing on exit

robbiefish, 2022-06-30:
This is an interesting case. I tested with your params file and saw the same behavior.

It appears that you need to add the leading `0` to negative subzero floats in our params file. For example, if you change your `gnss1_antenna_offset` param to:
```yml
gnss1_antenna_offset : [.17, -0.39, .19]
```
That should solve the problem

I would suggest adding a leading 0 to all the `gnssX_antenna_offset` values just in case you relocate your antennas and they become negative in the future.

I am going to leave this issue open, please let me know if that resolves your issue

closed 2022-08-07 by github-actions[bot]