LORD-MicroStrain/microstrain_inertial #165: Error:Failed to communicate with the device

HandsomeJunJun, 2022-06-24:
Hello,
I was ready to synchronize the 3DM-GX4-25 and robosens16 based on GPS, but I ran into some problems。
The pin7 of 3DM-GX4-25 is connected with the PPS of GPS。
/external_gps_time topic is published based GPS GPRMC messages。
And my params.yml as belows:
`
use_device_timestamp : true
publish_gps_corr : true
filter_enable_external_gps_time_update : true
filter_external_gps_time_topic         : "/external_gps_time"
gps_leap_seconds                       : 18.0
`
Issues:
[ INFO] [1656036255.567253947]: Setting sensor2vehicle frame rotation with euler angles [-0.000000, -0.000000, -0.000000]
[ INFO] [1656036255.569282816]: Resetting the filter after the configuration is complete.
[ INFO] [1656036255.590767895]: Publishing raw IMU data.
[ INFO] [1656036255.594711453]: Publishing Magnetometer data.
[ INFO] [1656036255.597451834]: Publishing IMU GPS correlation timestamp.
[ INFO] [1656036255.600972209]: Publishing Filter status data
[ INFO] [1656036255.604427654]: Publishing Filter heading message
[ INFO] [1656036255.607756695]: Publishing Filter heading state message
[ INFO] [1656036255.611897955]: Publishing Filter odometry message
[ INFO] [1656036255.615764870]: Publishing Filtered IMU data
[ INFO] [1656036255.618392671]: Publishing Dual Antenna Status message
[ INFO] [1656036255.777160031]: Setting spin rate to <200.000000> hz
[ INFO] [1656036255.792741647]: Subscribed to /external_gps_time for external GPS time
[ INFO] [1656036255.809861746]: Resuming the device data streams
[ INFO] [1656036255.815938240]: Starting Data Parsing
[ERROR] [1656036256.985583318]: Error: Failed to communicate with the device.
[ERROR] [1656036257.987063342]: Error: Failed to communicate with the device.
[ERROR] [1656036258.989130394]: Error: Failed to communicate with the device.
[ERROR] [1656036259.985348960]: Error: Failed to communicate with the device.
[ERROR] [1656036260.992800514]: Error: Failed to communicate with the device.
[ERROR] [1656036261.988745151]: Error: Failed to communicate with the device.
[ERROR] [1656036262.984808063]: Error: Failed to communicate with the device.
[ERROR] [1656036263.986935245]: Error: Failed to communicate with the device.
[ERROR] [1656036264.988696303]: Error: Failed to communicate with the device.
[ERROR] [1656036265.984329345]: Error: Failed to communicate with the device.
[ERROR] [1656036266.986945791]: Error: Failed to communicate with the device.
[ERROR] [1656036267.995006742]: Error: Failed to communicate with the device.
[ERROR] [1656036268.985307444]: Error: Failed to communicate with the device.

topic: /gps_corr

header: 
  seq: 5144
  stamp: 
    secs: 1661042732
    nsecs: 722000000
  frame_id: "sensor"
gps_cor: 
  gps_tow: 2750.722
  gps_week_number: 2224
  timestamp_flags: 7
---
header: 
  seq: 5145
  stamp: 
    secs: 1661042732
    nsecs: 732000000
  frame_id: "sensor"
gps_cor: 
  gps_tow: 2750.732
  gps_week_number: 2224
  timestamp_flags: 5
---
header: 
  seq: 5146
  stamp: 
    secs: 1661042732
    nsecs: 742000000
  frame_id: "sensor"
gps_cor: 
  gps_tow: 2750.742
  gps_week_number: 2224
  timestamp_flags: 5

Thanks.

msclissa, 2022-06-29:
Sorry for the confusion!

I did some testing and found that the GX4-25 just doesn't respond properly to the GPS time update command, so the application doesn't detect the response and thinks that device communication failed.

I confirmed in testing the command does go through and the device attempts to indicate success, just formatted incorrectly.

We no longer release firmware updates for GX4 devices, but we will update the code to catch the errors differently so both weeks and seconds will be updated even if the first command fails. It will still output an error message, as that is informative on devices that don't have this bug, but you can safely ignore it in this case.

We should be able to update the driver in a few days, but if you have the source you can make the following change in [microstrain_inertial_driver_common/src/microstrain_subscribers.cpp](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/b944ea82a166a1f25943400158d52e01a868a9b6/src/microstrain_subscribers.cpp#L176) `MicrostrainSubscribers::externalGpsTimeCallback()`:
```
void MicrostrainSubscribers::externalGpsTimeCallback(const TimeReferenceMsg& time)
{
  if (config_->inertial_device_)
  {
      int64_t utcTime = get_time_ref_sec(time.time_ref) + config_->gps_leap_seconds_ - UTC_GPS_EPOCH_DUR;

      int64_t secs = utcTime % static_cast<int32_t>(SECS_PER_WEEK);

      int weeks = (utcTime - secs) / SECS_PER_WEEK;
      
    try
    {
      config_->inertial_device_->setGPSTimeUpdate(mscl::MipTypes::TimeFrame::TIME_FRAME_SECONDS, secs);
    }
    catch (mscl::Error& e)
    {
      MICROSTRAIN_ERROR(node_, "Error: %s", e.what());
    }
    
    try
    {
      config_->inertial_device_->setGPSTimeUpdate(mscl::MipTypes::TimeFrame::TIME_FRAME_WEEKS, weeks);
    }
    catch (mscl::Error& e)
    {
      MICROSTRAIN_ERROR(node_, "Error: %s", e.what());
    }

    MICROSTRAIN_INFO(node_, "GPS Update: w%i, s%ld", weeks, secs);
  }
}
```

HandsomeJunJun, 2022-07-01:
栓Q！ I will have a try!

HandsomeJunJun, 2022-07-01:
> Sorry for the confusion!
>
> I did some testing and found that the GX4-25 just doesn't respond properly to the GPS time update command, so the application doesn't detect the response and thinks that [...]

A new problem I find.
[image]

closed 2024-05-05 by github-actions[bot]