LORD-MicroStrain/microstrain_inertial #166: Where is the roll, pitch and yaw? How can I get them

HoiCeong, 2022-06-27:
Hi,
I'm developing3DMGX5-AR and want to get the roll, pitch, yaw.
I have ported the program( microstrain_inertial ), but after running it, RPY is not found. So I want to ask where can these be found?

hhc@hhc-UP-APL01:~$ **rostopic list**
/device/status
/imu/data
/moving_ang
/nav/dual_antenna_status
/nav/filtered_imu/data
/nav/heading
/nav/heading_state
/nav/odom
/nav/status
/rosout
/rosout_agg
/tf

robbiefish, 2022-06-29:
Hi @HoiCeong,

The orientation of the device is published in a [`sensor_msgs/Imu`](http://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/Imu.html) message on the topic `/imu/data` for raw values, and on the topic `/nav/filtered_imu/data` for filtered values. The orientation can be found in the `orientation` variable of the `sensor_msgs/Imu` message

Please note that we publish the orientation as a quaternion. To get the Roll, Pitch, and Yaw, you can use the [tf2](http://wiki.ros.org/tf2) library from ROS. An example of converting a quaternion to RPY can be found [here](https://gist.github.com/marcoarruda/f931232fe3490b7fa20dbb38da1195ac)

closed 2022-08-07 by github-actions[bot]

JunHeo45, 2022-08-10:
Hi, I have a follow up question about this. I am using 3dm-gx5-45 GNSS/INS sensor. 

I am curious which frame does /nav/odom odometry message use for both linear and angular velocity? Do they use NED frame or Sensor frame? 
If they use NED frame, is there a topic that contains velocity data in Sensor frame?

nathanmillerparker, 2022-08-10:
The are the sensor frame (the moving frame) with-respect-to the inertial frame (the fixed frame.)  Velocity is currently reported with-respect-to the NED frame (or ENU frame if that option is selected.)

Hope that helps!

JunHeo45, 2022-08-10:
I really appreciate for the quick reply! 
I guess you already explained this but I am confusing about the first sentence. Could you please elaborate on that?
You are saying /nav/odom angular velocity(Roll Pitch Yaw rate) also reported w.r.t NED frame. Is that correct?

nathanmillerparker, 2022-08-10:
It is probably best to just think of the angular rate and acceleration as being in the sensor body frame.  An important note: if you have a sensor2vehicle transformation applied, these will be modified to be reported in the vehicle frame.