LORD-MicroStrain/microstrain_inertial #167: MSCL_LIB_PATH     linked by target "microstrain_inertial_driver" in directory /home/hhc/Desktop/lord/lord_ws_intact/src/microstrain_inertial

HoiCeong, 2022-06-29:
here is the log:

Please set them or make sure they are set and tested correctly in the CMake files:
MSCL_LIB_PATH
    linked by target "microstrain_inertial_driver" in directory /home/hhc/Desktop/lord/lord_ws_intact/src/microstrain_inertial/microstrain_inertial_driver

-- Configuring incomplete, errors occurred!

robbiefish, 2022-06-29:
Hi @HoiCeong,

1. What version of the driver are you using? Are you just cloning directly from github?
2. Are these all of the logs from a fresh build? I would expect to see more logs with some more helpful errors.
3. The driver will attempt to download MSCL at build time. If you have internet access this should work, but if you do not have internet access, or do not want to install the library automatically, you can download MSCL and specify the path manually with the `MSCL_LIB_PATH` variable. More information can be found [here](https://github.com/LORD-MicroStrain/microstrain_inertial#mscl)

HoiCeong, 2022-06-30:
> 
Thanks for your reply.

1. I downloaded the **c++-mscl_64.3.0_amd64.deb**, and excuted : **sudo dpkg -i c++-mscl_64.3.0_amd64.deb, sudo apt install -f** to install the driver. And then I cloned the project and **catkin_make**.
2. here is the LOG:
``` c++
-- +++ processing catkin package: 'microstrain_inertial_driver'
-- ==> add_subdirectory(microstrain_inertial/microstrain_inertial_driver)
-- Looking for MSCL tag v64.2.2 on github
/bin/bash: jq: command not found
curl: (60) SSL certificate problem: unable to get local issuer certificate
More details here: https://curl.haxx.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.
-- Downloading c++-mscl_64.2.2_amd64.deb to /home/xxxx/Desktop/lord/lord_ws_intact/build/microstrain_inertial/microstrain_inertial_driver/c++-mscl_64.2.2_amd64.deb
curl: no URL specified!
curl: try 'curl --help' or 'curl --manual' for more information
dpkg-deb: error: failed to read archive '/home/xxxx/Desktop/lord/lord_ws_intact/build/microstrain_inertial/microstrain_inertial_driver/c++-mscl_64.2.2_amd64.deb': No such file or directory
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
MSCL_LIB_PATH
    linked by target "microstrain_inertial_driver" in directory /home/xxxx/Desktop/lord/lord_ws_intact/src/microstrain_inertial/microstrain_inertial_driver

-- Configuring incomplete, errors occurred!
See also "/home/xxxx/Desktop/lord/lord_ws_intact/build/CMakeFiles/CMakeOutput.log".
See also "/home/xxxx/Desktop/lord/lord_ws_intact/build/CMakeFiles/CMakeError.log".
Makefile:2798: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

3. "you can download MSCL and specify the path manually with the MSCL_LIB_PATH variable"——In which file can I add this variable "MSCL_LIB_PATH" to specify the path manually.

closed 2022-08-07 by github-actions[bot]