LORD-MicroStrain/microstrain_inertial #168: an instance of 'mscl::Error_MipCmdFailed'   what():  The DeclinationSource command has failed.

Vikash-iitd, 2022-07-22:
Hi,
I tried to launch microstrain.launch using `roslaunch microstrain_inertial_driver microstrain.launch` but it throws error. I also tried by setting  `baudrate``:=921600` but still same error was there . In windows software -`SensorConnect`-  it's publishing some Hex Data. but when I tried to launch using `ros package LORD-MicroStrain/microstrain_inertial `it gives errors .

i am attaching screenshot below....

[image]
[image]
[image]

[image]

above you can see detailed output and error.

also `rostopic list` command also `not showing any topic` for lord-imu. Also searched for `mscl::Error_MipCmd` but couldn't find anything.

Thanks

robbiefish, 2022-07-22:
Hi @Vikash-iitd 

This error is due to the fact that the CV7 does not support the default [`filter_declination_source`](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/main/config/params.yml#L224). The comment has been updated to point this out, but you will need to change this to either `1` or `3`. For more information on what the different settings do, please see the documenentation for [Magnetic Field Declination Source Control (0x0D,0x43)](https://s3.amazonaws.com/files.microstrain.com/CV7+Online/external_content/dcp/Commands/filter_command/data/mip_filter_magnetic_declination_source.htm?Highlight=declination%20source)

If you already have a `params.yml` file you are using to configure the driver, you can simply add the following line to your file. If not, you should make a new yml file with the following contents:

```yaml
filter_declination_source: 1
```

Then you can provide the file to the `params_file` option when launching the node. Assuming you made the file at `/home/user/cv7_params.yml` you would run the node like so:

```bash
roslaunch microstrain_inertial_driver microstrain.launch params_file:=/home/user/cv7_params.yml
```

Vikash-iitd, 2022-07-27:
Hey , 
I tried what you suggested and the node moved somewhat further but then it again showed some error . I read param.yaml file comments and and tried to change according to that and it also worked and further moved ahead. now I am stuck in below error, please look into it.
[image]
[image]
[image]
[image]

This error is somewhat related to `Aiding measurement control ` in param.yaml file I guess . please look into it.

robbiefish, 2022-07-27:
It looks like you are using an old version of the driver. The version of the driver you are using does not work with the CV7.

* If you are building from source, either build from the `ros` branch or tag `2.6.0` or above.
* If installing from the ROS package index, please uninstall and reinstall the driver to get the most recent version like so (replace ROS_DISTRO with your ROS version e.g. noetic, melodic, etc.):
    ```bash
    sudo apt-get update
    sudo apt-get purge ros-ROS_DISTRO-microstrain-inertial-driver
    sudo apt-get install ros-ROS_DISTRO-microstrain-inertial-driver
    ```

After the driver has been updated, you should be able to run with the same command

Vikash-iitd, 2022-08-02:
Hey , Thanks for your help. I tried this and finally it worked!!

closed 2022-08-02 by Vikash-iitd