LORD-MicroStrain/microstrain_inertial #171: Set 3DM-GQ7 to idle from ROS1.

tocococa, 2022-07-29:
Hello,

I'm using a 3DM-GQ7 IMU using a serial connection and ROS1 Noetic, and I'd like to be able to set the IMU to idle from ROS. I know that it's possible to set the IMU to idle via serial because SensorConnect allows me to do so, but I haven't been able to find a way to set it to idle from ROS. I'd like to know if it's possible to do so, and if it is, how could I do it?

nathanmillerparker, 2022-08-01:
Hi,

When you refer to "IMU" do you mean all data coming out of the GQ7, or just the IMU data?  Also, could you describe the use case a little so we understand how to best approach the issue.

Thank you,

Nathan

tocococa, 2022-08-01:
Hi,

Yes, with IMU I'm talking about the GQ7 and the data stream coming from it. The problem I'm having is that when I launch a node that subscribes to the GQ7, if it's in streaming mode, the node fails to launch with `Error: Failed to communicate with the device.`. I can set the GQ7 to boot to idle from SensorConnect, such that the node can subscribe without problems on first boot. The issue I'm having is that if the node needs to be relaunched, I'd also need to set the GQ7 to idle again. In issue #55, it's stated that sending a hex code to the GQ7 sets it to idle, and for now we have a short script that does so, but I'd like to know if there's a way to do so from the ROS driver itself.

Thank you.

robbiefish, 2022-08-01:
Hi @tocococa,

The first thing the node attempts to do is to [set the device to idle](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/6484255afd6d426605ca13e155aa45204a405fc9/src/microstrain_config.cpp#L240). Starting in version `2.0.0` we should try to set the node to idle three times to avoid the issue you mentioned.

1. What version of the driver are you using? If it is a version below `2.0.0`, could you please upgrade to at least `2.0.0`, or preferably the latest?
2. Could you please post the logs you see when you launch the driver as well as any parameters you are modifying?

tocococa, 2022-08-08:
Sorry for the delay, we are using version 2.0.0 of your driver, but building it from source because it gave an error when running the version available through apt for ROS Noetic.
The whole log at launch is the following:

```
roslaunch microstrain_gps.launch 
... logging to /home/jetson/.ros/log/9505d916-150d-11ed-87e1-cde69192780f/roslaunch-nano-9035.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nano:33951/

SUMMARY
========

PARAMETERS
 * /microstrain_inertial_driver/aux_port: /dev/ttyACM1
 * /microstrain_inertial_driver/baudrate: 115200
 * /microstrain_inertial_driver/debug: False
 * /microstrain_inertial_driver/device_setup: True
 * /microstrain_inertial_driver/diagnostics: False
 * /microstrain_inertial_driver/enable_hardware_odometer: False
 * /microstrain_inertial_driver/filter_adaptive_level: 2
 * /microstrain_inertial_driver/filter_adaptive_time_limit_ms: 15000
 * /microstrain_inertial_driver/filter_aiding_measurement_summary_data_rate: -1
 * /microstrain_inertial_driver/filter_aiding_status_data_rate: -1
 * /microstrain_inertial_driver/filter_angular_zupt: True
 * /microstrain_inertial_driver/filter_angular_zupt_topic: /moving_ang
 * /microstrain_inertial_driver/filter_auto_heading_alignment_selector: 1
 * /microstrain_inertial_driver/filter_auto_init: True
 * /microstrain_inertial_driver/filter_child_frame_id: sensor
 * /microstrain_inertial_driver/filter_data_rate: 10
 * /microstrain_inertial_driver/filter_declination: 0.23
 * /microstrain_inertial_driver/filter_declination_source: 2
 * /microstrain_inertial_driver/filter_dynamics_mode: 1
 * /microstrain_inertial_driver/filter_enable_altimeter_aiding: False
 * /microstrain_inertial_driver/filter_enable_external_gps_time_update: False
 * /microstrain_inertial_driver/filter_enable_external_heading_aiding: False
 * /microstrain_inertial_driver/filter_enable_gnss_antenna_cal: False
 * /microstrain_inertial_driver/filter_enable_gnss_heading_aiding: True
 * /microstrain_inertial_driver/filter_enable_gnss_pos_vel_aiding: True
 * /microstrain_inertial_driver/filter_enable_magnetometer_aiding: False
 * /microstrain_inertial_driver/filter_enable_odometer_aiding: False
 * /microstrain_inertial_driver/filter_enable_vertical_gyro_constraint: False
 * /microstrain_inertial_driver/filter_enable_wheeled_vehicle_constraint: False
 * /microstrain_inertial_driver/filter_external_gps_time_topic: /external_gps_time
 * /microstrain_inertial_driver/filter_external_speed_topic: /external_speed
 * /microstrain_inertial_driver/filter_frame_id: sensor_wgs84
 * /microstrain_inertial_driver/filter_gnss_antenna_cal_max_offset: 0.1
 * /microstrain_inertial_driver/filter_gnss_dual_antenna_data_rate: -1
 * /microstrain_inertial_driver/filter_heading_data_rate: -1
 * /microstrain_inertial_driver/filter_heading_source: 1
 * /microstrain_inertial_driver/filter_heading_state_data_rate: -1
 * /microstrain_inertial_driver/filter_imu_data_rate: -1
 * /microstrain_inertial_driver/filter_init_attitude: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_init_condition_src: 0
 * /microstrain_inertial_driver/filter_init_position: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_init_reference_frame: 2
 * /microstrain_inertial_driver/filter_init_velocity: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_odom_data_rate: -1
 * /microstrain_inertial_driver/filter_pps_source: 1
 * /microstrain_inertial_driver/filter_relative_odom_data_rate: -1
 * /microstrain_inertial_driver/filter_relative_position_frame: 2
 * /microstrain_inertial_driver/filter_relative_position_ref: [0, 0, 0.01]
 * /microstrain_inertial_driver/filter_reset_after_config: True
 * /microstrain_inertial_driver/filter_sensor2vehicle_frame_selector: 1
 * /microstrain_inertial_driver/filter_sensor2vehicle_frame_transformation_euler: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_sensor2vehicle_frame_transformation_matrix: [1.0, 0.0, 0.0, 0...
 * /microstrain_inertial_driver/filter_sensor2vehicle_frame_transformation_quaternion: [0.0, 0.0, 0.0, 1.0]
 * /microstrain_inertial_driver/filter_speed_lever_arm: [0.0, 0.0, 0.0]
[... 79 lines cut ...]
process[rosout-1]: started with pid [9062]
started core service [/rosout]
process[foo-2]: started with pid [9079]
process[microstrain_inertial_driver-3]: started with pid [9083]
[ INFO] [1659738362.757357822]: Using MSCL Version: 64.2.2
[ INFO] [1659738362.791011780]: Attempting to open serial port </dev/ttyUSB0> at <115200>
[ INFO] [1659738362.795657249]: Setting to Idle: Stopping data streams and/or waking from sleep
[ WARN] [1659738363.046291623]: It looks like the device is streaming. Waiting a second before setting to idle again
[ INFO] [1659738364.046585634]: Setting to Idle: Stopping data streams and/or waking from sleep
[ WARN] [1659738364.296981311]: It looks like the device is streaming. Waiting a second before setting to idle again
[ INFO] [1659738365.297221466]: Setting to Idle: Stopping data streams and/or waking from sleep
[ WARN] [1659738365.547622977]: It looks like the device is streaming. Waiting a second before setting to idle again
[ERROR] [1659738366.798162820]: Error: Failed to communicate with the device.
[ERROR] [1659738366.798267612]: Failed to read configuration for node
[FATAL] [1659738366.798362299]: Unable to configure node base
terminate called after throwing an instance of 'mscl::Error_Communication'
[... cut ...] =/home/jetson/.ros/log/9505d916-150d-11ed-87e1-cde69192780f/microstrain_inertial_driver-3.log].
log file: /home/jetson/.ros/log/9505d916-150d-11ed-87e1-cde69192780f/microstrain_inertial_driver-3*.log
[foo-2] process has finished cleanly
log file: /home/jetson/.ros/log/9505d916-150d-11ed-87e1-cde69192780f/foo-2*.log
```
The only parameter I'm modifying is the USB port to which the 3DM is connected, which is set to `"/dev/ttyUSB0"`, which is the same port where I see the 3DR with grep.

Thank you.

robbiefish, 2022-08-09:
So this error usually occurs when the device is either streaming too much data for the baudrate, or when the serial port is not properly configured.
* Can you confirm that the device is connected via serial?
* If the device is connected via serial, can you check the baudrate that is set via SensorConnect? This should be visible on the `UART BaudRate` panel. This baudrate needs to be set at the same rate in SensorConnect as the rate in the driver, so please ensure that the rate is set to `115200` in SensorConnect.
* Please ensure that the serial port, or adapter you are using supports the baudrate you have set. [This answer](https://superuser.com/a/1482079) on SuperUser has good info on that.
* You can try increasing the baudrate. Remember that you will need to increase the baudrate on both SensorConnect and in the driver, and the baudrate needs to be supported by the port you are using.

tocococa, 2022-08-12:
Hello @robbiefish, I can confirm the device is connected via serial, the baudrate in SensorConnect show as 115200 and that's what I've set in the params file. 
I haven't had access to the Jetson to which we are connecting the 3DM, but from it's documentation, 115200 seems to be the upper limit for what baudrates it can handle reliably, I'll try with a lower baudrate on Monday and check back with you.

tocococa, 2022-08-26:
Hello @robbiefish. I solved the issue, it was a problem with the CH341 driver on the Jetson Nano we had the 3DM connected to. I connected the 3DM directly to the Jetson via UART on the GPIO pins instead of using the DB9 to USB adapter and the problem is gone. Sorry for the troubles.

closed 2022-10-02 by github-actions[bot]