LORD-MicroStrain/microstrain_inertial #176: Continuous drift on the 3DM-GQ7 solution

PaulZabalegui, 2022-09-21:
Hello,

I've performed multiple measurements with the 3DM-GQ7 device. Although some of them have outputted accurate solutions, lots of them have shown a similar performance: a correct initial position and a posterior drift on the Y-axis (being X-axis the direction of travel and Z-axis facing the ground).

Could this be caused by the IMU's bias? It coincides that the IMU's biggest bias occurs in the Y-axis, with a value of 0.016g, being this 10-100x higher than the one in the other axis.

What else could cause this phenomenon?

Thanks in advance

nathanmillerparker, 2022-09-21:
Hello Paul,

This is not expected behavior.  Could you please take the following steps (this assumes you are using USB)?

1. Provide your settings file (params.yml)
2. Do a short collection of data (approximately 10 minutes) with the settings below and provide us the resulting .bin file:
raw_file_enable : True
raw_file_include_support_data : True
raw_file_directory : "/home/your_name" <- need to modify this for your setup.

This will allow us to determine exactly what is happening.  

Thank you,

Nathan

PaulZabalegui, 2022-09-21:
Hello Nathan,

Thank you for your quick response.

We thought about using this device via the MSCL library on an Ubuntu platform but, as we could not make it work, we switched to the SensorConnect software.

I see that recording a .bin file is pretty easy but, will it include the requested fields?
Does the settings file you mentioned have a .json extension when using SensorConnect?

Thank you again for your time.

Regards,
Paul

Fecha: miércoles, 21 de septiembre de 2022, 22:13
Para: LORD-MicroStrain/microstrain_inertial ***@***.***>
CC: Zabalegui Landa, Paul ***@***.***>, Author ***@***.***>

Este mensaje puede contener información confidencial. Si usted no es el destinatario o lo ha recibido por error, por favor, bórrelo de sus sistemas y comuníquelo a la mayor brevedad al remitente. Los datos personales incluidos en los correos electrónicos que intercambie con el personal de Ceit podrán ser almacenados en la libreta de direcciones de su interlocutor y/o en los servidores de Ceit durante el tiempo fijado en su política interna de conservación de información. Ceit gestiona dichos datos con fines meramente operativos, para permitir el contacto por email entre sus trabajadores/colaboradores y terceros. Puede consultar la Política de Privacidad de Ceit en la dirección: https://ceit.es/es/informacion-legal

Mezu honek isilpeko informazioa izan dezake. Mezuaren hartzailea ez bazara edo akats baten ondorioz jaso baduzu, ezaba ezazu zure sistemetatik, mesedez, eta eman horren berri igorleari. Ceit-eko langileekin trukatutako mezu elektronikoetako datu pertsonalak solaskidearen helbide-libretan eta Ceit-en zerbitzarietan gorde daitezke denbora batez; epe hori Ceit-ek informazioa gordetzeari buruz duen barne-politikan zehazten da. Ceit-ek laneko eginkizunetarako soilik kudeatzen ditu datu horiek, bere langileen eta kolaboratzaileen eta hirugarren batzuen artean posta elektroniko bidezko harremana bideratzeko. Helbide honetan duzu Ceit-eko pribatutasun-politika https://ceit.es/es/informacion-legal

This email message may contain confidential information. If you are not the intended recipient of this message or their agent, or if this message has been addressed to you in error, please immediately alert the sender by reply email and then delete this message and any attachments.  The personal information included in email messages exchanged with employees of Ceit may be stored in the database of your interlocutor and/or the servers of Ceit for the time-period stipulated by its internal information storage policy. Ceit stores such data for purely administrative purposes, to facilitate e-mail contact between its employees and third parties. The Ceit Privacy Policy may be accessed at https://ceit.es/es/informacion-legal

Antes de imprimir este mensaje o sus documentos anexos, asegúrese de que es necesario. Proteger el medio ambiente está en nuestras manos.

Mezu hau edo erantsita dituen dokumentuak inprimatu aurretik, ziurtatu beharrezkoa dela. Ingurumena babestea gure esku dago.

Before printing this e-mail or attachments, be sure it is necessary. It is in our hands to protect the environment.

nathanmillerparker, 2022-09-21:
Hi Paul,

Yes, if you are using SensorConnect, please export the json settings file and ensure factory support channels are enabled via the "Sampling" tile (shown below)

[image]

PaulZabalegui, 2022-09-22:
Good afternoon Nathan,

Please find attached the requested files.

I don’t know if you received the second comment I added to my issue’s thread but, just for you to know (I guess it may be important), we got the following error when running the Commanded Built-In Tests:

Write: 75 65 01 02 02 05 E4 CA
Read:  75 65 01 0A 04 F1 05 00 06 83 01 00 00 00 69 81

Which, if I understood correctly, corresponds to “Receiver 1 is inoperative or is not reporting valid data.” 

Thanks for your help.

Regards,
Paul

[Parker3DM-GQ7_IssueAnalysis.zip](https://github.com/LORD-MicroStrain/microstrain_inertial/files/9625895/Parker3DM-GQ7_IssueAnalysis.zip)

nathanmillerparker, 2022-09-22:
Hey Paul,

Thank you for the data, that helped a lot.  Here's what I've been able to determine:

1. The main issue I see is that you have the filter set up to use dual antenna measurements, but the dual antenna subsystem did not compute a solution during the run.  This is consistent with your original data, which showed a deviation in path, likely due to no heading input.  You can see the fix type drop to 0 in the data you provided, implying no dual antenna fix:
[image]

2. The filter enumerated several warnings, pretty much from the start, so if we fix the issue that caused that, it will likely eliminate the issue in #1 above.
[image]

3. Experience suggests that something is likely not quite right in your antenna offsets and/or your sensor-to-vehicle transformation.  I would double check your antenna offsets and try and make sure they are within a couple of centimeters of the true value.  This link has some information on how to do this - https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/user_manual_content/installation/Antenna.htm

4. For the sensor-to-vehicle transformation, your setup has no transformation defined, which implies that the x axis of the GQ7 is pointed in the direction of travel of the vehicle, and the z axis of the GQ7 is pointed directly down.  If this is not the case, you will need to define the transformation correctly.  An important note is that the GNSS antenna offsets are defined in the vehicle frame with respect to the GQ7 origin.  The link above has an example of this.

5. I imported your settings file to my device and noticed a few settings I would change:

   The GNSS antenna lever arm calibration is good to have enabled:
[image]

   Not sure why, but it appears the PPS source has been set to disabled (assuming no import issue on my end.)  Please select 1 of the 2 internal GNSS sources (the default is GNSS1, as some people only use 1 antenna.)
[image]

6. I did read your concern about the BIT test failure, but wanted to review the data before providing a response.  Both GNSS modules appear to be working.... my guess would be that you may be splitting the GNSS signal between two systems?  If so, the GNSS would think there was a fault because the antenna would appear to be open circuit to the detection logic.  Let me know if that is not the case.

7. I can see from the stationary data that you are getting reasonable multipath off of the building.  This, of course, will affect the position solution accuracy.  I know that is not your normal test environment, but it was noticeable.

Please let me know if these suggestions resolve your issues.  I believe nailing down the antenna offsets and sensor-to-vehicle transformation (if applicable), will resolve everything.  On a side note, since you are using the dual antenna for heading, you can monitor the following data quantity in real-time to know if it is working or not: https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/external_content/dcp/Data/filter_data/data/mip_field_filter_gnss_dual_antenna_status.htm

I also recommend monitoring the filter status: https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/external_content/dcp/Data/filter_data/data/mip_field_filter_status.htm

I'm not sure if you are using the ROS driver or not, but these data quantities are available there as well.  

Nathan

PaulZabalegui, 2022-09-22:
Hello Nathan,

Thank you for your reply. I’ll give it a try and check what you said about the antenna offset (#3) but I’ve done nothing different from the measurements where I got great results.

Concerning comment #6, I’m afraid I’m not splitting the signal and that each GNSS is using its own Ublox patch antenna.

About #7, you are right, I just recorded 10min of data with the configuration I usually use, as I thought there might be some problem with the hardware.

I’ll let you know if I solve the problem with the tips you gave me.

Thanks again,

Paul

[image]<https://user-images.githubusercontent.com/69481927/191761112-e1a230d1-546e-44fc-ab72-b6de150d4b54.png>

[image]<https://user-images.githubusercontent.com/69481927/191761653-761710d6-942b-43ee-b3a3-d5362d629a10.png>

[image]<https://user-images.githubusercontent.com/69481927/191763891-bbcc1675-3f36-41dc-8396-bc00b0890326.png>

[image]<https://user-images.githubusercontent.com/69481927/191764190-c4fe3dcd-a502-47f5-8169-fe6482ad9b58.png>

nathanmillerparker, 2022-09-22:
Hi Paul, 

If you run the continuous built-in test command, it provides quite a bit more information... could you provide me the output of that after both GNSS modules have had time to get a lock (outside ~ 1 minute)?

https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/external_content/dcp/Commands/base_command/data/mip_cmd_base_continuous_bit.htm

The interpretation is at the bottom of this page: https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/user_manual_content/additional_features/Built-in%20Test.htm

Thanks,

Nathan

PaulZabalegui, 2022-09-22:
Hi Nathan,

I’m afraid we are not guessing how to run this test. We ran the Commanded BIT via SensorConnect, but we are not able to run the continuous one as we are not guessing how to use the doCommand function from the MSCL library, and couldn’t find any example about it.

We see that there are some implemented functions that correspond to the Base Commands (0x01) such as the Ping one, but we don’t know how to call the doCommand function as we are constantly getting an error related to the type of arguments and overloading the MipNode_doCommand function.

Could you please tell us how to call the continuous test from Python using MSCl?

Thank you again

Paul

https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/external_content/dcp/Commands/base_command/data/mip_cmd_base_continuous_bit.htm

nathanmillerparker, 2022-09-22:
Hi Paul,

Sorry about that... you can use the packet builder in SensorConnect to send this.
[image]

You will need to type in the field descriptor and response descriptor directly.  Then hit send and you should see an ack response with 16 bytes of data as shown here:
[image]

Hope that helps,

Nathan

PaulZabalegui, 2022-09-23:
Good morning Nathan,

I’m afraid I can’t send that message. That is the main reason I couldn’t run this test. As you can see in the image below, Field Descriptor 0x08 cannot be chosen. That is why I tried to run the Python MSCL doCommand, without success.

***@***.***

Regards,
Paul

[image]<https://user-images.githubusercontent.com/69481927/191785143-3453fc73-1a2a-4add-8ace-cc0a4c43c40c.png>

[image]<https://user-images.githubusercontent.com/69481927/191785038-4c3ea159-b107-4c28-a286-781196d44ccf.png>

PaulZabalegui, 2022-09-23:
> Hi Paul,
>
> For some reason your image did not come through, but you should be able to type in the descriptor instead of selecting it from the drop-down. If you are still experiencing trouble with that, please ensure to update to the latest SensorConnect. Please let me know if that did not fix your issue.

Hi Nathan,

I got pretty good measurements today, it seems that enabling the antenna offsets checkbox solved it. I didn’t expect it to be such a decisive parameter though, it never happened before.

Nevertheless, I still would like to be able to run the continuous BIT, just in case. So I’ll give it a try and let you know. I may try updating my SensorConnect version.

Thanks for your time.
Regards

Paul

nathanmillerparker, 2022-09-23:
Hi Paul,

I'm glad you have gotten better measurements.  The antenna offset error estimation is pretty important for when you can't take accurate measurements of the offsets (typical in a vehicle when you just want to put everything in and go... which we do all the time.)  The sensitivity for the overall system is that you lose heading measurements if the dual-antenna cannot calculate a solution, which on on relatively long straight paths means the heading will drift, which you saw.  

In addition to that change, the PPS source does not only affect the outgoing PPS signal, but also time alignment internally as well (since we allow for external PPS syncs.)  I believe the combination of not having the PPS source selected and having no dual antenna solution is what caused the large position drifts that you saw.

If you need help in the future, just reach out... although I hope you won't need to now :-)

Nathan

closed 2022-11-27 by github-actions[bot]