LORD-MicroStrain/microstrain_inertial #181: Configuration fails with 3DM-GX5-35

MarkMckessock, 2022-10-18:
**Describe the bug**
When attempting to launch the ROS2 inertial driver with the 3DM-GX5-35, I get the following error:
```
[INFO] [launch]: All log files can be found below /home/microstrain/.ros/log/2022-10-18-17-29-53-074139-e955a648ab3b-1
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [microstrain_inertial_driver_node-1]: process started with pid [42]
[microstrain_inertial_driver_node-1] [INFO] [1666114193.521127325] [microstrain_inertial_driver]: Using MSCL Version: 62.1.2
[microstrain_inertial_driver_node-1] [INFO] [1666114193.521638833] [microstrain_inertial_driver]: Attempting to open serial port </dev/ttyACM0> at <115200>
[microstrain_inertial_driver_node-1] [INFO] [1666114193.524674213] [microstrain_inertial_driver]: Setting to Idle: Stopping data streams and/or waking from sleep
[microstrain_inertial_driver_node-1] [INFO] [1666114193.542403090] [microstrain_inertial_driver]: 
[microstrain_inertial_driver_node-1]       #######################
[microstrain_inertial_driver_node-1]       Model Name:    3DM-GX5-35
[microstrain_inertial_driver_node-1]       Serial Number: 6252.81981
[microstrain_inertial_driver_node-1]       #######################
[microstrain_inertial_driver_node-1]     
[microstrain_inertial_driver_node-1] [INFO] [1666114193.554559414] [microstrain_inertial_driver]: Setting IMU data to stream at 100 hz
[microstrain_inertial_driver_node-1] [INFO] [1666114193.566150935] [microstrain_inertial_driver]: Note: Device does not support the declination source command.
[microstrain_inertial_driver_node-1] [INFO] [1666114193.572302982] [microstrain_inertial_driver]: Setting GNSS0 data to stream at 2 hz
[microstrain_inertial_driver_node-1] [ERROR] [1666114193.584426150] [microstrain_inertial_driver]: Could not set GNSS1 antenna offset!
[ERROR] [launch_ros.actions.lifecycle_node]: Failed to make transition 'TRANSITION_CONFIGURE' for LifecycleNode '/microstrain_inertial_driver'
[microstrain_inertial_driver_node-1] [ERROR] [1666114193.584562798] [microstrain_inertial_driver]: Failed to read configuration for node
[microstrain_inertial_driver_node-1] [ERROR] [1666114193.584607131] [microstrain_inertial_driver]: Failed to configure node base
```
From looking at the datasheet, it appears that the 35 doesn't support antenna offset configuration like the 45 does and so the node fails when it attempts to configure that property. This configuration step should be skipped for the 35 device.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch Node with provided params and 'configure:=true'
2. Sensor is detected but error occurs

**Expected behavior**
Driver node should launch and initialize properly

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Architecture: amd64
 - ROS Version: humble
 - Version: 2.7.0
 - Sensor(s): 3DM-GX5-35

**Launch Parameters**
```
microstrain_inertial_driver:
  ros__parameters:
    port  : "/dev/ttyACM0"
    gnss1_antenna_offset : [1.0, 1.0, 1.0]
```

robbiefish, 2022-10-18:
@MarkMckessock Thanks for bringing this to our attention. The linked PRs should fix the issue.

MarkMckessock, 2022-10-18:
@robbiefish Thanks so much. Tested your commit and everything appears to be working.

closed 2022-10-24 by robbiefish