LORD-MicroStrain/microstrain_inertial #186: The PpsSource command has failed

bbokser, 2022-11-01:
I am using a 3DMCV7-AR and am attempting to run the Docker image found [here](https://hub.docker.com/r/microstrain/ros-microstrain_inertial_driver) to create a ROS 1 node to subscribe to. I am encountering a

A summary of what I am doing:
1. The 3DMCV7-AR is connected to the C-Series Connectivity Board, which is connected by USB to my laptop.
    - OS: Ubuntu
    - Architecture: x86_64
    - ROS Version: noetic
2. I run `docker run -it --rm --net=host --device=/dev/ttyACM0 microstrain/ros-microstrain_inertial_driver:ros`
3. The following output is produced:

```
... logging to /home/microstrain/.ros/log/6d28530c-5a26-11ed-9f97-84fdd14bfcc6/roslaunch-bbokser-Inspiron-7580-1.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://bbokser-Inspiron-7580:42439/

SUMMARY
========

PARAMETERS
 * /microstrain_inertial_driver/aux_port: /dev/ttyACM1
 * /microstrain_inertial_driver/baudrate: 115200
 * /microstrain_inertial_driver/debug: False
 * /microstrain_inertial_driver/device_setup: True
 * /microstrain_inertial_driver/diagnostics: False
 * /microstrain_inertial_driver/enable_hardware_odometer: False
 * /microstrain_inertial_driver/filter_adaptive_level: 2
 * /microstrain_inertial_driver/filter_adaptive_time_limit_ms: 15000
 * /microstrain_inertial_driver/filter_angular_zupt: True
 * /microstrain_inertial_driver/filter_angular_zupt_topic: /moving_ang
 * /microstrain_inertial_driver/filter_auto_heading_alignment_selector: 1
 * /microstrain_inertial_driver/filter_auto_init: True
 * /microstrain_inertial_driver/filter_child_frame_id: sensor
 * /microstrain_inertial_driver/filter_data_rate: 10
 * /microstrain_inertial_driver/filter_declination: 0.23
 * /microstrain_inertial_driver/filter_declination_source: 2
 * /microstrain_inertial_driver/filter_dynamics_mode: 1
 * /microstrain_inertial_driver/filter_enable_altimeter_aiding: False
 * /microstrain_inertial_driver/filter_enable_external_gps_time_update: False
 * /microstrain_inertial_driver/filter_enable_external_heading_aiding: False
 * /microstrain_inertial_driver/filter_enable_gnss_antenna_cal: False
 * /microstrain_inertial_driver/filter_enable_gnss_heading_aiding: True
 * /microstrain_inertial_driver/filter_enable_gnss_pos_vel_aiding: True
 * /microstrain_inertial_driver/filter_enable_magnetometer_aiding: False
 * /microstrain_inertial_driver/filter_enable_odometer_aiding: False
 * /microstrain_inertial_driver/filter_enable_vertical_gyro_constraint: False
 * /microstrain_inertial_driver/filter_enable_wheeled_vehicle_constraint: False
 * /microstrain_inertial_driver/filter_external_gps_time_topic: /external_gps_time
 * /microstrain_inertial_driver/filter_external_speed_topic: /external_speed
 * /microstrain_inertial_driver/filter_frame_id: sensor_wgs84
 * /microstrain_inertial_driver/filter_gnss_antenna_cal_max_offset: 0.1
 * /microstrain_inertial_driver/filter_heading_source: 1
 * /microstrain_inertial_driver/filter_init_attitude: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_init_condition_src: 0
 * /microstrain_inertial_driver/filter_init_position: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_init_reference_frame: 2
 * /microstrain_inertial_driver/filter_init_velocity: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_pps_source: 1
 * /microstrain_inertial_driver/filter_relative_position_frame: 2
 * /microstrain_inertial_driver/filter_relative_position_ref: [0, 0, 0.01]
 * /microstrain_inertial_driver/filter_reset_after_config: True
 * /microstrain_inertial_driver/filter_sensor2vehicle_frame_selector: 1
 * /microstrain_inertial_driver/filter_sensor2vehicle_frame_transformation_euler: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_sensor2vehicle_frame_transformation_matrix: [1.0, 0.0, 0.0, 0...
 * /microstrain_inertial_driver/filter_sensor2vehicle_frame_transformation_quaternion: [0.0, 0.0, 0.0, 1.0]
 * /microstrain_inertial_driver/filter_speed_lever_arm: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_velocity_zupt: True
 * /microstrain_inertial_driver/filter_velocity_zupt_topic: /moving_vel
 * /microstrain_inertial_driver/gnss1_antenna_offset: [0.0, -0.7, -1.0]
 * /microstrain_inertial_driver/gnss1_data_rate: 2
 * /microstrain_inertial_driver/gnss1_frame_id: gnss1_antenna_wgs84
 * /microstrain_inertial_driver/gnss2_antenna_offset: [0.0, 0.7, -1.0]
 * /microstrain_inertial_driver/gnss2_data_rate: 2
 * /microstrain_inertial_driver/gnss2_frame_id: gnss2_antenna_wgs84
 * /microstrain_inertial_driver/gpio1_behavior: 0
 * /microstrain_inertial_driver/gpio1_feature: 0
 * /microstrain_inertial_driver/gpio1_pin_mode: 0
 * /microstrain_inertial_driver/gpio2_behavior: 0
 * /microstrain_inertial_driver/gpio2_feature: 0
 * /microstrain_inertial_driver/gpio2_pin_mode: 0
 * /microstrain_inertial_driver/gpio3_behavior: 0
 * /microstrain_inertial_driver/gpio3_feature: 0
 * /microstrain_inertial_driver/gpio3_pin_mode: 0
 * /microstrain_inertial_driver/gpio4_behavior: 0
 * /microstrain_inertial_driver/gpio4_feature: 0
 * /microstrain_inertial_driver/gpio4_pin_mode: 0
 * /microstrain_inertial_driver/gpio_config: False
 * /microstrain_inertial_driver/gps_leap_seconds: 18.0
 * /microstrain_inertial_driver/imu_angular_cov: [0.01, 0, 0, 0, 0...
 * /microstrain_inertial_driver/imu_data_rate: 100
 * /microstrain_inertial_driver/imu_frame_id: sensor
 * /microstrain_inertial_driver/imu_linear_cov: [0.01, 0, 0, 0, 0...
 * /microstrain_inertial_driver/imu_orientation_cov: [0.01, 0, 0, 0, 0...
 * /microstrain_inertial_driver/nmea_frame_id: nmea
 * /microstrain_inertial_driver/odometer_scaling: 0.0
 * /microstrain_inertial_driver/odometer_uncertainty: 0.0
 * /microstrain_inertial_driver/poll_max_tries: 60
 * /microstrain_inertial_driver/poll_port: False
 * /microstrain_inertial_driver/poll_rate_hz: 1.0
 * /microstrain_inertial_driver/port: /dev/ttyACM0
 * /microstrain_inertial_driver/publish_filter: True
 * /microstrain_inertial_driver/publish_gnss1: True
 * /microstrain_inertial_driver/publish_gnss2: True
 * /microstrain_inertial_driver/publish_gps_corr: False
 * /microstrain_inertial_driver/publish_imu: True
 * /microstrain_inertial_driver/publish_nmea: False
 * /microstrain_inertial_driver/publish_relative_position: False
 * /microstrain_inertial_driver/raw_file_directory: /home/your_name
 * /microstrain_inertial_driver/raw_file_enable: False
 * /microstrain_inertial_driver/raw_file_include_support_data: False
 * /microstrain_inertial_driver/rtcm_topic: /rtcm
 * /microstrain_inertial_driver/rtk_dongle_enable: False
 * /microstrain_inertial_driver/save_settings: False
 * /microstrain_inertial_driver/subscribe_rtcm: False
 * /microstrain_inertial_driver/use_device_timestamp: False
 * /microstrain_inertial_driver/use_enu_frame: False
 * /microstrain_inertial_driver/use_ros_time: False
 * /rosdistro: noetic
 * /rosversion: 1.15.13

NODES
  /
    microstrain_inertial_driver (microstrain_inertial_driver/microstrain_inertial_driver_node)

auto-starting new master
process[master]: started with pid [33]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6d28530c-5a26-11ed-9f97-84fdd14bfcc6
process[rosout-1]: started with pid [43]
started core service [/rosout]
process[microstrain_inertial_driver-2]: started with pid [50]
[ INFO] [1667335661.256415020]: Using MSCL Version: 63.1.0
[ INFO] [1667335661.261174967]: Attempting to open serial port </dev/ttyACM0> at <115200>
[ INFO] [1667335661.261955927]: Setting to Idle: Stopping data streams and/or waking from sleep
[ INFO] [1667335661.263576781]: 
      #######################
      Model Name:    3DMCV7-AR
      Serial Number: 6287.148232
      #######################
    
[ INFO] [1667335661.268166266]: Setting IMU data to stream at 100 hz
[ INFO] [1667335661.269237166]: Note: Device does not support the declination source command.
[ INFO] [1667335661.279056373]: Setting Filter data to stream at 10 hz
[ INFO] [1667335661.279615353]: Note: The device does not support the vehicle dynamics mode command.
[ INFO] [1667335661.279647125]: Setting PPS source to 0X01
terminate called after throwing an instance of 'mscl::Error_MipCmdFailed'
  what():  The PpsSource command has failed. (Error Code: 3)
[microstrain_inertial_driver-2] process has died [pid 50, exit code -6, cmd /opt/ros/noetic/lib/microstrain_inertial_driver/microstrain_inertial_driver_node __name:=microstrain_inertial_driver __log:=/home/microstrain/.ros/log/6d28530c-5a26-11ed-9f97-84fdd14bfcc6/microstrain_inertial_driver-2.log].
log file: /home/microstrain/.ros/log/6d28530c-5a26-11ed-9f97-84fdd14bfcc6/microstrain_inertial_driver-2*.log
```

**Modifications**
_Did you modify the source code in any way while testing? If so, please note those modifications here._
- No.

**Launch Parameters**
_If you have an updated `params.yml` file, please attach it to this issue._
- N/A

**Additional context**
I have checked SensorConnect: baud rate is 115200.

robbiefish, 2022-11-03:
Hi @bbokser,

Sorry about the delay. The docker images got out of date, so it is taking some time to update them.

The image you used (`microstrain/ros-microstrain_inertial_driver:ros`) has been updated last night to support the CV7, so you should be able to `docker pull` the image and run without problems.

Please note that when using a CV7 you will also need to set `filter_declination_source` to either 1 or 3. The dockerhub page has documentation on how to provide a custom `params.yml`, but here is what I would do for this case:

Make a `params.yml` file with the following contents:
```yaml
filter_declination_source: 1
```

Modify your command to:
```bash
docker run \
  -it \
  --rm \
  --net=host \
  --device=/dev/ttyACM0 \
  -v "$(pwd)/params.yml:/tmp/params.yml" \
  microstrain/ros-microstrain_inertial_driver:ros params_file:=/tmp/params.yml
```

Please let me know if you have any other issues

bbokser, 2022-11-03:
Hi Robbie,

Thanks for your response.

I'm not sure what I'm doing wrong, but this doesn't seem to be working for me.

Here are my steps:
1. `docker pull microstrain/ros-microstrain_inertial_driver`
2. `docker pull microstrain/ros-microstrain_inertial_driver:ros`
    These ran without issue.
3. Placed the params.yml file (empty except for `filter_declination_source: 1`) in my home directory. 
4. Ran the command you provided inside the home directory.

The output appears to be exactly the same:
```
... logging to /home/microstrain/.ros/log/5491c72a-5b92-11ed-8bfe-84fdd14bfcc6/roslaunch-bbokser-Inspiron-7580-1.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://bbokser-Inspiron-7580:39989/

SUMMARY
========

PARAMETERS
 * /microstrain_inertial_driver/aux_port: /dev/ttyACM1
 * /microstrain_inertial_driver/baudrate: 115200
 * /microstrain_inertial_driver/debug: False
 * /microstrain_inertial_driver/device_setup: True
 * /microstrain_inertial_driver/diagnostics: False
 * /microstrain_inertial_driver/enable_hardware_odometer: False
 * /microstrain_inertial_driver/filter_adaptive_level: 2
 * /microstrain_inertial_driver/filter_adaptive_time_limit_ms: 15000
 * /microstrain_inertial_driver/filter_angular_zupt: True
 * /microstrain_inertial_driver/filter_angular_zupt_topic: /moving_ang
 * /microstrain_inertial_driver/filter_auto_heading_alignment_selector: 1
 * /microstrain_inertial_driver/filter_auto_init: True
 * /microstrain_inertial_driver/filter_child_frame_id: sensor
 * /microstrain_inertial_driver/filter_data_rate: 10
 * /microstrain_inertial_driver/filter_declination: 0.23
 * /microstrain_inertial_driver/filter_declination_source: 2
 * /microstrain_inertial_driver/filter_dynamics_mode: 1
 * /microstrain_inertial_driver/filter_enable_altimeter_aiding: False
 * /microstrain_inertial_driver/filter_enable_external_gps_time_update: False
 * /microstrain_inertial_driver/filter_enable_external_heading_aiding: False
 * /microstrain_inertial_driver/filter_enable_gnss_antenna_cal: False
 * /microstrain_inertial_driver/filter_enable_gnss_heading_aiding: True
 * /microstrain_inertial_driver/filter_enable_gnss_pos_vel_aiding: True
 * /microstrain_inertial_driver/filter_enable_magnetometer_aiding: False
 * /microstrain_inertial_driver/filter_enable_odometer_aiding: False
 * /microstrain_inertial_driver/filter_enable_vertical_gyro_constraint: False
 * /microstrain_inertial_driver/filter_enable_wheeled_vehicle_constraint: False
 * /microstrain_inertial_driver/filter_external_gps_time_topic: /external_gps_time
 * /microstrain_inertial_driver/filter_external_speed_topic: /external_speed
 * /microstrain_inertial_driver/filter_frame_id: sensor_wgs84
 * /microstrain_inertial_driver/filter_gnss_antenna_cal_max_offset: 0.1
 * /microstrain_inertial_driver/filter_heading_source: 1
 * /microstrain_inertial_driver/filter_init_attitude: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_init_condition_src: 0
 * /microstrain_inertial_driver/filter_init_position: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_init_reference_frame: 2
 * /microstrain_inertial_driver/filter_init_velocity: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_pps_source: 1
 * /microstrain_inertial_driver/filter_relative_position_frame: 2
 * /microstrain_inertial_driver/filter_relative_position_ref: [0, 0, 0.01]
 * /microstrain_inertial_driver/filter_reset_after_config: True
 * /microstrain_inertial_driver/filter_sensor2vehicle_frame_selector: 1
 * /microstrain_inertial_driver/filter_sensor2vehicle_frame_transformation_euler: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_sensor2vehicle_frame_transformation_matrix: [1.0, 0.0, 0.0, 0...
[... 62 lines cut ...]

setting /run_id to 5491c72a-5b92-11ed-8bfe-84fdd14bfcc6
process[rosout-1]: started with pid [42]
started core service [/rosout]
process[microstrain_inertial_driver-2]: started with pid [49]
[ INFO] [1667491956.827517122]: Using MSCL Version: 62.1.2
[ INFO] [1667491956.831221844]: Attempting to open serial port </dev/ttyACM0> at <115200>
[ INFO] [1667491956.832070423]: Setting to Idle: Stopping data streams and/or waking from sleep
[ INFO] [1667491956.835245502]: 
      #######################
      Model Name:    3DMCV7-AR
      Serial Number: 6287.148232
      #######################
    
[ INFO] [1667491956.840054898]: Setting IMU data to stream at 100 hz
[ INFO] [1667491956.841203849]: Note: Device does not support the declination source command.
[ INFO] [1667491956.850873942]: Setting Filter data to stream at 10 hz
[ INFO] [1667491956.852041318]: Note: The device does not support the vehicle dynamics mode command.
[ INFO] [1667491956.852071837]: Setting PPS source to 0X01
terminate called after throwing an instance of 'mscl::Error_MipCmdFailed'
  what():  The PpsSource command has failed. (Error Code: 3)
[microstrain_inertial_driver-2] process has died [pid 49, exit code -6, cmd /opt/ros/noetic/lib/microstrain_inertial_driver/microstrain_inertial_driver_node __name:=microstrain_inertial_driver __log:=/home/microstrain/.ros/log/5491c72a-5b92-11ed-8bfe-84fdd14bfcc6/microstrain_inertial_driver-2.log].
log file: /home/microstrain/.ros/log/5491c72a-5b92-11ed-8bfe-84fdd14bfcc6/microstrain_inertial_driver-2*.log
```
Interestingly, I also tried changing using a bogus name for a nonexistent params.yml (e.g. `parmesan.yummy`) and got the same exact result. So I have no reason to believe the params.yml is being processed. Shouldn't there be some error message if the params.yml is not found?

robbiefish, 2022-11-03:
Sorry about that, there was an issue with deploying the images. This should be fixed now. Try repulling and rerunning

bbokser, 2022-11-03:
Hi Robbie,

I'm now getting a new error:

```
setting /run_id to 102a58e2-5bcc-11ed-b0e9-84fdd14bfcc6
process[rosout-1]: started with pid [42]
started core service [/rosout]
process[microstrain_inertial_driver-2]: started with pid [49]
[ INFO] [1667516752.774146639]: Using MSCL Version: 64.2.2
[ INFO] [1667516752.777876001]: Attempting to open serial port </dev/ttyACM0> at <115200>
[ INFO] [1667516752.778809236]: Setting to Idle: Stopping data streams and/or waking from sleep
[ INFO] [1667516752.857124918]: 
      #######################
      Model Name:    3DMCV7-AR
      Serial Number: 6287.148232
      #######################
    
[ INFO] [1667516752.864706819]: Note: Device does not support the declination source command.
[ WARN] [1667516752.864764707]: Attempted to stream MIP field with descriptor 0x8006 at a rate of 100 hz, but the device reported that it does not support it
[ INFO] [1667516752.878368842]: Note: The device does not support the vehicle dynamics mode command.
[ INFO] [1667516752.878413012]: Note: The device does not support the heading source command.
[ INFO] [1667516752.878441857]: Note: The device does not support the filter autoinitialization command.
[ INFO] [1667516752.878471904]: Setting autoadaptive options to: level = 2, time_limit = 15000
[ WARN] [1667516752.883102966]: Note: Filter aiding gnss pos/vel not supported, but it was requested. Disable in params file to remove this warning
[ WARN] [1667516752.883126232]: Note: Filter aiding gnss heading not supported, but it was requested. Disable in params file to remove this warning
[ INFO] [1667516752.883138997]: Note: Filter aiding altimeter not supported
[ INFO] [1667516752.883148855]: Note: Filter aiding odometer not supported
[ INFO] [1667516752.883158689]: Filter aiding magnetometer = 0
terminate called after throwing an instance of 'mscl::Error_MipCmdFailed'
  what():  The AidingMeasurementEnable command has failed. (Error Code: 3)
[microstrain_inertial_driver-2] process has died [pid 49, exit code -6, cmd /opt/ros/noetic/lib/microstrain_inertial_driver/microstrain_inertial_driver_node __name:=microstrain_inertial_driver __log:=/home/microstrain/.ros/log/102a58e2-5bcc-11ed-b0e9-84fdd14bfcc6/microstrain_inertial_driver-2.log].
log file: /home/microstrain/.ros/log/102a58e2-5bcc-11ed-b0e9-84fdd14bfcc6/microstrain_inertial_driver-2*.log
```
I tried changing `filter_enable_magnetometer_aiding` to `true`--didn't help. Please advise.

robbiefish, 2022-11-04:
I apologize for the continuous problems. The problem you encountered was an unknown problem in the driver with the version of MSCL we are using and the CV7-AR. A hotfix has been pushed in the following PRs:

ROS: #188
ROS2: #189
Submodule: LORD-MicroStrain/microstrain_inertial_driver_common#30

Please pull the image and try again.

closed 2022-11-04 by bbokser