LORD-MicroStrain/microstrain_inertial #194: Noisy data output from 3DMCV7-AHRS

Alex-Beh, 2022-11-14:
**Describe the bug**
The data from the IMU is very very noisy. As shown in the video below that is recorded with the IMU remain stationary, the linear acceleration of x & y are non-zero and the yaw oritentation keep increasing without any sense. I have performed the magnetic calibration but it seems don't solve the issue.

I have uploaded a bagfile to [google drive](https://drive.google.com/drive/folders/16mG9OQisqHUx9l2yvYh3Izk84hs3YCuZ?usp=share_link).

https://user-images.githubusercontent.com/66585558/201614212-801070ac-01ba-4737-87d8-1d55d80ccc5f.mp4

**To Reproduce**

Steps to reproduce the behavior:
1. ros2 launch microstrain_inertial_driver microstrain_launch.py
2. visualize the data in plotjuggler

**Expected behavior**
Expected to have stable and accurate data output.

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Architecture:  x86_64
 - ROS Version: foxy
 - Version: [e.g. 2.0.0]
 - Sensor(s): 3DM-CV7

**Modifications**
Did you modify the source code in any way while testing? If so, please note those modifications here.

**Launch Parameters**
I have updated the params.yaml. The main changes have been listed as below. I cannot uploaded the yaml file here so I change the extension to txt.
[params.txt](https://github.com/LORD-MicroStrain/microstrain_inertial/files/10001253/params.txt)

```
use_enu_frame : True
use_ros_time : True
filter_declination_source: 1
```

**Additional context**
Add any other context about the problem here.

nathanmillerparker, 2022-11-14:
Hi Alex,

Is the device sitting on or near something that could be vibrating (even a computer fan or something similar could be causing this as the accel is quite sensitive.)  It could even be a table if it is mechanically coupled to, say, the HVAC of the building.  The baseline would be noise 10x smaller than this in a quiescent environment.

As far as the heading drifting, you didn't indicate if you had a CV7-AR or -AHRS.  The AR model will have unbounded drift for yaw because it doesn't have a magnetometer.  If you do have the AHRS model, it is likely the magnetometer is not enabled as a measurement... please check your configuration has "filter_enable_magnetometer_aiding : True"

Hope that helps,

Nathan

Alex-Beh, 2022-11-15:
The IMU model is CV7-AHRS. The device is sitting on the table and nothing around the sensor is vibrating.

I have set  ```filter_enable_magnetometer_aiding : True``` but same issue. Anything else I can do for debugging?

nathanmillerparker, 2022-11-15:
To address the heading issue, is the filter state getting to "3" (AHRS)?  This can be found in the "/nav/status" topic.  In this state, the filter is using the magnetometer measurements so there shouldn't be unbounded drift. 

Here is the manual page that addresses filter initialization: https://s3.amazonaws.com/files.microstrain.com/CV7+Online/user_manual_content/ekf/Filter%20States.htm

Also, depending on the magnetic interference around the device, you may have to do a mag calibration, which would require using SensorConnect (https://www.microstrain.com/software/sensorconnect)  A poorly calibrated magnetometer may cause the filter to reject the measurements.

The noise is a bit perplexing, though.... are you connected with USB or serial?  If serial, what is your power supply?  Also, have you tried it on a different computer (if USB power)?  Trying to isolate potential external power issues with these questions.

nathanmillerparker, 2022-11-15:
Hey Alex,

I just noticed that you were plotting linear acceleration (in m/s^2) in the video above and not imu acceleration (in Gs.)  Sorry for the confusion as it was hard to tell from the video quality, but this level of noise is expected (thus my comment about it being 10x what it should be, because I thought it was the IMU acceleration, in Gs.)  Note: the noise of the accels in the device are 30 μg/√Hz @ 8Gs... and in this case, they are sampled at 1 kHz... so everything lines up.  Also note, the data quantities coming out of the filter are not anti-aliased. 

Nathan

Alex-Beh, 2022-11-16:
Yes the filter_state of /nav/status is 3. So I am assuming the magnetometer is being taken into the consideration.

I had performed the mag calibration. Since my robot only allowed to move in 2D plane, so only manage to capture the orientation data around z-axis. May I know is it sufficient for the calibration? Or I need to unmount it and move the IMU in all direction?

I powered the IMU using the [development kit](https://www.mouser.sg/ProductDetail/130-DEVELOPTKITRS232).

Sorry to make you confused..... May I know the difference between linear acceleration & IMU acceleration? What do you mean by anti-aliased? Not smoothing?

nathanmillerparker, 2022-11-16:
Hello Alex,

You would want to keep the device attached to the robot to do the calibration, if you did remove it to calibrate, it would invalidate the calibration because the interference from the robot is likely significant.  You can do a calibration as you have done (basically, only capturing a circle of data), but it is important to set the magnitude of the local magnetic field manually.  There are options in the calibration tool to do this... and you can download the expected magnetic field using your lat/lon in the tool.

The differences between the acceleration reported from the IMU topic and the one reported from the nav topic are as follows:

1. IMU acceleration is reported in Gs, nav is reported in m/s^2
2. The IMU acceleration will be what comes from the sensor after temperature compensation, the nav acceleration is similar, but any estimate of accel bias will be removed.  The data coming from the nav topic is after the kalman filter, which is needed to estimate biases.  The imu topic data is pre-kalman filter.
3. The IMU data will be low-pass filtered (anti-aliased) based on the message rate you have set.  So, if you set the rate to 100 Hz, a 200Hz low pass filter is applied to smooth out the data.  This 2x low pass filter setting is true for all rates in the IMU dat.  The filter data does not currently have this smoothing, but will probably in a future firmware release (I do not have a release estimate for that version.)

Hope that helps.

Nathan

Alex-Beh, 2022-12-12:
Thanks for the advice & information.

closed 2022-12-12 by Alex-Beh