LORD-MicroStrain/microstrain_inertial #198: Receiving RTCM Corrections from Radio via Aux Port

Jad-Tawil, 2022-11-20:
Hello, 

We have a radio receiving RTCM corrections from a base station.
It has a RS232 serial output. 
We would like to connect this to the Aux port.
Which parameters do we need to configure in the driver?
For example, should the rtk_dongle_enable_ variable be False|True?
What other variables are relevant (baudrate for the aux port for example needs to match the baudrate of the radio)

nathanmillerparker, 2022-11-21:
Hi Jad,

The rtk_dongle_enable variable only needs to be set if you have 2-way communications with your corrections service and need a NMEA GGA message output.  This is typically only required for virtual reference station (VRS) corrections. 

The default baudrate of the aux port is 115200, but it can be changed, if necessary.  Here is the relevant command in the manual: https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/external_content/dcp/Commands/base_command/data/mip_cmd_base_comm_speed.htm

Available baudrates for the GQ7 are on this page:
https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/user_manual_content/specifications/Main_Aux.htm

Sounds like you are going to be doing a direct connection, so you wouldn't use the following, but if you did have a program that sat in-between the radio and the GQ7, you can configure the driver to listen to an RTCM topic that you publish on with the following settings in the driver:

# (GQ7 Only) Allow the user to send RTCM messages to this node, and stream those messages to the GQ7
subscribe_rtcm : False
rtcm_topic     : "/rtcm"

Hope that helps,

Nathan

Jad-Tawil, 2022-11-21:
That makes sense to me. 
Few questions:
1. Is the branch "ros" the most up-to-date branch for using ROS1?
2. rtk_dongle_enable_ and subscribe_rtcm will be False for us, since it is a direct connection, like you said. In this case, what settings do i need to configure on the device to inform the receiver to expect RTCM corrections on the serial input of the AUX port? Is simply plugging it in by default adequate?

closed 2023-01-01 by github-actions[bot]