LORD-MicroStrain/microstrain_inertial #200: Heading has an offset of 90° with 3DMCV7 AHRS in ENU and NED frames

Anton1B, 2022-11-29:
**Describe the bug**
When I try to read the value of /filtered_imu/data/orientation/yaw_deg, I systematically find an offset on the heading.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch Node with command `roslaunch microstrain_inertial_driver microstrain.launch use_enu_frame:=true` or `roslaunch microstrain_inertial_driver microstrain.launch use_enu_frame:=false`
2. Launch Plotjuggler to display yaw_deg 
3. See Error

**Expected behavior**
When use_enu_frame = False, I get 0° when facing West and 90° when facing North
When use_enu_frame = True, I get -90° when facing East and 0° when facing North

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Architecture: x86_64
 - ROS Version: noetic
 - Version: 2.0.0
 - Sensor(s): 3DMCV7 AHRS

**Modifications**
I Did not modify the source code in any way while testing. 

**Launch Parameters**
filter_enable_magnetometer_aiding     : True

**Additional context**
I made the calibration before testing the IMU
The heading is stable, just offsetted

nathanmillerparker, 2022-11-29:
Hi Anton,

Are you using a sensor2vehicle frame transformation?  Is the CV7 attached to a robot or just the connectivity board?  I assume, since you calibrated the unit, you have SensorConnect available.  If you stream the Euler angles in SensorConnect, do you see the same behavior (will only match the use_enu_frame = False case)? 

These questions will help us figure out what is going on.

Thanks,

Nathan

Anton1B, 2022-11-30:
Hi Nathan

No, I am not using it

You are right I am testing it from my laptop

This is what I get with SensorConnect 
[image]

In this test I have been rotating back and forth between North and West

nathanmillerparker, 2022-11-30:
Hi Anton,

Can you verify that the filter state is achieving a value of "3", or "AHRS"?

Here is a reference for that:
https://s3.amazonaws.com/files.microstrain.com/CV7+Online/user_manual_content/ekf/Filter%20States.htm

If it is not entering AHRS mode, then it is using a relative heading, which could explain this behavior and will direct us on which way to go.

If that does not work, it may be worth resetting the device to factory defaults using SensorConnect, which can be found under the "Save/Load Settings" tile and starting over.  It is possible that some setting got written into the Eeprom that is causing this (sensor-to-vehicle frame transformation, manual declination angle, or something not quite right with the calibration stick out in my mind.)

Nathan

Anton1B, 2022-12-01:
Hi,

I echoed **/nav/status** and got the following message : 
```
---
filter_state: 3
dynamics_mode: 0
status_flags: 1
---
```
So I guess I have AHRS mode activated

I'll try resetting it and I'll keep you updated if anything changes

Antonin

Anton1B, 2022-12-01:
I have reset the device to factory defaults with SensorConnect and I don't see any difference.
Also, don't mind the yaw in blue, I started the device facing east
[image]

nathanmillerparker, 2022-12-01:
Hey Antonin,

This is very odd behavior... can you take a couple more steps for me and we will get this fixed for you one way or another.

1. Export your settings file from SensorConnect and send it to me so I can check all your settings (Save/Load settings tile -> Export button)
2. Please provide where you purchased the unit and the serial number.
3. As a final check, could you conduct this same test outside, away from any possible magnetic interference?  The magnitude of the magnetometer vector shown in SensorConnect is only 0.25 gauss, which at the very bottom of what is to be expected.  It is possible that something around your test area may be causing this behavior.

Thank you,

Nathan

Anton1B, 2022-12-05:
Hi,

Indeed... I'll send you anything you need 

1. The settings : https://drive.google.com/file/d/1N6m9Keckio2_JCFNlKy_jfSqGSKPvozf/view?usp=share_link

2. I bought the sensor through the distributor Alliantech, and I have this file if it helps...
[microstrain_lord_calibration_report.pdf](https://github.com/LORD-MicroStrain/microstrain_inertial/files/10152319/microstrain_lord_calibration_report.pdf)

3. I think I have already tested this sensor outside and the result were pretty much the same but I'll make a new test

nathanmillerparker, 2022-12-05:
Hi Antonin,

I think the best path forward would be to return the device to Alliantech via RMA and have them ship a new one.  I cannot decipher if there is an issue with the device or if there is something consistent in the testing that would be causing an issue, but getting a new device should remove it as a possibility.  

The only thing I noticed from your settings file is that "North Compensation" for the complementary filter is not enabled, but that should only affect the bottom left plot of the screen caps you posted.  

Sorry for the inconvenience.  This is not a common problem, but we need to take a look at the device.

Let me know if you have an issues getting the RMA... I don't typically initiate them like this, but if you send this thread to Alliantech, hopefully they can get the process started.

Nathan

closed 2023-02-26 by github-actions[bot]

jage92, 2023-06-05:
I have the same issue with gx5-45 but not with gx5-25.