LORD-MicroStrain/microstrain_inertial #213: How to publish 3dm-gx3-25 linear velocity and acceleration in rostopic

wlbyyqkn, 2022-12-05:
Hello, I would like to ask the official website program run by ros noetic to publish the linear acceleration and angular acceleration to the terminal.   Is there anything in the program that needs to be changed?

Display after the program is run

... logging to /home/nuc/.ros/log/2565ed0e-746b-11ed-99e5-f336a839863f/roslaunch-nuc-virtual-machine-8760.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nuc-virtual-machine:41271/

SUMMARY
========

PARAMETERS
 * /microstrain_inertial_driver/aux_port: /dev/ttyUSB1
 * /microstrain_inertial_driver/baudrate: 115200
 * /microstrain_inertial_driver/debug: False
 * /microstrain_inertial_driver/device_setup: True
 * /microstrain_inertial_driver/diagnostics: False
 * /microstrain_inertial_driver/enable_hardware_odometer: False
 * /microstrain_inertial_driver/filter_adaptive_level: 2
 * /microstrain_inertial_driver/filter_adaptive_time_limit_ms: 15000
 * /microstrain_inertial_driver/filter_aiding_measurement_summary_data_rate: -1
 * /microstrain_inertial_driver/filter_aiding_status_data_rate: -1
 * /microstrain_inertial_driver/filter_angular_zupt: True
 * /microstrain_inertial_driver/filter_angular_zupt_topic: /moving_ang
 * /microstrain_inertial_driver/filter_auto_heading_alignment_selector: 1
 * /microstrain_inertial_driver/filter_auto_init: True
 * /microstrain_inertial_driver/filter_child_frame_id: sensor
 * /microstrain_inertial_driver/filter_data_rate: 10
 * /microstrain_inertial_driver/filter_declination: 0.23
 * /microstrain_inertial_driver/filter_declination_source: 2
 * /microstrain_inertial_driver/filter_dynamics_mode: 1
 * /microstrain_inertial_driver/filter_enable_altimeter_aiding: False
 * /microstrain_inertial_driver/filter_enable_external_gps_time_update: False
 * /microstrain_inertial_driver/filter_enable_external_heading_aiding: False
 * /microstrain_inertial_driver/filter_enable_gnss_antenna_cal: False
 * /microstrain_inertial_driver/filter_enable_gnss_heading_aiding: True
 * /microstrain_inertial_driver/filter_enable_gnss_pos_vel_aiding: True
 * /microstrain_inertial_driver/filter_enable_magnetometer_aiding: False
 * /microstrain_inertial_driver/filter_enable_odometer_aiding: False
 * /microstrain_inertial_driver/filter_enable_vertical_gyro_constraint: False
 * /microstrain_inertial_driver/filter_enable_wheeled_vehicle_constraint: False
 * /microstrain_inertial_driver/filter_external_gps_time_topic: /external_gps_time
 * /microstrain_inertial_driver/filter_external_speed_topic: /external_speed
 * /microstrain_inertial_driver/filter_frame_id: sensor_wgs84
 * /microstrain_inertial_driver/filter_gnss_antenna_cal_max_offset: 0.1
 * /microstrain_inertial_driver/filter_gnss_dual_antenna_data_rate: -1
 * /microstrain_inertial_driver/filter_heading_data_rate: -1
 * /microstrain_inertial_driver/filter_heading_source: 1
 * /microstrain_inertial_driver/filter_heading_state_data_rate: -1
 * /microstrain_inertial_driver/filter_imu_data_rate: -1
 * /microstrain_inertial_driver/filter_init_attitude: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_init_condition_src: 0
 * /microstrain_inertial_driver/filter_init_position: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_init_reference_frame: 2
 * /microstrain_inertial_driver/filter_init_velocity: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_lever_arm_offset: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_nmea_gga_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_gll_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_hdt_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_prka_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_rmc_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_talker_id: 3
 * /microstrain_inertial_driver/filter_odom_data_rate: -1
 * /microstrain_inertial_driver/filter_pps_source: 1
 * /microstrain_inertial_driver/filter_relative_odom_data_rate: -1
 * /microstrain_inertial_driver/filter_relative_position_config: False
 * /microstrain_inertial_driver/filter_relative_position_frame: 2
[... 153 lines cut ...]
[ INFO] [1670223916.012310654]: Note: The device does not support the odometer settings command
[ INFO] [1670223916.035282365]: Note: The device does not support the factory streaming channels setup command
[ INFO] [1670223916.057294743]: Note: The device does not support the SBAS settings command
[ INFO] [1670223916.077359108]: Note: The device does not support the nmea message format command
[ INFO] [1670223916.098374903]: Note: Device does not support the RTK dongle config command
[ INFO] [1670223916.132357207]: Note: Device does not support the declination source command.
[ INFO] [1670223916.174355091]: Note: Device does not support GNSS antenna offsets
[ INFO] [1670223916.194362680]: Note: The device does not support the vehicle dynamics mode command.
[ INFO] [1670223916.214457694]: Note: The device does not support the heading source command.
[ INFO] [1670223916.236521239]: Note: The device does not support the filter autoinitialization command.
[ INFO] [1670223916.256378773]: Note: The device does not support the filter adaptive settings command.
[ INFO] [1670223916.276399799]: Note: The device does not support the filter aiding command.
[ INFO] [1670223916.298405614]: Note: The device does not support the relative position configuration command
[ INFO] [1670223916.318387364]: Note: The device does not support the filter speed lever arm command.
[ INFO] [1670223916.338399120]: Note: The device does not support the wheeled vehicle constraint command.
[ INFO] [1670223916.358457284]: Note: The device does not support the vertical gyro constraint command.
[ INFO] [1670223916.378536667]: Note: The device does not support the GNSS antenna calibration command.
[ INFO] [1670223916.398431934]: Note: The device does not support the next-gen filter initialization command.
[ WARN] [1670223916.438506508]: Note: The device does not support the sensor to vehicle transformation or rotation euler command
[ INFO] [1670223916.458475569]: Note: The device does not support the reference point lever arm command
[ INFO] [1670223916.458551953]: Note: The settings were not saved as startup settings. Power cycling will remove changes from device
[ WARN] [1670223916.480484530]: Device does not support the filter reset command
[ INFO] [1670223917.680722588]: Setting spin rate to <1.000000> hz
[ INFO] [1670223917.724801186]: Resuming the device data streams
[ INFO] [1670223917.729825691]: Starting Data Parsing

robbiefish, 2022-12-05:
@wlbyyqkn Thank you for bringing this to our attention. This should be fixed in #216 for ROS and #215 for ROS2. If possible, could you test on these branches and let me know if that fixes your issue? Otherwise, they should be merged by the end of the day

wlbyyqkn, 2022-12-05:
Of course, thank you very much for your answer. If you can update complete , please let me know or send me the link. Thank you very much！ 
 

curtain?
***@***.***

robbiefish, 2022-12-05:
@wlbyyqkn the fix has been merged into the main branches. Please update your clone with:
```bash
git pull origin ros --recurse-submodules
```
or if using ROS2
```bash
git pull origin ros2 --recurse-submodules
```

closed 2023-01-08 by github-actions[bot]