LORD-MicroStrain/microstrain_inertial #226: 3DM-GX5-25 new sensor configuration and usage with ROS

ninja-hu, 2023-02-04:
We have a couple of inertial sensors, however all of them seem to have some issue with the Yaw orientation (haven't tested the other values so far). It always drifts or gives an incorrect value every time the sensor is started.  

Following up the [conversation in this bug](https://github.com/LORD-MicroStrain/microstrain_inertial/issues/65), does setting the device_setup parameter in the launch file overwrite anything? It's likely that this was done for a couple of times. However I see that the save_settings parameter is set to false and this was not changed which would essentially mean that the any changes were not written in the device's memory. 

We did a factory reset of one of the sensors and did an auto calibration of the gyro and magnetometer. However the problem still persists. Can you please assist in debugging the problem? Thanks!

nathanmillerparker, 2023-02-06:
Hello,

Can you please provide the model of the device you are having trouble with?  This will help us provide the right recommendation.

Thank you,

Nathan

nathanmillerparker, 2023-02-06:
The GX5-25 uses the magnetometer as a heading reference and it is on by default.  The device_setup parameter does perform a function: if true, the ROS driver settings will take affect (recommended); if false, it is assumed you have set the device up exactly how you want it by some other means (SensorConnect or your own setup code) and stored those settings in the device (this is for advanced users.)  And you are correct, by default, the "save_settings" option is set to default so the ROS setup is not stored in the device.

Is it possible that you have placed the device too close to interfering magnetic fields?  To test for this, you can connect to the device with SensorConnect, while it is mounted to the robot and do a manual magnetic calibration (which is recommended on each installation.)   It is important that the device and robot are rotated together during this process, as doing a magnetic calibration with the device off of the robot would give invalid results.  Also, you want to mount the device as far away from changing magnetic fields (like motors) as possible.  

This user manual entry is for our newest device in this line, but the steps are very similar for the GX5:
https://s3.amazonaws.com/files.microstrain.com/CV7+Online/user_manual_content/installation/Magnetometer%20Calibration.htm

Let us know if that helps!

Nathan

nathanmillerparker, 2023-02-06:
No problem.

1. You may need to calibrate for changing environments.  We recommend you calibrate where you plan to operate.  Because the Earth's magnetic field is weak, building structure (and metal desks, etc.) can influence the quality of your calibration.  I'm not sure what your application is, so it is hard to provide a concrete suggestion, but following the suggestion to "calibrate where you plan to operate" is a good one.

2. The yaw coming from the "/imu/data" topic will be from the complementary filter, which is a backup to the regular Kalman filter output.  So, I'd use the "/nav/odom" quaternion output, which is the kalman filter.  The "nav/heading" is just a breakout topic for the heading in the "nav/odom" message for customers who requested it.

3. The "nav/relative_pos/odom" topic is only available on the GQ7 (GNSS/INS) as of now.

Hope that helps!

Nathan

ninja-hu, 2023-02-06:
Great! Thanks for the clarification. I'll do some tests in the next couple of days and close the bug once everything works fine.

jage92, 2023-06-08:
In respect with this, I have two issues. I used autocalibration of the magnetometer and before 10 minutes I get a good heading, before is impossible. I am using /nav/odom topic. During this time, I observed /imu/data orientation and /nav/odom orientation and are very different, when one is in the north the other is in the east, for example. After a time they are similar and the /nav/odom orientation is good. During this time, the robot moves and rotate. Why does it take so much time?

The other issue is that I keep the autocalibration configuration and sometimes after different test (more or less an hour) the heading is wrong and also the both topics have very different heading. 

Thanks for your time.

nathanmillerparker, 2023-06-08:
Hi,

Autocalibration isn't designed to take out all of the effects of a badly calibrated magnetometer... only to fine-tune the calibration.  I don't recommend that you rely entirely on the auto-cal, but instead perform a manual calibration and then allow the auto-cal to adjust it as necessary for any **_small_** changes in magnetic environment.    

From what you are describing, I believe 1 or 2 things are probably happening:

1. The robot has a varying magnetic field, either due to moving parts or magnetic interference sources (motors, etc.) 
2. The environment that you are operating has external magnetic interference causing changing magnetic fields that cannot be calibrated.

A lot can go into ensuring an AHRS will provide a stable, accurate heading... all of which revolve around making sure the AHRS is not experiencing these changing magnetic fields.  Placement of the AHRS on your robot is very important and can drastically impact the performance.  The operating environment can as well.

This is a difficult forum to provide support for this type of issue as images of your installation / operating environment and recorded data are how we typically diagnose these issues.

Please contact our support at https://www.microstrain.com/contact-support and reference this issue as well as my name - Nathan.  If you could prepare pictures of everything to send once we establish your support request, that would help!

Thanks,

Nathan

jage92, 2023-06-13:
As soon as possible, I will try to reproduce the issues and collect data to send to support. Thank you for your response.

nathanmillerparker, 2023-06-13:
Sounds good... please enable binary data recording in the driver... we will need the binary file, not a bag file, to do our analysis.

Thank you,

Nathan

jage92, 2023-09-26:
I calibrated manually in all directions near to the place that the IMU will be placed, and with the World Magnetic Model option I get better results. I will write to support to more personal help, but I wanted to leave this message to help other people.

When no need to use rotation to frame vehicle it is important to put the parameter in None=0, because with other option without rotation the attitude is relative to the start orientation.

closed 2024-05-05 by github-actions[bot]