LORD-MicroStrain/microstrain_inertial #231: can i know how to switch changing euler axis?

himhan34, 2023-02-12:
**Describe the bug**
https://github.com/LORD-MicroStrain/microstrain_inertial/issues/147

in the upper site, there is an Explanation of the changing the Euler axis 

 - Sensor(s): 3DM-GQ7
 - in this picture. 
[image]

what is meaning to let 0 to 3.14, is it just around rotating, a particular axis? 
I want to know that. because I thought the 0 to 3.14 was switching the, plus to minus.

nathanmillerparker, 2023-02-13:
Hello,

Not quite sure what is being asked, but hopefully this helps:

filter_sensor2vehicle_frame_transformation_euler : [roll, pitch, yaw] <- these are in radians

So, depending on how your device is oriented, you would set the values for those angles appropriately.  Please see this page for help: https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/user_manual_content/coordinate_frames/Vehicle%20Frame.htm

Hope that helps,

Nathan

himhan34, 2023-02-13:
umm, then you meant that wherever I am, I can change the angle of the imu?
is it possible to change the axis position? like, yaw axis to roll axis, or roll axis to pitch axis?

nathanmillerparker, 2023-02-13:
Yes, that is possible (with some limitations) and is the reason for this parameter.  

https://en.wikipedia.org/wiki/Euler_angles has a detailed discussion of Euler Angles and their use to help you understand what you would like to accomplish.

Hope that helps!

Nathan

himhan34, 2023-03-07:
Hi zyx,

The rotations about the IMU axis are not as described in the image above. A roll would be around the x axis, a pitch about the y, and a positive rotation on the z axis is in the opposite direction shown in the image.

To answer your question, we provide a sensor2vehicle transformation in the driver that will allow for the axis to change, but not the + rotation direction. For your purposes, it would be a yaw of 180, a roll of 180, and a pitch of 0. So you could use the Euler angle definition and put in [0, 3.14, 3.14] (these are in roll, pitch, yaw order.)

filter_sensor2vehicle_frame_selector : 1 filter_sensor2vehicle_frame_transformation_euler : [3.14,0, 3.14]

Hope that helps!

Nathan

hello i haveam using the same repo for convienence, if you are okey can you tell me why
the value in the euler to body frame is (3.14, 0, 3.14)?

himhan34, 2023-03-07:
i am seeing this repo as well 
#147
https://github.com/LORD-MicroStrain/microstrain_inertial/issues/147

closed 2023-04-16 by github-actions[bot]