LORD-MicroStrain/microstrain_inertial #238: 3DMGX5-AHRS Startup issue, "Failed to resume device data streams", Timeout

pjmara, 2023-03-17:
**Describe the bug**
Our 3DMGX5-AHRS has been experienceing intermittent startup issues. The error shows up in ros logs as "Failed to resume device data streams", then "unable to activate node base"

**To Reproduce**
Steps to reproduce the behavior:
1. Launch ROS microstrain inertial driver with the environment described below
2. See failure 10-30% of the time, resulting in no messages on the /imu/data topic

**Expected behavior**
Launch the ROS node and see messages publishing on the /imu/data topic

**Environment (please complete the following information):**
 - OS: Debian Buster
 - Architecture: arm64
 - ROS Version: noetic
 - Version: 3.0.1-1 (installed when it was in the ros-testing repo)
 - Sensor(s): 3DMGX5-AHRS 

**Modifications**
As noted above, we installed microstrain_inertial_driver back when it was in the ros-testing repository as we had been seeing rate issues (our imu is configured for raw data at 500 Hz)

**Launch Parameters**
```
# Standalone example params file for GX3, GX4, GX/CX5, RQ1 and GQ7 series devices
# Note: Feature support is device-dependent and some of the following settings may have no affect on your device.
# Please consult your device's documentation for supported features

# ******************************************************************
# NOTE: This file is formatted to work with ROS and will not work if specified as the params_file argument in ROS2.
#       If you want to override parameters for ROS2, start with https://github.com/LORD-MicroStrain/microstrain_inertial/blob/ros2/microstrain_inertial_driver/config/empty.yml
# ******************************************************************

# ******************************************************************
# General Settings
# ******************************************************************

# port is the main port that the device will communicate over. For all devices except the GQ7, this is the only available port.
# aux_port is only available for the GQ7 and is only needed when streaming RTCM corrections to the device from ROS, or if you want to publish NMEA sentences from this node
port        : "/dev/microstrain_imu"
aux_port    : "/dev/microstrain_imu"
baudrate    : 115200
debug       : False
diagnostics : False

# Frame IDs used in the different messages. By default these are set to arbitrary strings as not to interfere with other ROS services.
# For more information on common frame IDs, check out: https://www.ros.org/reps/rep-0105.html
#
# filter_frame_id and filter_child_frame_id are specifically useful as the node will also publish a transform to the /tf topic
# that contains the transform between these two frames. Many ROS tools such as RViz will use the /tf topic to display things like robot position.
imu_frame_id          : "sensor"
gnss1_frame_id        : "gnss1_antenna_wgs84"
gnss2_frame_id        : "gnss2_antenna_wgs84"
filter_frame_id       : "sensor_wgs84"
filter_child_frame_id : "sensor"
nmea_frame_id         : "nmea"

# Waits for a configurable amount of time until the device exists
# If poll_max_tries is set to -1 we will poll forever until the device exists
poll_port      : False
poll_rate_hz   : 1.0
poll_max_tries : 60

# Controls if the driver outputs data with-respect-to ENU frame
#      false - position, velocity, and orientation are WRT the NED frame (native device frame)
#      true  - position, velocity, and orientation are WRT the ENU frame
use_enu_frame : True

# Controls if the driver-defined setup is sent to the device
#      false - The driver will ignore the settings below and use the device's current settings
#      true  - Overwrite the current device settings with those listed below
[... 325 lines cut ...]
1678994963.314631132 INFO /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/microstrain_inertial_driver_common/src/config.cpp:504(Config::configureGNSS)] [topics: /rosout] Note: Device does not support the RTK dongle config command
1678994963.574404835 INFO /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/microstrain_inertial_driver_common/src/config.cpp:589(Config::configureFilter)] [topics: /rosout] Setting Declination Source to 2 0.230000
1678994963.583196171 INFO /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/microstrain_inertial_driver_common/src/config.cpp:632(Config::configureFilter)] [topics: /rosout] Note: Device does not support GNSS antenna offsets
1678994963.583283911 INFO /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/microstrain_inertial_driver_common/src/config.cpp:647(Config::configureFilter)] [topics: /rosout] Note: The device does not support the vehicle dynamics mode command.
1678994963.583320818 INFO /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/microstrain_inertial_driver_common/src/config.cpp:653(Config::configureFilter)] [topics: /rosout] Setting heading source to 0
1678994963.588540542 INFO /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/microstrain_inertial_driver_common/src/config.cpp:662(Config::configureFilter)] [topics: /rosout] Setting initial heading to 0.000000
1678994963.594661665 INFO /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/microstrain_inertial_driver_common/src/config.cpp:687(Config::configureFilter)] [topics: /rosout] Setting autoinitialization to 1
1678994963.600760547 INFO /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/microstrain_inertial_driver_common/src/config.cpp:711(Config::configureFilter)] [topics: /rosout] Note: The device does not support the filter adaptive settings command.
1678994963.600879065 INFO /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/microstrain_inertial_driver_common/src/config.cpp:727(Config::configureFilter)] [topics: /rosout] Note: The device does not support the filter aiding command.
1678994963.600913083 INFO /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/microstrain_inertial_driver_common/src/config.cpp:750(Config::configureFilter)] [topics: /rosout] Note: The device does not support the relative position configuration command
1678994963.600951527 INFO /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/microstrain_inertial_driver_common/src/config.cpp:765(Config::configureFilter)] [topics: /rosout] Note: The device does not support the filter speed lever arm command.
1678994963.600988212 INFO /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/microstrain_inertial_driver_common/src/config.cpp:780(Config::configureFilter)] [topics: /rosout] Note: The device does not support the wheeled vehicle constraint command.
1678994963.601015934 INFO /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/microstrain_inertial_driver_common/src/config.cpp:795(Config::configureFilter)] [topics: /rosout] Note: The device does not support the vertical gyro constraint command.
1678994963.601051637 INFO /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/microstrain_inertial_driver_common/src/config.cpp:810(Config::configureFilter)] [topics: /rosout] Note: The device does not support the GNSS antenna calibration command.
1678994963.601086507 INFO /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/microstrain_inertial_driver_common/src/config.cpp:837(Config::configureFilter)] [topics: /rosout] Note: The device does not support the next-gen filter initialization command.
1678994963.601135358 INFO /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/microstrain_inertial_driver_common/src/config.cpp:851(Config::configureFilter)] [topics: /rosout] Setting sensor to vehicle rotation euler to [-0.000000, -0.000000, -0.000000]
1678994963.606673468 INFO /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/microstrain_inertial_driver_common/src/config.cpp:970(Config::configureFilter)] [topics: /rosout] Note: The device does not support the reference point lever arm command
[... cut ...]
1678994965.074602071 ERROR /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/microstrain_inertial_driver_common/src/node_common.cpp:173(NodeCommon::activate)] [topics: /rosout, /imu/data, /mag, /gps_corr, /nav/status, /nav/heading, /nav/heading_state, /nav/odom, /nav/filtered_imu/data, /tf] Error(-4): Timed out
1678994965.074662941 FATAL /microstrain_inertial_driver [/tmp/binarydeb/ros-noetic-microstrain-inertial-driver-3.0.1/src/microstrain_inertial_driver.cpp:58(Microstrain::run)] [topics: /rosout, /imu/data, /mag, /gps_corr, /nav/status, /nav/heading, /nav/heading_state, /nav/odom, /nav/filtered_imu/data, /tf] Unable to activate node base
```

robbiefish, 2023-03-17:
I am unable to reproduce this issue, but the first thing that comes to mind is a bandwith issue. If the device is configured to stream more data than the connection can handle, or if it is right at the edge, the response from the resume command may get lost, which will cause the driver to error and crash.

* How is the device connected? serial or USB?
  * If it is connected via serial:
    * Do you have a USB cable you could use to connect, and does the problem persist with the USB cable?
    * Have you tried increasing the baudrate? Note that if you do increase the baudrate (would recommend 921600), you will need to set both `baudrate` and `set_baud` to `True` in order to set the device to communicate at the same baud.
* Could you launch the node with the option `debug:=true` and provide the output?

pjmara, 2023-03-20:
> * How is the device connected? serial or USB?
 * usb, using the provided USB cable from Microstrain
> * Could you launch the node with the option `debug:=true` and provide the output?
* working on this for you. Hoping to get it today

Is it possible to be a bandwidth issue over USB? would something like increasing the USB buffer sizes be something to try?

closed 2023-04-23 by github-actions[bot]

JMvanBruggen, 2023-06-22:
Hey, 

I have the same issue here, also on ARM64 architecture. 
I tried increasing the baudrate, but that did not work: 

`SUMMARY
========

PARAMETERS
 * /microstrain_inertial_driver/aux_port: /dev/ttyACM1
 * /microstrain_inertial_driver/baudrate: 921600
 * /microstrain_inertial_driver/debug: False
 * /microstrain_inertial_driver/device_setup: True
 * /microstrain_inertial_driver/diagnostics: False
 * /microstrain_inertial_driver/enable_hardware_odometer: False
 * /microstrain_inertial_driver/filter_adaptive_level: 2
 * /microstrain_inertial_driver/filter_adaptive_time_limit_ms: 15000
 * /microstrain_inertial_driver/filter_aiding_measurement_summary_data_rate: -1
 * /microstrain_inertial_driver/filter_aiding_status_data_rate: -1
 * /microstrain_inertial_driver/filter_angular_zupt: True
 * /microstrain_inertial_driver/filter_angular_zupt_topic: /moving_ang
 * /microstrain_inertial_driver/filter_auto_heading_alignment_selector: 1
 * /microstrain_inertial_driver/filter_auto_init: True
 * /microstrain_inertial_driver/filter_child_frame_id: sensor
 * /microstrain_inertial_driver/filter_data_rate: 10
 * /microstrain_inertial_driver/filter_declination: 0.23
 * /microstrain_inertial_driver/filter_declination_source: 2
 * /microstrain_inertial_driver/filter_dynamics_mode: 1
 * /microstrain_inertial_driver/filter_enable_altimeter_aiding: False
 * /microstrain_inertial_driver/filter_enable_external_gps_time_update: False
 * /microstrain_inertial_driver/filter_enable_external_heading_aiding: False
 * /microstrain_inertial_driver/filter_enable_gnss_antenna_cal: False
 * /microstrain_inertial_driver/filter_enable_gnss_heading_aiding: True
 * /microstrain_inertial_driver/filter_enable_gnss_pos_vel_aiding: True
 * /microstrain_inertial_driver/filter_enable_magnetometer_aiding: False
 * /microstrain_inertial_driver/filter_enable_odometer_aiding: False
 * /microstrain_inertial_driver/filter_enable_vertical_gyro_constraint: False
 * /microstrain_inertial_driver/filter_enable_wheeled_vehicle_constraint: False
 * /microstrain_inertial_driver/filter_external_gps_time_topic: /external_gps_time
 * /microstrain_inertial_driver/filter_external_speed_topic: /external_speed
 * /microstrain_inertial_driver/filter_frame_id: sensor_wgs84
 * /microstrain_inertial_driver/filter_gnss_antenna_cal_max_offset: 0.1
 * /microstrain_inertial_driver/filter_gnss_dual_antenna_data_rate: -1
 * /microstrain_inertial_driver/filter_heading_data_rate: -1
 * /microstrain_inertial_driver/filter_heading_source: 1
 * /microstrain_inertial_driver/filter_heading_state_data_rate: -1
 * /microstrain_inertial_driver/filter_imu_data_rate: -1
 * /microstrain_inertial_driver/filter_init_attitude: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_init_condition_src: 0
 * /microstrain_inertial_driver/filter_init_position: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_init_reference_frame: 2
 * /microstrain_inertial_driver/filter_init_velocity: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_lever_arm_offset: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_nmea_gga_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_gll_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_hdt_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_prka_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_rmc_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_talker_id: 3
 * /microstrain_inertial_driver/filter_odom_data_rate: -1
 * /microstrain_inertial_driver/filter_pps_source: 1
 * /microstrain_inertial_driver/filter_relative_odom_data_rate: -1
 * /microstrain_inertial_driver/filter_relative_position_config: False
 * /microstrain_inertial_driver/filter_relative_position_frame: 2
 * /microstrain_inertial_driver/filter_relative_position_ref: [0, 0, 0.01]
 * /microstrain_inertial_driver/filter_reset_after_config: True
 * /microstrain_inertial_driver/filter_sensor2vehicle_frame_selector: 1
 * /microstrain_inertial_driver/filter_sensor2vehicle_frame_transformation_euler: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_sensor2vehicle_frame_transformation_matrix: [1.0, 0.0, 0.0, 0...
 * /microstrain_inertial_driver/filter_sensor2vehicle_frame_transformation_quaternion: [0.0, 0.0, 0.0, 1.0]
[... 100 lines cut ...]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 518d34c4-10e6-11ee-a7f2-48b02d83f722
process[rosout-1]: started with pid [99016]
started core service [/rosout]
process[microstrain_inertial_driver-2]: started with pid [99023]
[ INFO] [1687429190.168461490]: Running microstrain_inertial_driver version: 3.0.0
[ INFO] [1687429190.169327386]: Using MIP SDK version: v1.0.0-21-gb1e6278
[ INFO] [1687429190.172803898]: Attempting to open serial port </dev/serial/by-id/usb-Lord_Microstrain_Lord_Inertial_Sensor_0000_6251.159161-if00> at <921600>
[ INFO] [1687429190.175870391]: Setting device to idle in order to configure
[ INFO] [1687429196.182765007]: Note: Attempting to open device at different bauds to change the baudrate
[ INFO] [1687429196.191814147]: Attempting to open serial port </dev/serial/by-id/usb-Lord_Microstrain_Lord_Inertial_Sensor_0000_6251.159161-if00> at <115200>
[ INFO] [1687429202.211204432]: Attempting to open serial port </dev/serial/by-id/usb-Lord_Microstrain_Lord_Inertial_Sensor_0000_6251.159161-if00> at <460800>
[ INFO] [1687429208.226524247]: Attempting to open serial port </dev/serial/by-id/usb-Lord_Microstrain_Lord_Inertial_Sensor_0000_6251.159161-if00> at <230400>
[... cut ...]
log file: /home/ada/.ros/log/518d34c4-10e6-11ee-a7f2-48b02d83f722/microstrain_inertial_driver-2*.log`

For me the only option to get it to work again is unplug and plug the usb again, or reset usb ports of my device, which is not a very sophisticated solution. Any more ideas on what this could be? 

Note:
I also have this error with release 2.7.1. There it iss really predictable. It works only the first run, second run it has the issue above. For some reason 3.0.0 and 3.0.1 have it only ~10% of the time.