LORD-MicroStrain/microstrain_inertial #241: mip_ interface_ recv_ from_ Device() function cannot read data. The amount(variable count = 0) of data read is 0, resulting in no updates to the data of the imu.

HoiCeong, 2023-03-20:
Hi,
  I ran the Lord's official driver form `https://github.com/LORD-MicroStrain/microstrain_inertial`. And found that most of the time, the amount of data read is 0 (as shown in the following image), resulting in no updates to the data of the imu.

I want to know the reason for this phenomenon, and **I hope that every time this function is run, the amount of data read out cannot be zero.**
[image]
[image]

robbiefish, 2023-03-20:
Hi @HoiCeong,

That function call will return 0 bytes when the port has no data available to be read, so this behavior may not be a problem depending on the circumstances. I need some more information to help you with this issue.

* Did you modify any of the driver's configuration when conducting this test? If so, can you please provide me with the changed configuration
* Can you please provide the full logs from the driver after launching with the argument `debug:=true`?

HoiCeong, 2023-03-21:
- **Did you modify any of the driver's configuration when conducting this test? If so, can you please provide me with the changed configuration**
 I haven't made any changes except to change the output frequency from 100 to 500.(imu_data_rate : 500).Just clone directly from github, and then compile and run directly.
- **Can you please provide the full logs from the driver after launching with the argument debug:=true?**
 here is the log with the argument `debug:=true`
```
# haichang @ haichang-Latitude-7490 in ~/Desktop/shanghai/lord_test_ws [10:37:39] 
$ source devel/setup.zsh

# haichang @ haichang-Latitude-7490 in ~/Desktop/shanghai/lord_test_ws [10:37:40] 
$ roslaunch microstrain_inertial_driver microstrain.launch debug:=true
... logging to /home/haichang/.ros/log/5bd13168-c791-11ed-928f-e4b97a4c68de/roslaunch-haichang-Latitude-7490-22019.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://haichang-Latitude-7490:40567/

SUMMARY
========

PARAMETERS
 * /microstrain_inertial_driver/aux_port: /dev/ttyACM1
 * /microstrain_inertial_driver/baudrate: 115200
 * /microstrain_inertial_driver/debug: True
 * /microstrain_inertial_driver/device_setup: True
 * /microstrain_inertial_driver/diagnostics: False
 * /microstrain_inertial_driver/enable_hardware_odometer: False
 * /microstrain_inertial_driver/filter_adaptive_level: 2
 * /microstrain_inertial_driver/filter_adaptive_time_limit_ms: 15000
 * /microstrain_inertial_driver/filter_aiding_measurement_summary_data_rate: -1
 * /microstrain_inertial_driver/filter_aiding_status_data_rate: -1
 * /microstrain_inertial_driver/filter_angular_zupt: True
 * /microstrain_inertial_driver/filter_angular_zupt_topic: /moving_ang
 * /microstrain_inertial_driver/filter_auto_heading_alignment_selector: 1
 * /microstrain_inertial_driver/filter_auto_init: True
 * /microstrain_inertial_driver/filter_child_frame_id: sensor
 * /microstrain_inertial_driver/filter_data_rate: 10
 * /microstrain_inertial_driver/filter_declination: 0.23
 * /microstrain_inertial_driver/filter_declination_source: 2
 * /microstrain_inertial_driver/filter_dynamics_mode: 1
 * /microstrain_inertial_driver/filter_enable_altimeter_aiding: False
 * /microstrain_inertial_driver/filter_enable_external_gps_time_update: False
 * /microstrain_inertial_driver/filter_enable_external_heading_aiding: False
 * /microstrain_inertial_driver/filter_enable_gnss_antenna_cal: False
 * /microstrain_inertial_driver/filter_enable_gnss_heading_aiding: True
 * /microstrain_inertial_driver/filter_enable_gnss_pos_vel_aiding: True
 * /microstrain_inertial_driver/filter_enable_magnetometer_aiding: False
 * /microstrain_inertial_driver/filter_enable_odometer_aiding: False
 * /microstrain_inertial_driver/filter_enable_vertical_gyro_constraint: False
 * /microstrain_inertial_driver/filter_enable_wheeled_vehicle_constraint: False
 * /microstrain_inertial_driver/filter_external_gps_time_topic: /external_gps_time
 * /microstrain_inertial_driver/filter_external_speed_topic: /external_speed
 * /microstrain_inertial_driver/filter_frame_id: sensor_wgs84
 * /microstrain_inertial_driver/filter_gnss_antenna_cal_max_offset: 0.1
 * /microstrain_inertial_driver/filter_gnss_dual_antenna_data_rate: -1
 * /microstrain_inertial_driver/filter_heading_data_rate: -1
 * /microstrain_inertial_driver/filter_heading_source: 1
 * /microstrain_inertial_driver/filter_heading_state_data_rate: -1
 * /microstrain_inertial_driver/filter_imu_data_rate: -1
 * /microstrain_inertial_driver/filter_init_attitude: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_init_condition_src: 0
 * /microstrain_inertial_driver/filter_init_position: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_init_reference_frame: 2
 * /microstrain_inertial_driver/filter_init_velocity: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_lever_arm_offset: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_nmea_gga_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_gll_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_hdt_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_prka_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_rmc_data_rate: 0
[... 434 lines cut ...]
mip_interface_default_update wait_time : 0, max_count: 242, count: 0.
mip_interface_default_update wait_time : 0, max_count: 242, count: 0.
mip_interface_default_update wait_time : 0, max_count: 242, count: 0.
mip_interface_default_update wait_time : 0, max_count: 242, count: 198.
mip_interface_default_update wait_time : 0, max_count: 44, count: 0.
mip_interface_default_update wait_time : 0, max_count: 44, count: 0.
mip_interface_default_update wait_time : 0, max_count: 44, count: 0.
mip_interface_default_update wait_time : 0, max_count: 44, count: 0.
mip_interface_default_update wait_time : 0, max_count: 44, count: 0.
mip_interface_default_update wait_time : 0, max_count: 44, count: 44.
mip_interface_default_update wait_time : 0, max_count: 980, count: 154.
mip_interface_default_update wait_time : 0, max_count: 870, count: 0.
mip_interface_default_update wait_time : 0, max_count: 870, count: 0.
mip_interface_default_update wait_time : 0, max_count: 870, count: 0.
mip_interface_default_update wait_time : 0, max_count: 870, count: 0.
^C^C[rosout-1] killing on exit
^C[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

# haichang @ haichang-Latitude-7490 in ~/Desktop/shanghai/lord_test_ws [10:37:51] 
$ 

```

HoiCeong, 2023-03-24:
@robbiefish 
Thank you for your answers and suggestions. Next, I plan to use the hardware serial port  to transmit data.
And here comes the question,  
- Does this software driver on github(`https://github.com/LORD-MicroStrain/microstrain_inertial`) fit into the hardware serial port？
- Can I obtain data through the hardware serial port after the following modifications？If not, where can I find the software code for using the hardware serial port？
[image]

closed 2023-04-23 by github-actions[bot]