LORD-MicroStrain/microstrain_inertial #243: Failed to make transition

bbokser, 2023-03-24:
I am using a 3DMCV7-AHRS and am attempting to run the Docker image found [here](https://hub.docker.com/r/microstrain/ros-microstrain_inertial_driver) to create a **ROS 2** node to subscribe to. I am encountering an error preventing the node from starting.

A summary of what I am doing:

1. The 3DMCV7-AHRS is connected to the C-Series Connectivity Board, which is connected by USB to my laptop.
        OS: **WSL2** Ubuntu (Windows 11 Host)
        Architecture: x86_64
        ROS Version: humble
        USB attached following [this guide](https://learn.microsoft.com/en-us/windows/wsl/connect-usb)
 2. I attempt to start the node:
   ```
   sudo docker run \
      -it \
      --rm \
      --net=host \
      --device=/dev/ttyACM0 \
      -v "/home/ben-wsl/git_workspace/source/up-xtreme-workspace/src/sub_cv7/params.yml:/tmp/params.yml" \
      microstrain/ros-microstrain_inertial_driver:ros2 params_file:=/tmp/params.yml \
      configure:=true activate:=true \
      port:=/dev/ttyACM0
[INFO] [launch]: All log files can be found below /home/microstrain/.ros/log/2023-03-24-17-19-44-286299-docker-desktop-1
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [microstrain_inertial_driver_node-1]: process started with pid [61]
[microstrain_inertial_driver_node-1] [INFO] [1679678384.710413052] [microstrain_inertial_driver]: Using MSCL Version: 64.2.2
[microstrain_inertial_driver_node-1] [INFO] [1679678384.710553115] [microstrain_inertial_driver]: Attempting to open serial port </dev/ttyACM0> at <921600>
[ERROR] [launch_ros.actions.lifecycle_node]: Failed to make transition 'TRANSITION_ACTIVATE' for LifecycleNode '/microstrain_inertial_driver'
[microstrain_inertial_driver_node-1] [ERROR] [1679678384.710674980] [microstrain_inertial_driver]: Device Disconnected
[ERROR] [launch_ros.actions.lifecycle_node]: Failed to make transition 'TRANSITION_CONFIGURE' for LifecycleNode '/microstrain_inertial_driver'
[microstrain_inertial_driver_node-1] [ERROR] [1679678384.710684949] [microstrain_inertial_driver]: Failed to read configuration for node
[microstrain_inertial_driver_node-1] [ERROR] [1679678384.710688858] [microstrain_inertial_driver]: Failed to configure node base
[microstrain_inertial_driver_node-1] [WARN] [1679678384.710835338] [rcl_lifecycle]: No transition matching 3 found for current state unconfigured
[microstrain_inertial_driver_node-1] [ERROR] [1679678384.710842826] []: Unable to start transition 3 from current state unconfigured: Transition is not registered., at ./src/rcl_lifecycle.c:355
```

Here is my params.yml for reference:
```
microstrain_inertial_driver:
  ros__parameters:
    port  : "/dev/ttyACM0"

    use_enu_frame : True

    baudrate    : 921600  # max

    device_setup : True

    filter_sensor2vehicle_frame_selector: 1
    filter_vel_in_vehicle_frame : True
    filter_declination_source: 1

    filter_data_rate : 500

    filter_use_compensated_accel : True
```

robbiefish, 2023-03-28:
Hi @bbokser,

The docker containers, and the driver in general has not been tested on WSL or WSL2. Since it looks like the device is not able to connect, I would guess that this problem is somewhere in your environment. Here are a few things to try:

1. Before attaching the device to WSL, can you access the device with [SensorConnect](https://www.parker.com/us/en/divisions/noise-vibration-and-harshness-division/solutions/microstrain-sensors/microstrain-software/sensorconnect.html)?
1. How is the device attached to the computer? Is it via USB, or via serial? If using serial, you either need to have `set_baud` set to true, reduce your baudrate to `115200` or use SensorConnect to set the baudrate to `921600` and save your settings as startup
1. After attaching the device to WSL, can you interact with it outside of docker? If you configure the device to stream via SensorConnect, you should be able to run `cat /dev/ttyACM0` and see a bunch of binary data printed to the screen.

bbokser, 2023-03-29:
Hi @robbiefish,

1. I have accessed the device with SensorConnect and verified that it works.
2. I have configured the baudrate to `921600` in SensorConnect.
3. Running `cat /dev/ttyACM0` does indeed output binary data. `lsusb` and `dmesg | grep tty` also work. I can also verify that the device is connected to wsl2 by unplugging it and trying the same commands (it doesn't show up).
4. I finally tried this on a physical Ubuntu computer and the issue does not occur. However, I'm unable to find the node using `ros2 topic list` or `ros2 node list`. But maybe there's something else I'm doing wrong with ROS2 there. Anyway, I'm not sure why this isn't working on wsl2 when all of the aforementioned tests work.

closed 2023-04-30 by github-actions[bot]

HamRadioDXer, 2023-10-13:
Hello, have you solved your problem? I had the same problem

bbokser, 2023-10-13:
My main problem was due to the use of WSL2, as noted in my last comment. Switching to physical Ubuntu solved the issue for me.