LORD-MicroStrain/microstrain_inertial #247: Can we get GNSS raw measurements of 3DM gq7 like Ublox?

JokerJohn, 2023-04-16:
**Is your feature request related to a problem? Please describe.**
I want to get the GNSS raw measurements. like the UBX-RXM-RAWX(Multi-GNSS-Raw-Measurement Data), UBX-RXM-SFRBX and UBX-NAV-PVT  in Ublox F9p.  since we want to do a tightly-coupled methods with other sensors,such as camera and lidar.

**Describe the solution you'd like**
Multi-GNSS-Raw-Measurement, GNSS velocity(**not the 2 dimension speed from NMEA GPRMC**), and so on.

**Additional context**
Add any other context or screenshots about the feature request here.

nathanmillerparker, 2023-04-17:
Hello,

The device outputs this information in our MIP format, please see these links for reference: https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/external_content/dcp/Data/gnss_recv_1/gnss_recv_1_links.htm
https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/external_content/dcp/Data/gnss_recv_2/gnss_recv_2_links.htm
https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/external_content/dcp/Data/gnss_corrections/gnss_corrections_links.htm

We have not made this information available in our ROS driver yet as an overwhelming majority of our ROS customers use our computed solution.  There are provisions for accessing this information in the MIP SDK, which the driver is built on.  You could modify the driver code to collect and publish this data for your purposes.  If you choose to, please consider creating a pull request so others may benefit from it as well.  I will add a task to our backlog, but I cannot provide an estimate of when this may be complete.

Thank you,

Nathan

JokerJohn, 2023-04-18:
> Hello,
>
> The device outputs this information in our MIP format, please see these links for reference: [...]

Thank you for your patient response. I will try to complete this task and, if successful, I will contribute it.

closed 2023-04-18 by JokerJohn