LORD-MicroStrain/microstrain_inertial #248: IMU report timestamp warning

HomieRegina, 2023-05-25:
When I tried to locate using 3dm-gx5-25 data, the terminal reported an error, as shown below:

Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame sensor at time 168492036.961000 according to authority unknown_publisher 
     at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.6/src/buffer_core.cpp

When we replaced another IMU, we can use them normally without reporting any errors. Is this related to the driver configuration? How to solve this problem? 
Looking forward to a reply! Thank you!

HomieRegina, 2023-05-26:
> We get the same from our gx5-15 (or gx5-AR, depending on which web page you're looking at). Didn't have time to track down the issue, so just commented-out the call to `sendTransform()` in `Publishers::publish()`. I know, it's a hack, not a solution :)

Thank you for your reply and effective methods. I have referred to your method to cancel the release of TF transformation. The above error has been solved. And I found that the IMU data publishing frequency I set is 100Hz. When running on one of my industrial control computers, the TF conversion is only 11 Hz, but this issue will not occur on other industrial control computers. Do you know the reason of this problem?
Looking forward to your reply. Thank you!

rickarmstrong, 2023-05-26:
> Thank you for your reply and effective methods. I have referred to your method to cancel the release of TF transformation. The above error has been solved. And I found that the IMU data publishing [...]

I think you're right about the differing publish rates. It's possible that simply updating the header timestamp before sending the transform might do the trick, something (sorta) like:
```
if (filter_relative_odom_pub_ && filter_relative_odom_pub_->updated())
  filter_relative_transform_msg_.header.stamp = ros::Time::now();
  transform_broadcaster_->sendTransform(filter_relative_transform_msg_);
```
I haven't looked at the transform update code, so I don't know if this is thread-safe. Might require synchronization (e.g. a mutex).

HomieRegina, 2023-05-29:
> > Thank you for your reply and effective methods. I have referred to your method to cancel the release of TF transformation. The above error has been solved. And I found that the IMU data publishing frequency I set is 100Hz. When running on one of my industrial control computers, the TF conversion is only 11 Hz, but this issue will not occur on other industrial control computers. Do you know the reason of this problem? Looking forward to your reply. Thank you!
> 
> I think you're right about the differing publish rates. It's possible that simply updating the header timestamp before sending the transform might do the trick, something (sorta) like:
> 
> ```
> if (filter_relative_odom_pub_ && filter_relative_odom_pub_->updated())
>   filter_relative_transform_msg_.header.stamp = ros::Time::now();
>   transform_broadcaster_->sendTransform(filter_relative_transform_msg_);
> ```
> 
> I haven't looked at the transform update code, so I don't know if this is thread-safe (probably not). Might require synchronization (e.g. a mutex)...thinking about it a little more, perhaps adding an additional check to see if the transform is updated would be better, something like: `if (filter_relative_odom_pub_ && filter_relative_odom_pub_->updated()) && updated(filter_relative_transform_msg_))`

Thanks for your reply and excellent solution. I will try the method you provided.

robbiefish, 2024-03-25:
This problem should be resolved in version `4.0.0` and later of the driver. I am closing this issue now. Please feel free to reopen if you continue to see this problem

closed 2024-03-25 by robbiefish