LORD-MicroStrain/microstrain_inertial #255: Expose Hard and Soft Iron Corrections in Params File

cocharland, 2023-06-14:
**Is your feature request related to a problem? Please describe.**
As a user of a microstrain device (GQ7) that is installed in a hard to reach location, and attached to a non windows computer, having to connect to the sensor over USB to load hard & soft iron is a pain. I can set it through the service calls but it won't persist over power cycles.

**Describe the solution you'd like**
I'd like to be able to specify the matrices from the params file so they can be set by my system at launch.

**Describe alternatives you've considered**
Auto call the services at launch, which is clunky and not extensible to other systems

**Additional context**

nathanmillerparker, 2023-07-06:
Hello, 

You can use Device Settings service to store your hard/soft iron config across power cycles.  We are working on a new cross-platform interface program for our devices that we hope to have out by the end of the year to make this process easier.

Hope that helps!

Nathan

jage92, 2023-09-26:
I calibrated manually in all directions near to the place that the IMU will be placed, and with the World Magnetic Model option I get better results. I will write to support to more personal help, but I wanted to leave this message to help other people. 

When no need to use rotation to frame vehicle it is important to put the parameter in None=0, because with other option without rotation the attitude is relative to the start orientation.

jage92, 2023-09-26:
Hello,

when the IMU is working for some time and the heading is correct. I call the get_hard_iron and get_soft_iron services to obtain the correct matrix to then use this calibration params in other execution, but the matrix is always the configured with Sensor Connect. I understood in your previous message that the service will return the correct data when the IMU make autocalibration. 

I do not know if it is a bug, or I am wrong.