LORD-MicroStrain/microstrain_inertial #257: Cannot set imu_data_rate >100Hz for 3DM-CV7 AHRS

shahyaan, 2023-06-20:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch Node with command `docker run --privileged --net=host -v /home/imu_params.yml:/home/imu_params.yml --device /dev/ttyS0 microstrain/ros-microstrain_inertial_driver:ros params_file:=/home/imu_params.yml port:=/dev/ttyS0
` 
3. See Error
terminate called after throwing an instance of 'mscl::Error_Communication'
what():  Failed to communicate with the device.

**Expected behavior**
When imu_data_rate: 100 in imu_params.yml file, the sensor does indeed publish on topic /imu/data at 100 Hz and the data looks valid (header, orientation, orientation_covariance, angular velocity, angular_velocity_covariance, linear_acceleration, linear_acceleration_covariance). However, the Ros node issues the following warnings:
[ WARN] [1687275568.570441788]: Attempted to stream MIP field with descriptor 0x8214 at a rate of 10.000000 hz, but the device reported that it does not support it                                         
[ WARN] [1687275568.580466708]: Attempted to stream MIP field with descriptor 0x8201 at a rate of 10.000000 hz, but the device reported that it does not support it                                         
[ WARN] [1687275568.580549429]: Attempted to stream MIP field with descriptor 0x8208 at a rate of 10.000000 hz, but the device reported that it does not support it                                         
[ WARN] [1687275568.580593910]: Attempted to stream MIP field with descriptor 0x8202 at a rate of 10.000000 hz, but the device reported that it does not support it                                         
[ WARN] [1687275568.581006222]: Attempted to stream MIP field with descriptor 0x8209 at a rate of 10.000000 hz, but the device reported that it does not support it                                         
[ WARN] [1687275568.592788868]: Attempted to stream MIP field with descriptor 0x8242 at a rate of 10.000000 hz, but the device reported that it does not support it                                         
[ WARN] [1687275568.592860719]: Attempted to stream MIP field with descriptor 0x8208 at a rate of 10.000000 hz, but the device reported that it does not support it                                         
[ WARN] [1687275568.592958866]: Attempted to stream MIP field with descriptor 0x8202 at a rate of 10.000000 hz, but the device reported that it does not support it                                         
[ WARN] [1687275568.593171439]: Attempted to stream MIP field with descriptor 0x8209 at a rate of 10.000000 hz, but the device reported that it does not support it

**Environment (please complete the following information):**
 - OS: Raspbian Buster
 - Architecture: arm32 [e.g. x86_64, arm32, arm64]
 - ROS Version: noetic 
 - Version: 2.0.0
 - Sensor(s): 3DM-CV7

**Modifications**
No modifications to code, simply created a new parameters.yml file for sensor to reference

**Launch Parameters**
If you have an updated `params.yml` file, please attach it to this issue.
imu_params.yml file:
declination_source: 1
imu_data_rate: 250
filter_enable_gnss_pos_vel_aiding: False
filter_enable_gnss_heading_aiding: False

**Additional context**
Add any other context about the problem here.

robbiefish, 2023-06-20:
I think these logs be kind of confusing. There are many topics that your device does not support, which is what those logs are saying,

Can you please run `rostopic hz /imu/data` or replace `/imu/data` with whatever topic you are interested in and let me know the rate in hertz you are getting? If it is around 250 hertz, then your device is functioning as expected

robbiefish, 2023-06-21:
Okay, thank you for the feedback, I notice that you are using an old version of the driver. The version you are using was build before the CV7 was released, so it has not been tested with that device. Could you try updating to the latest version? It should be available via the ROS package index for your environment

Installation instructions for installing from the buildfarm can be found [here](https://github.com/LORD-MicroStrain/microstrain_inertial#buildfarm)

If you can't install from the buildfarm, you should be able to build from source by following [these](https://github.com/LORD-MicroStrain/microstrain_inertial#source) instructions

shahyaan, 2023-06-21:
I am using the Raspbian Buster OS which is armv7. 

When I pull the docker image with tag "latest" I get the following error:
docker pull microstrain/ros-microstrain_inertial_driver:latest
latest: Pulling from microstrain/ros-microstrain_inertial_driver
no matching manifest for linux/arm/v7 in the manifest list entries

So instead I pulled "[latest-noetic-arm32v7] (https://hub.docker.com/layers/microstrain/ros-microstrain_inertial_driver/latest-noetic-arm32v7/images sha256-19a070228899bf9528154bf761a5536a390b62688fbaac977f505ea4d2e6a357?context=explore)
which does work, but results in the issue of not being able to set imu_data_rate>100Hz. 

Is there another driver that you are referring to in the last message that I am unaware of?

Thanks

shahyaan, 2023-06-22:
I was able to build the driver from source. However, now I get the following issue when I run: 
roslaunch microstrain_inertial_driver microstrain.launch

process[microstrain_inertial_driver-2]: started with pid [25009]
[ INFO] [1687444676.505751494]: Running microstrain_inertial_driver version: 3.0.1-3-g527e630
[ INFO] [1687444676.508793506]: Using MIP SDK version: v1.0.0-21-gb1e6278
[ INFO] [1687444676.521890559]: Attempting to open serial port </dev/ttyS0> at <115200>
[ INFO] [1687444676.522569382]: Setting device to idle in order to configure
[ INFO] [1687444676.535056221]: Main Connection Info:
    #######################
    Model Name:       3DMCV7-AHRS
    Serial Number:    6286.155873
    Firmware Version: 1.0.07
    #######################
[ INFO] [1687444676.538108529]: Note: Not opening aux port because RTK dongle enable was not set to true.
[ INFO] [1687444676.564968551]: Note: The device does not support publishing the topic gnss1/aiding_status
[ INFO] [1687444676.565282880]: Note: The device does not support publishing the topic gnss1/antenna_offset_correction
[ INFO] [1687444676.565499839]: Note: The device does not support publishing the topic gnss1/fix
[ INFO] [1687444676.565692522]: Note: The device does not support publishing the topic gnss1/fix_info
[ INFO] [1687444676.565881704]: Note: The device does not support publishing the topic gnss1/odom
[ INFO] [1687444676.566068924]: Note: The device does not support publishing the topic gnss1/rf_error_detection
[ INFO] [1687444676.566279458]: Note: The device does not support publishing the topic gnss1/sbas_info
[ INFO] [1687444676.566468529]: Note: The device does not support publishing the topic gnss1/time_ref
[ INFO] [1687444676.566671230]: Note: The device does not support publishing the topic gnss2/aiding_status
[ INFO] [1687444676.566858635]: Note: The device does not support publishing the topic gnss2/antenna_offset_correction
[ INFO] [1687444676.567045576]: Note: The device does not support publishing the topic gnss2/fix
[ INFO] [1687444676.567230926]: Note: The device does not support publishing the topic gnss2/fix_info
[ INFO] [1687444676.567415960]: Note: The device does not support publishing the topic gnss2/odom
[ INFO] [1687444676.567598328]: Note: The device does not support publishing the topic gnss2/rf_error_detection
[ INFO] [1687444676.567816473]: Note: The device does not support publishing the topic gnss2/sbas_info
[ INFO] [1687444676.567999711]: Note: The device does not support publishing the topic gnss2/time_ref
[ INFO] [1687444676.568193541]: Note: The device does not support publishing the topic nav/aiding_summary
[ INFO] [1687444676.568377983]: Note: The device does not support publishing the topic nav/dual_antenna_status
[ INFO] [1687444676.568571073]: Note: The device does not support publishing the topic nav/heading_state
[ INFO] [1687444676.568768681]: Note: The device does not support publishing the topic rtk/status
[ INFO] [1687444676.568965660]: Note: The device does not support publishing the topic rtk/status_v1
[ INFO] [1687444676.638279756]: Note: Not configuring GPIO
[ INFO] [1687444676.638630399]: Setting PPS source to 0x0001
[ERROR] [1687444676.640628463]: Failed to configure PPS source
[ERROR] [1687444676.640869941]:   Error(3): Invalid Parameter
[ERROR] [1687444676.641085494]: Failed to read configuration for node
[FATAL] [1687444676.641295435]: Unable to configure node base
[microstrain_inertial_driver-2] process has died [pid 25009, exit code 2, cmd /opt/ros/noetic/lib/microstrain_inertial_driver/microstrain_inertial_driver_node __name:=microstrain_inertial_driver __log:=/home/mezmeriz/.ros/log/5eb151ca-110a-11ee-868e-e45f014ed061/microstrain_inertial_driver-2.log].
log file: /home/mezmeriz/.ros/log/5eb151ca-110a-11ee-868e-e45f014ed061/microstrain_inertial_driver-2*.log

shahyaan, 2023-06-22:
Hi Rob,

Setting filter_pps_source: 0 worked. The node runs and data is streaming from the IMU. I think the filter_declination_soure: 1 is correct for the CV7, and not 0. If I set it to 0 then the node stops communicating. 

I am able to get data rates over 100Hz now, however, there still seems to be a limit somewhere that prevents me from achieving an imu_data_rate of 250 Hz. When I change the imu_data_rate: 250 in 'src/microstrain_inertial/microstrain_inertial_driver/microstrain_inertial_driver_common/config/params.yml', here's the output of the node:

started roslaunch server http://CM4-8G-HEAD3:35625/

SUMMARY
========

PARAMETERS
 * /microstrain_inertial_driver/aux_port: /dev/ttyACM1
 * /microstrain_inertial_driver/baudrate: 115200
 * /microstrain_inertial_driver/debug: False
 * /microstrain_inertial_driver/device_setup: True
 * /microstrain_inertial_driver/diagnostics: False
 * /microstrain_inertial_driver/enable_hardware_odometer: False
 * /microstrain_inertial_driver/filter_adaptive_level: 2
 * /microstrain_inertial_driver/filter_adaptive_time_limit_ms: 15000
 * /microstrain_inertial_driver/filter_aiding_measurement_summary_data_rate: -1
 * /microstrain_inertial_driver/filter_aiding_status_data_rate: -1
 * /microstrain_inertial_driver/filter_angular_zupt: True
 * /microstrain_inertial_driver/filter_angular_zupt_topic: /moving_ang
 * /microstrain_inertial_driver/filter_antenna_offset_correction_data_rate: -1
 * /microstrain_inertial_driver/filter_auto_heading_alignment_selector: 1
 * /microstrain_inertial_driver/filter_auto_init: True
 * /microstrain_inertial_driver/filter_child_frame_id: sensor
 * /microstrain_inertial_driver/filter_data_rate: 10
 * /microstrain_inertial_driver/filter_declination: 0.23
 * /microstrain_inertial_driver/filter_declination_source: 1
 * /microstrain_inertial_driver/filter_dynamics_mode: 1
 * /microstrain_inertial_driver/filter_enable_altimeter_aiding: False
 * /microstrain_inertial_driver/filter_enable_external_gps_time_update: False
 * /microstrain_inertial_driver/filter_enable_external_heading_aiding: False
 * /microstrain_inertial_driver/filter_enable_gnss_antenna_cal: True
 * /microstrain_inertial_driver/filter_enable_gnss_heading_aiding: False
 * /microstrain_inertial_driver/filter_enable_gnss_pos_vel_aiding: False
 * /microstrain_inertial_driver/filter_enable_magnetometer_aiding: False
 * /microstrain_inertial_driver/filter_enable_odometer_aiding: False
 * /microstrain_inertial_driver/filter_enable_vertical_gyro_constraint: False
 * /microstrain_inertial_driver/filter_enable_wheeled_vehicle_constraint: False
 * /microstrain_inertial_driver/filter_external_gps_time_topic: /external_gps_time
 * /microstrain_inertial_driver/filter_external_speed_topic: /external_speed
 * /microstrain_inertial_driver/filter_frame_id: sensor_wgs84
 * /microstrain_inertial_driver/filter_gnss_antenna_cal_max_offset: 0.1
 * /microstrain_inertial_driver/filter_gnss_dual_antenna_data_rate: -1
 * /microstrain_inertial_driver/filter_heading_data_rate: -1
 * /microstrain_inertial_driver/filter_heading_source: 1
 * /microstrain_inertial_driver/filter_heading_state_data_rate: -1
 * /microstrain_inertial_driver/filter_imu_data_rate: -1
 * /microstrain_inertial_driver/filter_init_attitude: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_init_condition_src: 0
 * /microstrain_inertial_driver/filter_init_position: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_init_reference_frame: 2
 * /microstrain_inertial_driver/filter_init_velocity: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_lever_arm_offset: [0.0, 0.0, 0.0]
 * /microstrain_inertial_driver/filter_nmea_gga_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_gll_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_hdt_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_prka_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_rmc_data_rate: 0
 * /microstrain_inertial_driver/filter_nmea_talker_id: 3
 * /microstrain_inertial_driver/filter_odom_data_rate: -1
 * /microstrain_inertial_driver/filter_pps_source: 0
 * /microstrain_inertial_driver/filter_relative_odom_data_rate: -1
 * /microstrain_inertial_driver/filter_relative_position_config: False
 * /microstrain_inertial_driver/filter_relative_position_frame: 2
 * /microstrain_inertial_driver/filter_relative_position_ref: [0, 0, 0.01]
 * /microstrain_inertial_driver/filter_relative_position_source: 1
[... 178 lines cut ...]
subscribed to [/imu/data]
average rate: 128.791
        min: 0.006s max: 0.018s std dev: 0.00221s window: 128
average rate: 129.640
        min: 0.006s max: 0.018s std dev: 0.00221s window: 259
average rate: 129.339
        min: 0.004s max: 0.020s std dev: 0.00228s window: 388
average rate: 129.505
        min: 0.004s max: 0.020s std dev: 0.00225s window: 518
average rate: 129.553
        min: 0.004s max: 0.020s std dev: 0.00224s window: 648
average rate: 129.875
        min: 0.004s max: 0.020s std dev: 0.00218s window: 780
average rate: 129.570
        min: 0.004s max: 0.020s std dev: 0.00220s window: 908
average rate: 129.531
        min: 0.004s max: 0.020s std dev: 0.00223s window: 1037
average rate: 129.332
        min: 0.004s max: 0.020s std dev: 0.00225s window: 1165
average rate: 129.347
        min: 0.004s max: 0.020s std dev: 0.00226s window: 1295
average rate: 129.202
        min: 0.004s max: 0.020s std dev: 0.00225s window: 1423
average rate: 129.232
        min: 0.004s max: 0.020s std dev: 0.00225s window: 1523

nathanmillerparker, 2023-06-27:
Looks like it might be a baudrate issue?  

From "[ INFO] [1687450343.429608030]: Attempting to open serial port </dev/ttyS0> at <115200>"

Are you using a RS232 serial port at 115200 baud?  If so, can you up it to 921600?  The driver will output a number of messages from the device to support as many of the features of the driver as possible.  You can tune the publishing rate of data in the driver to some extent to try to maximize your bandwidth.

Hope that helps!

Nathan

shahyaan, 2023-06-27:
Hi Nathan,

I've tried that (and other baudrates) and the node gives me the following error. Seems like when the device tries to reset itself to accept the new configuration it hangs.

NODES
  /
    microstrain_inertial_driver (microstrain_inertial_driver/microstrain_inertial_driver_node)

ROS_MASTER_URI=http://CM4-8G-HEAD3:11311

process[microstrain_inertial_driver-1]: started with pid [1192]
[ INFO] [1687876111.025718907]: Running microstrain_inertial_driver version: 3.0.1-3-g527e630
[ INFO] [1687876111.028669443]: Using MIP SDK version: v1.0.0-21-gb1e6278
[ INFO] [1687876111.041987929]: Attempting to open serial port </dev/ttyS0> at <230400>
[ INFO] [1687876111.042548733]: Setting device to idle in order to configure
[ERROR] [1687876117.048503613]: Unable to set device to idle
[ERROR] [1687876117.048761349]:   Error(-4): Timed out
[ERROR] [1687876117.048905383]: Failed to read configuration for node
[FATAL] [1687876117.049043714]: Unable to configure node base
[microstrain_inertial_driver-1] process has died [pid 1192, exit code 2, cmd /opt/ros/noetic/lib/microstrain_inertial_driver/microstrain_inertial_driver_node __name:=microstrain_inertial_driver __log:=/home/mezmeriz/.ros/log/cb4205ce-14f6-11ee-b463-e45f014ed061/microstrain_inertial_driver-1.log].
log file: /home/mezmeriz/.ros/log/cb4205ce-14f6-11ee-b463-e45f014ed061/microstrain_inertial_driver-1*.log

shahyaan, 2023-06-27:
Here is the same error with 921600 baudrate:
NODES
  /
    microstrain_inertial_driver (microstrain_inertial_driver/microstrain_inertial_driver_node)

ROS_MASTER_URI=http://CM4-8G-HEAD3:11311

process[microstrain_inertial_driver-1]: started with pid [2518]
[ INFO] [1687876351.086200558]: Running microstrain_inertial_driver version: 3.0.1-3-g527e630
[ INFO] [1687876351.089013060]: Using MIP SDK version: v1.0.0-21-gb1e6278
[ INFO] [1687876351.101635418]: Attempting to open serial port </dev/ttyS0> at <921600>
[ INFO] [1687876351.102213040]: Setting device to idle in order to configure
[ERROR] [1687876357.111125789]: Unable to set device to idle
[ERROR] [1687876357.111578154]:   Error(-4): Timed out
[ERROR] [1687876357.112717065]: Failed to read configuration for node
[FATAL] [1687876357.113029117]: Unable to configure node base
[microstrain_inertial_driver-1] process has died [pid 2518, exit code 2, cmd /opt/ros/noetic/lib/microstrain_inertial_driver/microstrain_inertial_driver_node __name:=microstrain_inertial_driver __log:=/home/mezmeriz/.ros/log/cb4205ce-14f6-11ee-b463-e45f014ed061/microstrain_inertial_driver-1.log].
log file: /home/mezmeriz/.ros/log/cb4205ce-14f6-11ee-b463-e45f014ed061/microstrain_inertial_driver-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

nathanmillerparker, 2023-06-27:
Those errors indicate that the port is not opening correctly at those baudrates.  Is there a port configuration option on your target hardware that needs to be set?  We provide low-level serial drivers for standard Linux & Windows installs, but the initialization may look different for your target hardware.  This is where I'd look for reference in how we set up the serial port (our ROS driver uses our MIP SDK): https://github.com/LORD-MicroStrain/mip_sdk/tree/b1e62780dfdfe6a9166480e9b05a8fdc051e6dd9/src/mip/utils

Hope that helps,
Nathan

shahyaan, 2023-06-27:
Hi Nathan,

That resolved the problem, however, the only way I could change the baudrate was using the sensor connect software, disconnecting from the device, and then attaching it to y raspberry Pi module. I was unable to get the device to respect a change in the baud rate using the parameter file. But once it was set in sensor connect to 921600 it stays that way and allows imu_data_rates up to 1Khz, and I can live with that. Thank you both Nathan and Rob for helping me resolve this issue!

Shahyaan

robbiefish, 2023-06-28:
@shahyaan, I am glad Nathan was able to help you with the issue in my absence, saving the baudrate setting using SensorConnect is one solution, you could also change the parameter `set_baud: True` in the `params.yml`, this will configure the device to use the requested baudrate and then open the port at the requested baudrate as well.

closed 2023-08-06 by github-actions[bot]