LORD-MicroStrain/microstrain_inertial #258: Unable to relaunch the node successful after stopping the node with ctrl + C

bennettbe, 2023-06-22:
**Describe the bug**
After plugging in the 3DMGX5-AHRS I open a terminal and successfully run the node from the launch file. I then successfully configured and activated the node. 

The issue starts when I ctrl + C out of the running program.

Without unplugging the 3DMGX5-AHRS from the usb port I then relaunch the node from the terminal. I then cannot configure the node.

I have to unplug the 3DMGX5-AHRS from the usb port and then plug it back in to get it to successfully configure after exiting with ctrl + C from a successful run. I have not been able to execute two consecutive successful runs.

This code is completely unmodified and taken directly from the ros2 branch.

**To Reproduce**
Steps to reproduce the behavior:

1. Unplug and plug back in the 3DMGX5-AHRS

2. Launch Node in terminal 1:
rosdev@ros2:/ws$ ros2 launch microstrain_inertial_driver microstrain_launch.py 
[INFO] [launch]: All log files can be found below /home/rosdev/.ros/log/2023-06-22-22-19-56-285176-ros2-2923
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [microstrain_inertial_driver_node-1]: process started with pid [2935]

3. Configure the node in terminal 2:
rosdev@ros2:/ws$ ros2 lifecycle set /microstrain_inertial_driver configure
Transitioning successful

4. Output in terminal 1 from step 3:
[microstrain_inertial_driver_node-1] [INFO] [1687472402.745780235] [microstrain_inertial_driver]: Running microstrain_inertial_driver version: unknown
[microstrain_inertial_driver_node-1] [INFO] [1687472402.745931487] [microstrain_inertial_driver]: Using MIP SDK version: v0.0.0
[microstrain_inertial_driver_node-1] [INFO] [1687472402.746082239] [microstrain_inertial_driver]: Attempting to open serial port </dev/ttyACM0> at <115200>
[microstrain_inertial_driver_node-1] [INFO] [1687472402.747326799] [microstrain_inertial_driver]: Setting device to idle in order to configure
[microstrain_inertial_driver_node-1] [INFO] [1687472402.764998603] [microstrain_inertial_driver]: Main Connection Info:
[microstrain_inertial_driver_node-1]     #######################
[microstrain_inertial_driver_node-1]     Model Name:       3DMGX5-AHRS
[microstrain_inertial_driver_node-1]     Serial Number:    6253.152431
[microstrain_inertial_driver_node-1]     Firmware Version: 1.1.76
[microstrain_inertial_driver_node-1]     #######################
[microstrain_inertial_driver_node-1] [INFO] [1687472402.765137938] [microstrain_inertial_driver]: Note: Not opening aux port because RTK dongle enable was not set to true.
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776686734] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss1/aiding_status
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776779402] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss1/antenna_offset_correction
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776788444] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss1/fix
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776796069] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss1/fix_info
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776803319] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss1/odom
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776809319] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss1/rf_error_detection
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776815611] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss1/sbas_info
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776822111] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss1/time_ref
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776828986] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss2/aiding_status
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776836278] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss2/antenna_offset_correction
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776843903] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss2/fix
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776850236] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss2/fix_info
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776856403] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss2/odom
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776862820] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss2/rf_error_detection
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776868862] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss2/sbas_info
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776874737] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss2/time_ref
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776898362] [microstrain_inertial_driver]: Note: The device does not support publishing the topic imu/overrange_status
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776909404] [microstrain_inertial_driver]: Note: The device does not support publishing the topic nav/aiding_summary
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776917321] [microstrain_inertial_driver]: Note: The device does not support publishing the topic nav/dual_antenna_status
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776957863] [microstrain_inertial_driver]: Note: The device does not support publishing the topic rtk/status
[microstrain_inertial_driver_node-1] [INFO] [1687472402.776966530] [microstrain_inertial_driver]: Note: The device does not support publishing the topic rtk/status_v1
[microstrain_inertial_driver_node-1] [INFO] [1687472402.824835238] [microstrain_inertial_driver]: Note: The device does not support the GPIO config command
[microstrain_inertial_driver_node-1] [INFO] [1687472402.824884905] [microstrain_inertial_driver]: Note: The device does not support the PPS source command
[microstrain_inertial_driver_node-1] [INFO] [1687472402.824891072] [microstrain_inertial_driver]: Note: The device does not support the odometer settings command
[microstrain_inertial_driver_node-1] [INFO] [1687472402.824896572] [microstrain_inertial_driver]: Note: The device does not support the factory streaming channels setup command
[microstrain_inertial_driver_node-1] [INFO] [1687472402.824940323] [microstrain_inertial_driver]: Note: The device does not support the SBAS settings command
[microstrain_inertial_driver_node-1] [INFO] [1687472402.824947115] [microstrain_inertial_driver]: Note: The device does not support the nmea message format command
[microstrain_inertial_driver_node-1] [INFO] [1687472402.824981615] [microstrain_inertial_driver]: Note: Device does not support the RTK dongle config command
[microstrain_inertial_driver_node-1] [INFO] [1687472402.825138368] [microstrain_inertial_driver]: Setting Declination Source to 2 0.230000
[microstrain_inertial_driver_node-1] [INFO] [1687472402.830698950] [microstrain_inertial_driver]: Note: Device does not support GNSS antenna offsets
[microstrain_inertial_driver_node-1] [INFO] [1687472402.830805326] [microstrain_inertial_driver]: Note: The device does not support the vehicle dynamics mode command.
[microstrain_inertial_driver_node-1] [INFO] [1687472402.830817327] [microstrain_inertial_driver]: Setting heading source to 1
[... 52 lines cut ...]
**Environment (please complete the following information):**\
The code is being run inside a Docker container in a Parallels VM running Ubuntu on a Mac M1.
 - Docker 
 - - OS: Ubuntu 22.04.2 LTS
 - - Architecture: aarch64
 - Parallels VM
 - - OS: Ubuntu 22.04.2 LTS
 - - Architecture: aarch64
 - Mac M1 (host) 
 - - OS: macOS 13.4
 - - Kernel: Darwin 22.5.0
 - - Architecture: arm64
 - ROS Version: humble
 - version: latest
 - Sensor: 3DM-GX5-AHRS
 - microstrain_inertial branch: origin/ros2

**Modifications**
No modifications

**Launch Parameters**
no updates params.yaml

**Additional context**
Add any other context about the problem here.

gseqBE, 2023-07-06:
Could we have an update on this from Microstrain?

gseqBE, 2023-07-11:
Hi Nathan,
Definitely understand there is the VM layer. That could be the issue. 
As a data point: we have joysticks and other USB peripherals connected and they all seem ok.

I''d like to keep this open. Maybe we could provide your team with a docker container to reproduce? We see similar issues from others, but they get closed as stale.

For the moment, the below will work, before CTRL+C
`ros2 lifecycle set /microstrain_inertial_driver shutdown`

The other option is creating a node that manages the IMU's lifecycle, but we would like to avoid that.

Thank you.