LORD-MicroStrain/microstrain_inertial #260: Hardware Trigger Setup for microstrain 3DM-CV7-AHRS

Alex-Beh, 2023-07-11:
Hi, I am trying to configure the hardware triggering application on microstrain 3DM-CV7-AHRS. I am following the tutoral [ynchronizing Data Output with an External System](https://s3.amazonaws.com/files.microstrain.com/CV7+Online/user_manual_content/app_notes/Example%20Synchronized%20Data%20Output.htm). I use packet builder to send all the commands (as shown below).

[image]

However, the first issue I encounter is I cannot see the event 2 in data streaming.
[image]

I want to make sure the configuration is okay first before move to ROS wrapper. But I am unable to stream the event 2 from SensorConnect.

Here is the json config file. Have renamed the extension to txt because I am unable to upload json file here.
[3DM-CV7-AHRS_config.txt](https://github.com/LORD-MicroStrain/microstrain_inertial/files/12012556/3DM-CV7-AHRS_config.txt)

Alex-Beh, 2023-07-12:
@microstrain-sam Thanks for your reply and it make the entire thing more clear to me now. I have also tried to add other data into the action such as [Scaled Gyro (0x80,0x05)](https://s3.amazonaws.com/files.microstrain.com/CV7+Online/external_content/dcp/Data/sensor_data/data/mip_field_sensor_scaled_gyro.htm) but I remember it will not appear in the list of Event 1. May I know is that the case for you also? Or the action only can stream certain type of data from the 0x80 and 0x82?

Besides, does the current ROS integration stream the event data? If not, do you mind to give some hint which or where should I add the relevant code? Will do the PR after testing.

Alex-Beh, 2023-07-14:
> You could do that [here](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/ecbb25bd5909b7887a8825e09ee114bdc0978474/src/utils/mappings/mip_publisher_mapping.cpp#L173) by sending an empty array to this function.

The only way to do this is hardcode in the source code? Or there is a way to configure in the launch file?

I only care about the event data so what I did now is configure the IMU to only output the event data and set the `device_setup` to false in the launch file. Everything works fine and I manage to get the IMU data from hardware triggering. I am intended to use the params.yml to configure what I did in SensorConnect but it seem not everything is supported from the parameter file right now? Only can find the gpio relevant setup

Guoc0529, 2024-02-22:
> > 您可以通过向此函数发送一个空数组来完成[此](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/ecbb25bd5909b7887a8825e09ee114bdc0978474/src/utils/mappings/mip_publisher_mapping.cpp#L173)操作。
> 
> 做到这一点的唯一方法是在源代码中硬编码？或者有没有办法在启动文件中配置？
> 
> 我只关心事件数据，所以我现在所做的是将 IMU 配置为仅输出事件数据，并`device_setup`在启动文件中将其设置为 false。一切工作正常，我设法从硬件触发获取 IMU 数据。我打算使用 params.yml 来配置我在 SensorConnect 中所做的操作，但现在参数文件似乎并不支持所有内容？只能找到gpio相关设置

@Alex-Beh 
Hello, have you set up the 3dmcv7 hardware trigger? I tried setting it up in the Windows workstation, but it doesn't seem to work. My main goal is to configure it in the ROS driver and trigger it through FPGA. Do you have any experience with this?

Alex-Beh, 2024-02-22:
Yes eventually I have successfully setup the hardware trigger for 3dmcv7. Have you followed the steps here? https://s3.amazonaws.com/files.microstrain.com/CV7+Online/user_manual_content/app_notes/Example%20Synchronized%20Data%20Output.htm

Guoc0529, 2024-02-22:
I want to use hardware synchronization in ROS, how do I set up the yaml file? Are there any other settings needed?

> Yes eventually I have successfully setup the hardware trigger for 3dmcv7. Have you followed the steps here? https://s3.amazonaws.com/files.microstrain.com/CV7+Online/user_manual_content/app_notes/Example%20Synchronized%20Data%20Output.htm

I want to use hardware synchronization in ROS, how do I set up the yaml file? Are there any other settings needed?

Alex-Beh, 2024-02-23:
No special file, just follow the steps few times and you will achieve what you want. I can share you my json file once I have access to my sensor next time.

Can I have your contact? would like to have deeper discussion with you.

Guoc0529, 2024-02-23:
> 无需特殊文件，只需按照步骤几次即可实现您想要的。下次我可以访问我的传感器时，我可以与您分享我的 json 文件。
> 
> 能给我你的联系方式吗？想与您进行更深入的讨论。

My email is guoc171@163.com. Thank you very much!

Guoc0529, 2024-02-26:
> My email is [guoc171@163.com](mailto:guoc171@163.com). Thank you very much!

Hello! Can we chat when you're free? Are you too busy, do you think?

Guoc0529, 2024-02-26:
> No special file, just follow the steps few times and you will achieve what you want. I can share you my json file once I have access to my sensor next time.
> 
> Can I have your contact? would like to have deeper discussion with you.

@Alex-Beh
Hello! Can we chat when you're free? Are you too busy, do you think?

Guoc0529, 2024-02-27:
Hello, I have successfully set up event triggering on Windows. When there is a pulse, Event 1 outputs data, and it doesn't output anything when there's no pulse. However, I want to use this in ROS and I'm not sure how to set it up to work the same way as in Windows. Alternatively, how can I use the data from Event 1?

Guoc0529, 2024-02-27:
> Hi @Alex-Beh, the driver does not support event data at all right now.
> 
> When event data is streamed it will be output like normal data, so the logic in [`publishers.cpp`](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/ecbb25bd5909b7887a8825e09ee114bdc0978474/src/publishers.cpp) should handle parsing and populating the ROS messages just fine, the issue is that it will be interspersed with non event data.
> 
> If you only care about getting event data, this could be fixed by simply setting an empty message format. You could do that [here](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/ecbb25bd5909b7887a8825e09ee114bdc0978474/src/utils/mappings/mip_publisher_mapping.cpp#L173) by sending an empty array to this function. In this case, we will still configure all the ROS publishers, but we will only publish a ROS message when we get event data as that is the only kind of data the driver will receive from the device.
> 
> If you do care about both event data and non event data, that is a much more complicated issue. You would need to set up new publishers for each event, and depending on the [event source](https://s3.amazonaws.com/files.microstrain.com/CV7+Online/external_content/dcp/Data/shared_data/data/mip_field_shared_event_source.htm?Highlight=event%20source) populate the appropriate message.
> 
> In regards to the data not showing up in the list of Event 1, @microstrain-sam and I tested this and saw similar behavior when we first configured the event in SensorConnect, but after reconnecting the device (via the SensorConnect disconnect and reconnect button), the data started showing up in SensorConnect

Hello, I have successfully set up event triggering on Windows. When there is a pulse, Event 1 outputs data, and it doesn't output anything when there's no pulse. However, I want to use this in ROS and I'm not sure how to set it up to work the same way as in Windows. Alternatively, how can I use the data from Event 1?

closed 2026-10-01 by aidan-laracy-HBK