LORD-MicroStrain/microstrain_inertial #268: Is there a driver for ROS2 Foxy for the 3DMCV7-AHRS?

SmnGdn, 2023-07-31:
Hello,

It's all in the title, I build from source with the ROS2 branch, started the driver and then the window hangs at the state described below:

ros2 launch microstrain_inertial_driver microstrain_launch.py 
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-07-31-18-39-22-732665-ubuntu-PX60-6QE-13321
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [microstrain_inertial_driver_node-1]: process started with pid [13333]
[microstrain_inertial_driver_node-1] [INFO] [1690817962.982359653] [microstrain_inertial_driver]: This version of ROS2 does not support changing the log level in C++

There is only the topic /microstrain_inertial_driver/transition_event which is published. I suppose after connecting the sensor and starting the driver node I should expect also topics like /imu/data and other data? 

Thanks in advance!

SmnGdn, 2023-08-01:
When I sniff the port /dev/ttyACM0 (where the device is communicating), with the command `jpnevulator --ascii --timing-print --tty /dev/ttyACM0 --read` I can see data going through the port, approximately once every 30 seconds, and I verified with sensorConnect t(on a Windows system) that the device can send data,and it works there. Just when I plug it and try to read data through ROS2, nothing happens... Yet the green light blinking very fast on the device, so it is communicating. Could there be some information about how to get the device to publish data to ROS? I followed the steps in the Readme but either they are not sufficient, or there is something wrong with the branch ros2 or with how I use it.

robbiefish, 2023-08-02:
Hi @SmnGdn, the ROS2 driver is built as a lifecycle node, so when it starts, it is not configured or activated. If you want to run right away, you can add `configure:=true` and `activate:=true` to your launch command, otherwise you can transition the lifecycle node after it is started. We have some documentation talking about that [here](https://github.com/LORD-MicroStrain/microstrain_inertial/tree/ros2#lifecycle-node)

closed 2023-09-03 by github-actions[bot]