LORD-MicroStrain/microstrain_inertial #269: Bloom failed for 3.1.0 release

stephen-cpr, 2023-08-08:
https://github.com/LORD-MicroStrain/microstrain_inertial/actions/runs/5660073866/job/15334986376

robbiefish, 2023-08-08:
Hi @stephen-cpr,

That github action really needs to be removed. It was an attempt by me to automate part of the release process, but stopped working shortly after adding it. Is there a specific concern related to that failed job?

I manually released using bloom back when I tagged 3.1.0, and it should be available for [noetic](https://index.ros.org/p/microstrain_inertial_driver/github-LORD-MicroStrain-microstrain_inertial/#noetic), [humble](https://index.ros.org/p/microstrain_inertial_driver/github-LORD-MicroStrain-microstrain_inertial/#humble), [iron](https://index.ros.org/p/microstrain_inertial_driver/github-LORD-MicroStrain-microstrain_inertial/#iron), and [rolling](https://index.ros.org/p/microstrain_inertial_driver/github-LORD-MicroStrain-microstrain_inertial/#rolling)

stephen-cpr, 2023-08-08:
Hey Robbie, main concern was just getting 3.1.0. I had tried apt installing ros-noetic-microstrain-inertial-driver after the initial merge and was still getting 3.0, but just tested now and got 3.1.0 okay. Probably should have checked again before opening the ticket -- sorry!