LORD-MicroStrain/microstrain_inertial #271: Subscribing to RTCM corrections over main-port using NTRIP-Client

Acerox41, 2023-08-22:
**Describe the bug**
I'm using dual antenna gnss system with the 3DM-GQ7. I want to use RTK and subscribe to rtcm corrections coming from the [ntrip client](https://github.com/LORD-MicroStrain/ntrip_client): Setting `rtk_dongle_enable : True` and `subscribe_rtcm : True`.

To get this work I need also to publish NMEA sentence messages to the ntrip client: `publish_nmea : True`
Problem: After the first `roslaunch` driver want to open aux-port and returns an error (obviously because nothing is connected).
Pressing `Ctrl-C` and again using `roslaunch`, the driver attempts to open the main-port and fails after iterating through all baud-rates. Then I need to replug the sensor otherwise I get no connection anymore.

Note: The driver works like charm with the current `my_params_ros.yml` file for different baudrates - without  setting rtcm, and nmea parameters mentioned above.

**To Reproduce**
Steps to reproduce the behavior:
1. roslaunch microstrain_inertial_driver microstrain.launch params_file:="/home/me/my_params_ros.yml"
2. See Error:

  [ INFO] [1692691009.480278759]: Running microstrain_inertial_driver version: 3.1.0
  [ INFO] [1692691009.481124144]: Using MIP SDK version: v1.0.0-42-gc332b1a
  [ INFO] [1692691009.484580117]: Attempting to open serial port </dev/ttyACM0> at <921600>
  [ INFO] [1692691009.487616214]: Setting device to idle in order to configure
  [ INFO] [1692691009.487860696]: Note: Setting UART baudrate to 921600
  [ INFO] [1692691009.518224636]: Main Connection Info:
      #######################
      Model Name:       3DM-GQ7
      Serial Number:    6284.154117
      Firmware Version: 1.1.02
      #######################
  [ INFO] [1692691009.522674156]: Attempting to open serial port </dev/ttyACM1> at <921600>
  
  [ERROR] [1692691009.522770509]: MIP SDK: Unable to open port (2): No such file or directory
  [ERROR] [1692691009.522816173]: Failed to read configuration for node
  [FATAL] [1692691009.522840334]: Unable to configure node base
  [microstrain_inertial_driver-2] process has died [pid 16895, exit code 2, cmd /home/me/ros_ws/devel/lib/microstrain_inertial_driver/microstrain_inertial_driver_node __name:=microstrain_inertial_driver __log:=/home/me/.ros/log/72541e2e-40c1-11ee-bbe7-48b02de0c823/microstrain_inertial_driver-2.log].
  log file: /home/me/.ros/log/72541e2e-40c1-11ee-bbe7-48b02de0c823/microstrain_inertial_driver-2*.log

**Expected behavior**
Subscribing to `/rtcm`topic over the main-port.

**Environment (please complete the following information):**
 - Hardware system: Nvidia Jetson Orin DevKit
 - OS: Ubuntu 20.04.6 LTS
 - Architecture: arm64
 - ROS Version: noetic 1.16.0
 - microstrain_inertial_driver version: 3.1.0
 - MIP SDK version: v1.0.0-42-gc332b1a
 - Sensor: 3DM-GQ7

**Modifications**
No modifications in source code.

**Launch Parameters**
"my_params_ros.yml" as zip attached
[my_params_ros.zip](https://github.com/LORD-MicroStrain/microstrain_inertial/files/12406468/my_params_ros.zip)

**Additional context**
I'm also using  the ntrip_client node that is succesfull connected.

nathanmillerparker, 2023-08-22:
Please note, the GQ7 can only receive RTCM corrections over the Aux port.

Hope that helps,

Nathan

Acerox41, 2023-08-22:
Thanks Nathan, so in my setup I just need to start `ntrip_client` and `microstrain_inertial_driver` with another usb cable to the aux-port?

Thanks in advance

Acerox41, 2023-08-22:
> Thanks Nathan, so in my setup I just need to start `ntrip_client` and `microstrain_inertial_driver` with another usb cable to the aux-port?
> 
> Thanks in advance

I just tested it. I works. Thank you very much Nathan.

closed 2023-08-22 by Acerox41