LORD-MicroStrain/microstrain_inertial #272: Difficulty in receiving the correct position and orientation

ojindal, 2023-08-28:
**Describe the bug**

Hi there,

I am using the Parker 3DMGQ7 device along with a Reach Emlid M2 as a base station for RTK correction support. I am using the RFD900x radios for their mutual communication. I am able to receive RTCM messages and the Parker sensor is showing the 'fix' state on both gnss1 and gnss2.

The issue is that when I subscribe to the topic: nav/relative_pos/odom, I am supposed to get the position and orientation in the NED frame which is not happening in my case. When I place the X direction of the sensor in the North, it doesn't show a near 0 yaw. Also, I tried to crosscheck the issue by echoing the message on the topic: nav/heading which showed that the heading/yaw is drifting continuously (keeps incrementing) even when the sensor setup is stationary.

**Environment:**
 - OS: Ubuntu 20.04
 - ROS Version: noetic
 - Sensor(s): 3DM-GQ7

**Launch Parameters**
The `params.yml` file is not updated, used the default one.

nathanmillerparker, 2023-08-28:
Hello,

For the sensor to get a heading fix, you will need to edit the antenna offsets in your params file to match what they are in your setup.  Please reference the following for how to measure these:

https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/user_manual_content/getting_started/Getting%20Started.htm
https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/user_manual_content/installation/Antenna.htm

Please monitor the filter status topic as well and make sure the filter enters the "Full Navigation" mode.

Hope that helps!

Nathan

ojindal, 2023-08-28:
Hi, Thanks for the reply.

I already made appropriate configurations for the same on SensorConnect, will it automatically fetch the data or should I make changes in the params file separately?

Best,
Orish

nathanmillerparker, 2023-08-28:
The default params file has the following flag set, which overrides any of the device's saved settings:

device_setup : True

You would need to set this to "False" to bypass the ROS driver's setup of the device.  Be warned, that certain features may not work if the setup you have in SensorConnect does not provide the equivalent of what ROS expects.  We consider this as an advanced option and should only be used is special cases... overriding the params for ROS is the preferred method of setup when using ROS.

Hope that helps!

Nathan

ojindal, 2023-08-28:
I changed the params.yml according to my requirements, please refer to the file at [https://github.com/ojindal/params.yml/blob/main/params.yml](https://github.com/ojindal/params.yml/blob/main/params.yml). I am currently using dual antenna for the filter heading. 

When I echo the messages on the topic /nav/aiding_summary, it shows:
dual_antenna: 
  enabled: False
  used: False
  
Please guide me further. Thanks

nathanmillerparker, 2023-08-28:
Hello,

Is the system outside, with both antennas pointed towards the sky with unobstructed views?  Also, if you are near a building, you may be experiencing multipath, which may prevent a dual antenna heading fix.  

Assuming the above is okay, we would need either pictures or a diagram of how the system is mounted to check for any errors in setup.  If you would like to provide this privately, you can email me at nathan.miller(at)parker.com.

Hope that helps,

Nathan

ojindal, 2023-08-28:
Hi Nathan,

The test setup is indeed outside with mostly unobstructed views. It seems that we don't have much trouble getting both gnss 1 & 2 into a fix state. Interestingly the dual antenna status topic and the /nav/heading topic are reporting different headings (1.22 rad & 0.627 rad)

Here is the fix status of gnss 1:

[image]

Here is the fix status of gnss 2:

[image]

Here is the output of the dual antenna status msg:

[image]

Here is the output of /nav/heading topic:

[image]

Here is the output of /nav/aiding_summary:

[image]

nathanmillerparker, 2023-08-28:
Interesting... looks like the dual antenna has a valid solution as well.  You can see at the top of the /nav/aiding_summary that the gps_tow is 0, meaning no aiding has been processed in the filter yet (implying it is not in full nav mode.) 

Could you provide the output of the /nav/status topic?

Thanks,

Nathan

ojindal, 2023-08-28:
This the output of the /nav/status topic

```
filter_state: 2
dynamics_mode: 1
status_flags: 1
```

nathanmillerparker, 2023-08-28:
Ah, ok, sorry, I should have saw this earlier...

Your heading alignment selector is set as follows:
filter_auto_heading_alignment_selector : 0

But this is a bitfield, so if you want to enable dual antenna, you need to set bit 1, which is a value of 1:

filter_auto_heading_alignment_selector : 1

Please give that a go and let me know if the filter initializes.  

Nathan

ojindal, 2023-08-28:
This is the output of the /nav/status topic now:

```
filter_state: 2
dynamics_mode: 1
status_flags: 2
```

and this is the output of /nav/aiding_summary

```
gps_tow: 0.0
gnss1: 
  enabled: True
  used: True
  residual_high_warning: False
  sample_time_warning: False
  configuration_error: False
  max_num_meas_exceeded: False
gnss2: 
  enabled: True
  used: False
  residual_high_warning: False
  sample_time_warning: False
  configuration_error: False
  max_num_meas_exceeded: False
dual_antenna: 
  enabled: True
  used: True
  residual_high_warning: False
  sample_time_warning: False
  configuration_error: False
  max_num_meas_exceeded: False
heading: 
  enabled: False
  used: False
  residual_high_warning: False
  sample_time_warning: False
  configuration_error: False
  max_num_meas_exceeded: False
pressure: 
  enabled: False
  used: False
  residual_high_warning: False
  sample_time_warning: False
  configuration_error: False
  max_num_meas_exceeded: False
magnetometer: 
  enabled: False
  used: False
  residual_high_warning: False
  sample_time_warning: False
  configuration_error: False
  max_num_meas_exceeded: False
speed: 
  enabled: False
  used: False
  residual_high_warning: False
  sample_time_warning: False
  configuration_error: False
  max_num_meas_exceeded: False

```

ojindal, 2023-08-28:
But the heading shown by /nav/heading is still icreasing continuously while the sensor setup is stationary.

nathanmillerparker, 2023-08-28:
Another way we can debug this is to output a raw file that we can process on our end.  If you change the settings below to the values listed, the driver will generate a raw data file.  You can send this to the email I provided above and I can take a look at it.  Please record at least 5 minutes of data.  After recording this test data, you will probably want to disable it for your testing by setting raw_file_enable : **False**

# Controls if the driver creates a raw binary file
#      false - Do not create the file
#      true  - Create the file
#
#      Notes: 1) The filename will have the following format -
#                model_number "_" serial_number "_" datetime (year_month_day_hour_minute_sec) ".bin"
#                example: "3DM-GX5-45_6251.00001_20_12_01_01_01_01.bin"
#             2) This file is useful for getting support from the manufacturer
raw_file_enable : **True**

# (GQ7/CV7 only) Controls if the driver requests additional factory support data to be included in the raw data file
#      false - Do not request the additional data
#      true  - Request the additional channels (please see notes below!)
#
#      Notes: **We recommend only enabling this feature when specifically requested by Microstrain.**
#      
#      Including this feature increases communication bandwidth requirements significantly... for serial data connections
#      please ensure the baudrate is sufficient for the added data channels.     
raw_file_include_support_data : **True**

# The directory to store the raw data file (no trailing '/')
raw_file_directory : "/home/your_name" **<- Change this to a directory on your computer**

ojindal, 2023-08-28:
Hi Nathan, 
Data sent.
Seems like the device is in Full Nav mode most of the time when testing, although I have seen instances where this drops. Currently our system is stationary so there should be no major vibrations.

Thanks!

closed 2024-03-24 by github-actions[bot]