LORD-MicroStrain/microstrain_inertial #273: 3dm-cv7 - failed to configure PPS source

hannahloygecko, 2023-08-29:
Hi, I am using 3dm-cv7 with ROS2 foxy on ubuntu 20.04.  When I launch the node with the command 'ros2 launch microstrain_inertial_driver microstrain_launch.py configure:=true', I get the following error('Failed to configure PPS source'), preventing the node from configuring:

[image]

microstrain-build, 2023-08-29:
This appears to be the same issue as #168, see [this comment](https://github.com/LORD-MicroStrain/microstrain_inertial/issues/168#issuecomment-1192737248) specifically

hannahloygecko, 2023-08-29:
I tried changing filter_declination_source to both 1 and 3, and I also changed filter_heading_source to 0 since I am using an -AR product.  I am still getting the same error.

robbiefish, 2023-08-29:
Hi @hannahloygecko,

Apologies, I mistook which option was causing the issue, try setting `filter_pps_source` to 0.
For more information on the possible options for that command, see the `Source` section of [this page](https://s3.amazonaws.com/files.microstrain.com/CV7+Online/external_content/dcp/Commands/3dm_command/data/mip_cmd_3dm_pps_source.htm?Highlight=pps)

hannahloygecko, 2023-08-29:
That worked, thank you!

closed 2023-10-01 by github-actions[bot]