LORD-MicroStrain/microstrain_inertial #274: INS level arm related to ground car rear axle

ayoubasritii, 2023-08-31:
**Describe the bug**
I'm currently using your GQ7 (INS) on a buggy project and have encountered some issues that I'm hoping you could help me resolve.

Vehicle Reference Point Offset: Our INS is installed on a buggy. The system is shifted 8cm from the X axis and positioned 1 meter in front of the rear axle, where the lateral velocity is zero. I have activated the automotive model and Wheeled Vehicle Constraint in the system. However, I am unsure how to configure the INS to account for the 1-meter distance from the rear axle (very important for the ackerman model). I suspect that this involves setting the Vehicle Reference Point Offset, but I am not certain. Could you confirm if this is the case, and if so, provide the syntax for the relevant command to be added to the microstrain.yaml file?

Issue with INS Solution: I have attempted some tests without setting the Vehicle Reference Point Offset. While the antenna positions are accurate, I noticed some discrepancies in the INS solution, especially during turns. I've attached a bag file, the microstrain.yaml file, and the configuration exported using SensorConnect for your reference.
[Inquiry on INS GQ7 Configuration.zip](https://github.com/LORD-MicroStrain/microstrain_inertial/files/12482841/Inquiry.on.INS.GQ7.Configuration.zip)

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.6 LTS
 - Architecture: x86_64
 - ROS Version: noetic
 - Firmware Version: 1.0.10
 - Sensor(s): 3DM-GQ7

**Modifications**
No

**Launch Parameters**
Attached

**Additional context**

robbiefish, 2023-09-08:
Hi @ayoubasritii,

I apologize for not getting back to you sooner. The reason you would not have seen a difference after configuring the reference lever arm with SensorConnect is because the ROS driver will override that setting when we configure the device.

If you set [this line](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/3d3d678c0e2562151464d741094c75fd37705bf0/config/params.yml#L290) in your `microstrain.yaml` to your desired lever arm offset, that should give you the results you expect

ayoubasritii, 2023-09-11:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

Hello Rob,

Thanks for your answer, I will test it and let you know.
Do you think setting the reference to the rear axle is enough to tell the INS that is where the lateral velocity is null or I need to mention it by another mean?

Sincerely,
Ayoub.

closed 2023-10-22 by github-actions[bot]