LORD-MicroStrain/microstrain_inertial #280: 3DMGX5-AHRS Time sync with external GPS

jangirrishabh, 2023-10-26:
Hello, I am trying to sync a VLP 16 lidar and the AHRS imu with an external GPS clock. Hoping I could get some information about doing it here.

The AHRS module does not have an internal GPS PPS clock like the GNSS module, but is it possible to sync it with an external GPS clock?  If yes, could you please explain how to do it.

nathanmillermicrostrain, 2023-10-27:
Hello,

This can be done on a GX5... it requires one of our craft cables, providing the hardware PPS to the device (TTL-level), and providing the time offset via a command (within 0.25 sec of the PPS pulse). 

In ROS, the offset command is listened for and configured on this topic (obviously needs to be enabled as well):

https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/3d3d678c0e2562151464d741094c75fd37705bf0/config/params.yml#L275

https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/3d3d678c0e2562151464d741094c75fd37705bf0/src/subscribers.cpp#L148

Not trying to up-sell you, but if you can, I would suggest that you look into our 7-series devices, as we designed them specifically for applications like this, with many advanced timing and event features.  You can find links to them below.

https://www.microstrain.com/inertial-sensors/3dm-cv7-ahrs
https://www.microstrain.com/inertial-sensors/3DM-CV7-INS (same as above, but you can provide GNSS inputs to it to get pos & vel)
https://www.microstrain.com/inertial-sensors/3DM-GV7-AHRS

This page of the CV7 manual covers this specific topic:

https://s3.amazonaws.com/files.microstrain.com/CV7+Online/user_manual_content/additional_features/PPS.htm

Hope that helps,

Nathan

jangirrishabh, 2023-10-27:
Thank you for the quick response.

I appreciate your recommendation for the CV7 devices. For future purchases, we will look into that. In that case, could you suggest a device that then already has an internal GPS as well such that it can generate its own PPS clock that can then also be used to sync the VLP 16 sensor? 

Just to be clear, we want a system that does not need any other devices other than the PC, VLP 16 sensor, and the IMU. In ideal scenario, either the PC /IMU generates the PPS signal and the lidar+IMU/lidar sync itself to the signal. 

So, either the IMU has to be able to generate its own signal or subscribe to a signal from PC or GPS. But it seems to me there is not a way in which the current IMU I have (AHRS) can get PPS signal from the GPS device without needing TTL-level manipulation.

In case you recommend an IMU that has a GPS clock for PPS signal, it would be great to have a recommendation that can directly connect to the lidar or a PC connected to the lidar and send the clock signal.

This would be super helpful for a lot of folks in the community IMHO. Thank you!

nathanmillermicrostrain, 2023-10-27:
Hello,

It is a little difficult to decipher exactly what you may be asking for, but if you'd like a module that has it's own GNSS receiver and will output IMU data that is sync'd to GPS time, we have a few options, I have listed them below.  All of these will require you to mount a GNSS antenna and have a clear view of the sky.  If you need the actual PPS hardware signal, it is available from all of them at a TTL level, which currently requires our craft cable (flying leads.)  In the near future we will have a cable with 2 TNC connectors for the I/O, but it will still be TTL.  It is unclear to me how you intend to sync one of our IMUs with the Lidar without a hardware PPS line.  Also, the GQ7 is the only device with built-in GNSS that outputs NMEA, which can be used to sync with the Velodyne.

On a side note, the CV7 can generate a PPS, but with less accuracy than GPS.  (The GQ7 can do this too, but is more expensive.)

Hope that helps,

Nathan

https://www.microstrain.com/inertial-sensors/3dm-gx5-45 (On-board kalman filter)
https://www.microstrain.com/inertial-sensors/3dm-gx5-35 (No on-board kalman filter)
https://www.microstrain.com/inertial-sensors/3dm-gq7 (On-board kalman filter, dual antenna heading, and much better IMU than the GX5.)

jangirrishabh, 2023-10-30:
I see, yeah its difficult for me to correctly lay forth my requirements as its a little difficult to wrap my head around all the different versions and signals. Thank you for your patience.

I guess it would be better to just go back to my original question. If you could just expand on this more, and let me know which craft cables and GNSS module do you recommend to get GX5-AHRS working in sync with a VLP 16 lidar.

> This can be done on a GX5... it requires one of our craft cables, providing the hardware PPS to the device (TTL-level), and providing the time offset via a command (within 0.25 sec of the PPS pulse).

Any blogs/resources that do the same, and maybe show a connection diagram would be extremely helpful!

closed 2024-03-24 by github-actions[bot]