LORD-MicroStrain/microstrain_inertial #297: Unexpected heading acquired during navigation using AHRS GX5

FLaiRobotics, 2024-02-20:
We are facing an issue while using the orientation data for inferring the absolute heading of an autonomous ground vehicle from an 3DM-GX5-AHRS IMU. As shown in the image below, when navigating concentric square paths we see odometry alternating acute and obtuse angles where there appears to be a missing transformation or incorrect parameter configuration causing unexpected heading calculations at the turns.

[image]

Raw magnetometer data from _/mag_ and raw orientation datum from IMU data from _/imu/data/orientation_ gave similar results. We have calibrated the IMU (also hard and soft iron) using SensorConnect.

When tested with a different IMU, we were able to track a square shaped path as can be seen in the image below.

[image]

Environment:
- Ubuntu 20.04 LTS
- Architecture: x86_64
- ROS Version: Noetic
- Sensor: 3DM-GX5-AHRS

Appreciate your support investigating this issue.

Best,
Debjit Ghosh

> Listed below are  the contents of params.xml:

```json
# Standalone example params file for GX3, GX4, GX/CX5, RQ1 and GQ7 series devices
# Note: Feature support is device-dependent and some of the following settings may have no affect on your device.
# Please consult your device's documentation for supported features

# ****************************************************************** 
# NOTE: This file is formatted to work with ROS and will not work if specified as the params_file argument in ROS2.
#       If you want to override parameters for ROS2, start with https://github.com/LORD-MicroStrain/microstrain_inertial/blob/ros2/microstrain_inertial_driver/config/empty.yml
# ****************************************************************** 

# ****************************************************************** 
# General Settings 
# ****************************************************************** 

# port is the main port that the device will communicate over. For all devices except the GQ7, this is the only available port.
# aux_port is only available for the GQ7 and is only needed when streaming RTCM corrections to the device from ROS, or if you want to publish NMEA sentences from this node
port        : "/dev/ttyUSB0"
aux_port    : "/dev/ttyACM1"
baudrate    : 115200
debug       : False
diagnostics : False

# If set to true, this will configure the requested baudrate on the device.
# Note that this will be set on both USB and serial, but will only actually affect the baudrate of a serial connection.
set_baud : False

# Frame IDs used in the different messages. By default these are set to arbitrary strings as not to interfere with other ROS services.
# For more information on common frame IDs, check out: https://www.ros.org/reps/rep-0105.html
#
# filter_frame_id and filter_child_frame_id are specifically useful as the node will also publish a transform to the /tf topic
# that contains the transform between these two frames. Many ROS tools such as RViz will use the /tf topic to display things like robot position.
imu_frame_id          : "robot_footprint"
gnss1_frame_id        : "gnss1_antenna_wgs84"
gnss2_frame_id        : "gnss2_antenna_wgs84"
filter_frame_id       : "robot_footprint"
filter_child_frame_id : "robot_footprint"
nmea_frame_id         : "nmea"

# Waits for a configurable amount of time until the device exists
# If poll_max_tries is set to -1 we will poll forever until the device exists
poll_port      : False
poll_rate_hz   : 1.0
poll_max_tries : 60

# Number of times to attempt to reconnect to the device if it disconnects while running
# If configure_after_reconnect is true, we will also reconfigure the device after we reconnect
#      NOTE: It is strongly recommended to configure after reconnect unless device_setup is set to false
[... 435 lines cut ...]
gnss2_nmea_gsv_data_rate: 0  # Note that this message_id will not use the gnss1_talker_id since the talker ID will come from the actual constellation the message originates from
gnss2_nmea_rmc_data_rate: 0
gnss2_nmea_vtg_data_rate: 0
gnss2_nmea_hdt_data_rate: 0
gnss2_nmea_zda_data_rate: 0

# NMEA messages in the filter descriptor set
# In order to enable a message, set nmea_message_config to true, and then change the 'data_rate' of the sentences you want to the desired rate in hertz
#
# Note: In order to publish, 'publish_nmea' must also be set to True
#
# Note: filter_nmea_talker_id can be any of the follwing numeric values:
#         1 - Sentences will start with GN
#         2 - Sentences will start with GP
#         3 - Sentences will start with GA
#         4 - Sentences will start with GL
#       The purpose of the talker ID is to differentiate when the same message_id comes from different data sources (descriptor sets)
#       The filter_nmea_talker_id will be applied to all NMEA messages from the filter descriptor set
filter_nmea_talker_id: 3
filter_nmea_gga_data_rate: 0
filter_nmea_gll_data_rate: 0
filter_nmea_rmc_data_rate: 0
filter_nmea_hdt_data_rate: 0
filter_nmea_prka_data_rate: 0  # Note that this message_id will not have any talker ID on it since it is proprietary and can only come from the filter descriptor set
```

robbiefish, 2024-02-22:
Hi @FLaiRobotics,

A couple things to note:
1. The hard and soft iron calibration needs to be done with the unit mounted on the robot. If you did the calibration in your hand or another way please redo it while mounted on the robot
2.  Dynamic magnetic interference is common in mobile robots due to the proximity of electric motors and other sources.  A good way to check interference is to have the sensor mounted on the robot with the robot off.  Record the magnetometer output. Turn the robot on and while preventing it from moving, spin up the motors or other potential interference sources (gimbals, lidars, etc.)  Stop the recording and plot the magnetometer output... did it change drastically?  If so, the unit should be moved to a place on the robot that minimizes interference.

FLaiRobotics, 2024-02-22:
Hi @robbiefish,

Thanks for your response. 

Yes, we did perform soft and hard calibration with the sensor mounted on the robot.

Given the consistent pattern at each turn, could it be possible that the magnetometer/on-board sensors may be damaged or that the sensor is not using the stored calibrations (or) I am not configuring the params correctly?

Best,
Debjit Ghosh

robbiefish, 2024-02-22:
The parameters appear to be correct as far as I can tell.

It is always possible that the sensor is damaged in some way, but are you able to attempt number 2 from my comment above? I would like to rule out external magnetic interference first

FLaiRobotics, 2024-02-26:
Hi @robbiefish,

No drastic change in values found: +/- 0.01 

See comparison of x/y/z raw magnetometer readings; blue/green motors, etc. turned off/on:

[image]

Best,
Debjit Ghosh

closed 2024-03-31 by github-actions[bot]