LORD-MicroStrain/microstrain_inertial #303: Dual antenna system: Receiving heading estimate?

Acerox41, 2024-03-04:
I want to get the heading estimate using the 3DM-GQ7 with two multiband-antennas.
I recognized a small difference of the estimated heading/yaw between the topics `ekf/dual_antenna_heading` and `ekf/odometry_map`.
What is the difference?

Using SensorConnect-Software I get the same difference between the MIP-Messages `GNSS Dual Antenna Status (0x82,0x49)` and `Euler Angles (0x82,0x05)`.

robbiefish, 2024-03-06:
Hi @Acerox41,

Can you checkout the `microstrain_inertial_driver_common` submodule to [`bugfix/dual_antenna_heading`](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/tree/bugfix/dual_antenna_heading) and let me know if that fixes your problem?

If you are in a terminal in the root of `microstrain_inertial`, you can do this by running the following commands
```bash
cd microstrain_inertial_driver/microstrain_inertial_driver_common
git checkout bugfix/dual_antenna_heading
```

robbiefish, 2024-04-03:
This should be fixed on `4.0.1`. Closing for now, feel free to reopen if this is still an issue

closed 2024-04-03 by robbiefish