LORD-MicroStrain/microstrain_inertial #306: ROS IMU data topics not showing

Zahin-Nambiar, 2024-03-08:
**Describe the bug**I have a MicroStrain 3DM-GV7-AHRS unit confirmed connected to a linux machine Ubuntu 22 ROS2 IRON "/dev/ttyACM0". I am trying to get the IMU topics publishing using "ros2 launch microstrain_inertial_driver microstrain_launch.py" I have the params file modified, see below. When I launch, only the "/microstrain_inertial_driver/transition_event" topic shows. 

# Standalone example params file for GX3, GX4, GX/CX5, RQ1 and GQ7 series devices
# Note: Feature support is device-dependent and some of the following settings may have no affect on your device.
# Please consult your device's documentation for supported features

# ****************************************************************** 
# NOTE: This file is formatted to work with ROS and will not work if specified as the params_file argument in ROS2.
#       If you want to override parameters for ROS2, start with https://github.com/LORD-MicroStrain/microstrain_inertial/blob/ros2/microstrain_inertial_driver/config/empty.yml
# ****************************************************************** 

# ****************************************************************** 
# General Settings 
# ****************************************************************** 

# port is the main port that the device will communicate over. For all devices except the GQ7, this is the only available port.
# aux_port is only available for the GQ7 and is only needed when streaming RTCM corrections to the device from ROS, or if you want to publish NMEA sentences from this node
port        : "/dev/ttyACM0"
aux_port    : "/dev/ttyACM1"
baudrate    : 115200
debug       : True
diagnostics : True

# If set to true, this will configure the requested baudrate on the device.
# Note that this will be set on both USB and serial, but will only actually affect the baudrate of a serial connection.
set_baud : False

# Frame IDs used in the different messages. By default these are set to arbitrary strings as not to interfere with other ROS services.
# For more information on common frame IDs, check out: https://www.ros.org/reps/rep-0105.html
#
# filter_frame_id and filter_child_frame_id are specifically useful as the node will also publish a transform to the /tf topic
# that contains the transform between these two frames. Many ROS tools such as RViz will use the /tf topic to display things like robot position.
imu_frame_id          : "sensor"
gnss1_frame_id        : "gnss1_antenna_wgs84"
gnss2_frame_id        : "gnss2_antenna_wgs84"
filter_frame_id       : "sensor_wgs84"
filter_child_frame_id : "sensor"
nmea_frame_id         : "nmea"

# Waits for a configurable amount of time until the device exists
# If poll_max_tries is set to -1 we will poll forever until the device exists
poll_port      : False
poll_rate_hz   : 1.0
poll_max_tries : 60

# Number of times to attempt to reconnect to the device if it disconnects while running
# If configure_after_reconnect is true, we will also reconfigure the device after we reconnect
#      NOTE: It is strongly recommended to configure after reconnect unless device_setup is set to false
#            as the device will likely initialize in a different state otherwise
reconnect_attempts : 0
configure_after_reconnect : True

# Controls if the driver outputs data with-respect-to ENU frame
#      false - position, velocity, and orientation are WRT the NED frame (native device frame)
#      true  - position, velocity, and orientation are WRT the ENU frame
use_enu_frame : False

# Controls if the driver-defined setup is sent to the device
#      false - The driver will ignore the settings below and use the device's current settings
#      true  - Overwrite the current device settings with those listed below
device_setup : True

# Controls if the driver-defined settings are saved
#      false - Do not save the settings
#      true  - Save the settings in the device's non-volatile memory
save_settings : False

# Controls if the driver uses the device generated timestamp (if available) for timestamping messages
#      false - Use PC received time for timestamping
#      true  - Use device generated timestamp
[... 428 lines cut ...]
#         4 - Sentences will start with GL
#       The purpose of the talker ID is to differentiate when the same message_id comes from different data sources (descriptor sets)
#       The filter_nmea_talker_id will be applied to all NMEA messages from the filter descriptor set
filter_nmea_talker_id: 3
filter_nmea_gga_data_rate: 0
filter_nmea_gll_data_rate: 0
filter_nmea_rmc_data_rate: 0
filter_nmea_hdt_data_rate: 0
filter_nmea_prka_data_rate: 0  # Note that this message_id will not have any talker ID on it since it is proprietary and can only come from the filter descriptor set

**To Reproduce**
Steps to reproduce the behavior:
1. Launch Node with command `...`
2. ...
3. See Error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Environment (please complete the following information):**
 - OS: [e.g. Ubuntu]
 - Architecture: [e.g. x86_64, arm32, arm64]
 - ROS Version: [e.g. melodic, noetic, foxy, galactic, rolling]
 - Version: [e.g. 2.0.0]
 - Sensor(s): [e.g. 3DM-GQ7, 3DM-GX5-45]

robbiefish, 2024-03-12:
HI @Zahin-Nambiar, please see the [lifecycle node](https://github.com/LORD-MicroStrain/microstrain_inertial/tree/ros2-3.x.x?tab=readme-ov-file#lifecycle-node) section of the README.

To summarize, you should launch the driver with
```
ros2 launch microstrain_inertial_driver microstrain_launch.py configure:=true activate:=true
```

Zahin-Nambiar, 2024-03-22:
resolved, thank you

closed 2024-03-22 by Zahin-Nambiar