LORD-MicroStrain/microstrain_inertial #322: colcon build error : 3DMGX5-AHRS

kevinhan75, 2024-04-24:
**Describe the bug**
I did install 'ros2' branch of microstrain_inertial in ubuntu 20.04, ROS2 foxy for installing 3DMGX5-AHRS
When I executed 'colcon build', I got the error below:

kevinhan@kevinhan-locomotion:~/microstrain_ws$ colcon build
Starting >>> microstrain_inertial_msgs
Starting >>> microstrain_inertial_description
Finished <<< microstrain_inertial_description [0.63s]                                                                       
Finished <<< microstrain_inertial_msgs [16.2s]                       
Starting >>> microstrain_inertial_driver
Starting >>> microstrain_inertial_rqt                      
Finished <<< microstrain_inertial_rqt [0.69s]                                                                  
--- stderr: microstrain_inertial_driver                          
CMake Error at CMakeLists.txt:92 (find_package):
  By not providing "Findnmea_msgs.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "nmea_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "nmea_msgs" with
  any of the following names:

    nmea_msgsConfig.cmake
    nmea_msgs-config.cmake

  Add the installation prefix of "nmea_msgs" to CMAKE_PREFIX_PATH or set
  "nmea_msgs_DIR" to a directory containing one of the above files.  If
  "nmea_msgs" provides a separate development package or SDK, be sure it has
  been installed.

---
Failed   <<< microstrain_inertial_driver [1.22s, exited with code 1]

Summary: 3 packages finished [17.7s]
  1 package failed: microstrain_inertial_driver
  1 package had stderr output: microstrain_inertial_driver
  1 package not processed

How can I solve this problem?

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Architecture: x86_64
 - ROS Version: foxy
 - Version: 
 - Sensor(s): 3DMGX5-AHRS]

robbiefish, 2024-05-15:
This seems like a problem with the overall rosdep configuration. All of those packages should be able to be found. Have you setup rosdep? Documentation for rosdep can be found [here](https://wiki.ros.org/rosdep)

kevinhan75, 2024-06-10:
I did something as the overall rosdep configuration.
But still same problem occur.

closed 2024-07-14 by github-actions[bot]