LORD-MicroStrain/microstrain_inertial #333: How to configure device timestamping?

stephen-cpr, 2024-05-23:
I'm using a GQ7 and would like the driver to output IMU data based on the internal (presumably GPS time sync'd) clock, not the ROS time when the packet was received. It looks like in the past there was a 'use_device_timestamp' parameter you could configure, but this has since been removed. Is there still a way to do this?

Aposhian, 2024-05-24:
It seems like we should be able to add that parameter back, and just have that option be read in the `updateHeaderTime` method. The GPS timestamp is already being passed into that method, but just isn't being used.

hofbauerc, 2024-10-03:
Are there any methods on how to use the "use_device_timestamp" parameter with the current version? It seems weird that this functionality has been removed.

robbiefish, 2024-10-03:
I am working on re implementing this for the driver. This was removed when releasing 4.0.0 in order to update and improve the timestamping feature, but due to some other priorities never got added back. I will link to the PR once it is open

hofbauerc, 2024-10-03:
Ok thanks for the reply. 

So currently the timestamp is based on the system time of the PC am I right?

hofbauerc, 2024-11-04:
@robbiefish any news on when we can expect that feature to be added back?

When we ordered our GQ7 we thought this would be the default way in terms of timestamping and now we basically wait for this feature to be added back to properly continue our projects with it.

robbiefish, 2024-11-07:
This has been readded in `4.5.0`, please see the [`timestamp_source`](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/62db5238278f8a6092b35a3d06e5b6694ed6c0f4/config/params.yml#L79) parameter for more information.

If you want to go back to using the exact device timestamp, you can set `timestamp_source` to 1, but keep in mind that when the GQ7 gets a GPS fix, the timestamp will jump considerably. If you want a mostly consistent timestamp that should track ROS time, but with a dt similar to the device timestamp, you can set `timestamp_source` to 2

closed 2024-11-07 by robbiefish

robbiefish, 2024-11-25:
Hi @lincob, the `gnss_x/time` topic you subscribed to is a special case. That message will **always** contain the device timestamp in the `time_ref` field, and be stamped with ROS time.

The purpose of that message is to show the relationship between the device GNSS time and ROS time. All other messages should be stamped with the requested timestamp

lincob, 2024-12-19:
Thanks @robbiefish that actually makes sense.