LORD-MicroStrain/microstrain_inertial #335: How to completely disable the magnetometer of GV7 Models?

karlji1021, 2024-06-03:
Hello, I recently changed the IMU model from 3DM-GX5-AHRS to 3DM-GV7-AHRS.
Before changing the model, I used a code snippet like the one below to disable a magnetometer.

"""
mscl::HeadingUpdateOptions headingUpdateOptions;
headingUpdateOptions.useInternalMagnetometer = false;
node_->setHeadingUpdateControl(headingUpdateOptions);
"""

However, for the GV7 model, "setHeadingUpdateControl" method doesn't work anymore.
The method gives "Error_NotSupported" error.

So, what I've tried to solve this problem are the ones below.

Disable aiding measurement source
"""
enableDisableAidingMeasurement(InertialTypes::AidingMeasurementSource::MAGNETOMETER_AIDING, false);
"""

Disable magnetometer usage in SensorConnect.
The magnetometer usage was set to be disabled, and this setup is saved as the initial settings.
I've checked that the disabled setting remained after disconnecting/reconnecting the IMU cable.

To test if the magnetometer is disabled, a small magnet was put close to the IMU.
Then, the estimated direction changed depending on the direction of the magnet.

I wonder how to completely disable its magnetometer.

Thank you.

robbiefish, 2024-06-03:
Not relevant to the ROS driver, please do not cross post. Original question can be found at LORD-MicroStrain/MSCL#391

closed 2024-06-03 by robbiefish

karlji1021, 2024-06-03:
@robbiefish Do you know anyone supporting the MSCL repository? I've been waiting for a long time but there's no response.