LORD-MicroStrain/microstrain_inertial #343: Python Package Errors in ROS2 rqt launch

Owen-Liuyuxuan, 2024-07-22:
**Describe the bug**
I got an error while running `ros2 launch microstrain_inertial_rqt quickview_launch.py` on ROS humble.
```bash
[rqt_gui-1] Traceback (most recent call last):
[rqt_gui-1]   File "/opt/ros/humble/lib/python3.10/site-packages/rqt_gui/ros_plugin_provider.py", line 80, in load
[rqt_gui-1]     module = __builtin__.__import__(
[rqt_gui-1]   File "/opt/ros/humble/lib/python3.10/site-packages/microstrain_inertial_rqt/quickview.py", line 4, in <module>
[rqt_gui-1]     from .utils.widgets import MicrostrainWidget
[rqt_gui-1]   File "/opt/ros/humble/lib/python3.10/site-packages/microstrain_inertial_rqt/utils/widgets.py", line 9, in <module>
[rqt_gui-1]     from .services import DeviceReportMonitor
[rqt_gui-1]   File "/opt/ros/humble/lib/python3.10/site-packages/microstrain_inertial_rqt/utils/services.py", line 3, in <module>
[rqt_gui-1]     from microstrain_inertial_msgs.srv import DeviceReport
[rqt_gui-1] ImportError: cannot import name 'DeviceReport' from 'microstrain_inertial_msgs.srv' (/opt/ros/humble/local/lib/python3.10/dist-packages/microstrain_inertial_msgs/srv/__init__.py)
```

**To Reproduce**
Steps to reproduce the behavior:
1. On a machine with basic ROS2 humble installed. Install the packages with `sudo apt install ros-humble-microstrain-inertial-driver ros-humble-microstrain-inertial-rqt`
2. I launched the driver with no significant error `ros2 launch microstrain_inertial_driver microstrain_launch.py` and I can see the topics and data flowing correctly.
3. I run `ros2 launch microstrain_inertial_rqt quickview_launch.py ` and the error messages can be seen.

https://github.com/LORD-MicroStrain/microstrain_inertial_msgs_common/tree/269c6c062ee0a8cb0bbf56510fdd66a652f26015
I can not see any messages with `DeviceReport` name

**Expected behavior**
I expected the rqt_gui to launch correctly. But there are python errors as posted before and plugin loading failure (probably due to python error)

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Architecture: x86_64
 - ROS Version: humble
 - Sensor(s): 3DM-GX5-15

**Modifications**
I have not changed the code because I just installed the packages with apt.

**Launch Parameters**
I do not have a launch parameter file.

**Additional context**
Add any other context about the problem here.

ebnerluca, 2024-07-23:
I have the same problem in ros 1 noetic. I installed the rqt tools with `apt-get install ros-noetic-microstrain-inertial-rqt` and when I run `oslaunch /opt/ros/noetic/share/microstrain_inertial_rqt/launch/quickview.launch` I get the error
```
[...]
ImportError: cannot import name 'DeviceReport' from 'microstrain_inertial_msgs.srv' (/opt/ros/noetic/lib/python3/dist-packages/microstrain_inertial_msgs/srv/__init__.py)
```

robbiefish, 2024-07-25:
The `microstrain_inertial_rqt` package is considerably broken since the release of `4.0.0`. I have been meaning to fix it for some time, but have had other priorities, and unfortunately, that will probably continue for some time.

If anyone is interested in providing a fix for this, I am happy to review a PR. Otherwise, I will update this issue once the package gets fixed

snwu1996, 2024-08-06:
I created a [draft PR](https://github.com/LORD-MicroStrain/microstrain_inertial_rqt_common/pull/4) for this issue in the `microstrain_intertial_rqt_common` submodule.

It's still work in progress but it is ~80% usable at the moment. Several widgets are still not being mapped to data properly because  the data format of certain topics changed.

Rhyssmcm, 2024-11-19:
Currently using ros2 (Jazzy) on Ubuntu 24.04 and seeing the exact same issue, Just starting to use this package and wondering how this could be fixed since the file that is being imported not existing in any file directories " ImportError: cannot import name 'DeviceReport' from 'microstrain_inertial_msgs.srv'", an explanation of how this worked previously could help a future fix.

aidan-laracy-HBK, 2026-10-01:
We are closing this issue as it has been inactive for a while. If you are still experiencing this problem, please leave a comment and we will gladly reopen it.

closed 2026-10-01 by aidan-laracy-HBK