LORD-MicroStrain/microstrain_inertial #367: GPS and PPS time sync status

amir-zviran, 2024-11-26:
Hi,

I'm using 3DM-CV7. I connect its PPS input to my GPS module, and sending it time reference messages on the `/external_gps_time` topic. I can see it is synced by looking at the timestamps of the messages, but I wonder if I can rely on that to tell whether the sync is working well.

I have a few questions:
1. Will the timestamps of the messages change to the GPS time (i.e. the time reference time) even if it is not yet synced with the PPS signal?
2. Is there a way to read the Time Sync Status (0xA0,0x02) register through the ROS driver?
3. Is reading the time sync status enough to tell the sync is "complete", i.e. both the PPS signal is valid and the time is correctly set according to the time reference? Or I have to look at the time_valid field in the GPS Timestamp (0xFF,0xD3) register too?
4. If I need to look at the GPS Timestamp (0xFF,0xD3) register, can I read it through the ROS driver?

Many thanks!
Amir

amir-zviran, 2025-01-09:
@robbiefish Can you help with this one?
Thanks!

robbiefish, 2025-01-09:
Hi @amir-zviran

1. Yes, it will change to GPS time as soon as it gets a valid GPS time. The reverse is also true. If PPS is acheived before GPS time, the fractional seconds will be updated correctly. Note that the newest version of the driver supports multiple different timestamp types. Please see [this section](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/62db5238278f8a6092b35a3d06e5b6694ed6c0f4/config/params.yml#L72-L79) of the config for more information, you will probably want `timestamp_source: 1` to get the raw device time.
2. Not currently no
3. The Time Sync Status field will tell you if the fractional seconds are valid. You will need to check the `time_valid` field of the GPS timestamp register to see if the time is correctly set
4. All MIP messages published by the ROS driver will have a `header.gps_timestamp` field that will contain [this message](https://github.com/LORD-MicroStrain/microstrain_inertial_msgs_common/blob/2fad96bbc33df626c2127fa3245bb507cb4cd1d3/msg/MipGpsTimestamp.msg). Note that it does not contain the `time_valid` field

With all of this being said, it looks like the current version of the driver is missing some of the information to do what you are trying to do. I think this is a valid use case, so I have made the [`feature/ros_time_sync_status`](https://github.com/LORD-MicroStrain/microstrain_inertial/tree/feature/ros_time_sync_status) and [`feature/ros2_time_sync_status`](https://github.com/LORD-MicroStrain/microstrain_inertial/tree/feature/ros2_time_sync_status) branches that should provide all of the information you are looking for.

To look at the state of the time sync, set `mip_system_time_sync_status_data_rate` to a non-zero value, then subscribe to `mip/system/time_sync_status`. You can check `time_sync` to check that the fractional seconds are set properly, and `header.gps_timestamp.time_valid` to see if the time is valid.

I do not have a setup to test PPS here, so please let me know how this works for you

amir-zviran, 2025-01-14:
Hi @robbiefish ,

Can you have a look at that?

Thanks!

robbiefish, 2025-01-16:
Hi @amir-zviran,

> The whole header of the time_sync_status message doesn't seem to be updated. It shows a constant timestamp, and the rest of the fields are all zeros/False:

This was a bug, and should be resolved in the most recent commit to the branch. To get the changes, you will need to do a `git pull` as normal, and then run `git submodule update --recursive --init` to ensure that the submodule has been updated to include the change

> I realized I must set this flag:
subscribe_ext_time : True
And use the topic /ext/time for sending my time reference. This is what I used to test the above.
Is that the correct way to go? Is filter_enable_external_gps_time_update still being used anywhere?

That is the correct way to go. That flag and the corresponding subscriber work nearly identical to the old callback that you used.

`filter_enable_external_gps_time_update` was replaced with the flag you mentioned, so you do not need to use it anymore

amir-zviran, 2025-01-16:
Thanks @robbiefish !

Is there a chance you forgot to update the ROS1 branch too? I don't see any new commits there

robbiefish, 2025-01-20:
@amir-zviran that is exactly what I did, I apologize. Should be pushed now

amir-zviran, 2025-01-20:
@robbiefish I'm still not seeing any new commit on [feature/ros_time_sync_status](https://github.com/LORD-MicroStrain/microstrain_inertial/tree/feature/ros_time_sync_status)

amir-zviran, 2025-02-05:
Hi @robbiefish ,

I tested and it looks better now, except for the time_valid field which still remains False.
I think you might have missed updating it here:
https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/a77c248991345dceee206ae21e4845931b65c287/src/publishers.cpp#L2147

A small note - I noticed that all the flags inside header.gps_timestamp.valid_flags change to False too once time_sync changes to False. It doesn't matter much for me, but I'm mentioning it here in case someone looks at it in the future.

Thanks!

amir-zviran, 2025-02-11:
Hi @robbiefish ,

Can you have a look at the above issues?

Thanks!

amir-zviran, 2025-03-02:
Hi @robbiefish ,

Can you please look at the issues I mentioned above?

Thanks a lot!

amir-zviran, 2025-03-23:
Hi @robbiefish ,

Can you please have a look at this?

Thanks!

Alex-Beh, 2025-03-27:
Hi, [`/external_gps_time`](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/62db5238278f8a6092b35a3d06e5b6694ed6c0f4/config/params.yml#L306) doesn't seem used by the code. May I know which branch you using now?

amir-zviran, 2025-03-28:
Hi @Alex-Beh , please look at the full correspondence. Indeed `/external_gps_time` isn't being used anymore. You should use `/ext/time` instead.

robbiefish, 2025-04-15:
Hi @amir-zviran,

Thank you very much for your patience. I am sorry it took me so long to get to this. The issue has been resolved, and the new release has been pushed to the ROS package index

amir-zviran, 2025-04-15:
Thanks @robbiefish!
I'll give the new version a try and let you know if I notice any issues.

amir-zviran, 2025-04-16:
Hi @robbiefish,
I don't see it on the ROS package index. Should it be there already?

robbiefish, 2025-04-16:
@amir-zviran, no I apologize for being confusing. I have contributed it to `rosdistro` which is what I need to do to get the package released in the index.

The ROS maintainers should merge the PR within the next couple of days and then a few days after that, the package index will be updated. I would say to keep an eye out for this in the package index next week or the week after

amir-zviran, 2025-04-16:
Cool, thanks for the clarification

amir-zviran, 2025-04-23:
Hi @robbiefish,
I'm using ROS noetic. I can see the new version (4.6.0) on the index - https://index.ros.org/p/microstrain_inertial_driver/#noetic
But when trying to install it using apt install, I'm still getting the older version (4.5.0).
Do you have any idea why? Maybe it is not released for ROS noetic?
Thanks!

closed 2025-05-11 by amir-zviran