LORD-MicroStrain/microstrain_inertial #368: microstrain_inertial_driver]: Unable to set device to idle

devvaibhav455, 2024-12-03:
**Describe the bug**
I boot the Jetson and sometimes the IMU driver is not working. After a reboot, it works.

**To Reproduce**
Steps to reproduce the behavior:
1. Star the jetson. Get below error sometimes (not always)

**Fail Log**
```
Jan 01 00:00:53 echo-jetson-orin-nx clearpath-sensors-start[3429]: [INFO] [microstrain_inertial_driver_node-6]: process started with pid [3542]
Jan 01 00:00:54 echo-jetson-orin-nx clearpath-sensors-start[3429]: [microstrain_inertial_driver_node-6] [INFO] [0000000054.226971680] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Running microstrain_inertial_driver version: unknown
Jan 01 00:00:54 echo-jetson-orin-nx clearpath-sensors-start[3429]: [microstrain_inertial_driver_node-6] [INFO] [0000000054.227219488] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Using MIP SDK version: v0.0.0
Jan 01 00:00:54 echo-jetson-orin-nx clearpath-sensors-start[3429]: [microstrain_inertial_driver_node-6] [WARN] [0000000054.227242240] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: No relative position configured. We will not publish relative odometry or transforms.
Jan 01 00:00:54 echo-jetson-orin-nx clearpath-sensors-start[3429]: [microstrain_inertial_driver_node-6] [WARN] [0000000054.227253120] [a200_0524.sensors.imu_0.microstrain_inertial_driver]:   Please configure relative position to publish relative position data
Jan 01 00:00:54 echo-jetson-orin-nx clearpath-sensors-start[3429]: [microstrain_inertial_driver_node-6] [INFO] [0000000054.227768096] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Attempting to open serial port </dev/microstrain_main> at <115200>
Jan 01 00:00:54 echo-jetson-orin-nx clearpath-sensors-start[3429]: [microstrain_inertial_driver_node-6] [INFO] [0000000054.231347136] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Setting device to idle in order to configure
Jan 01 00:01:00 echo-jetson-orin-nx clearpath-sensors-start[3429]: [microstrain_inertial_driver_node-6] [ERROR] [0000000060.236806784] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Unable to set device to idle
Jan 01 00:01:00 echo-jetson-orin-nx clearpath-sensors-start[3429]: [microstrain_inertial_driver_node-6] [ERROR] [0000000060.236954336] [a200_0524.sensors.imu_0.microstrain_inertial_driver]:   Error(-4): Timed out
Jan 01 00:01:00 echo-jetson-orin-nx clearpath-sensors-start[3429]: [microstrain_inertial_driver_node-6] [ERROR] [0000000060.237021312] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Failed to read configuration for node
Jan 01 00:01:00 echo-jetson-orin-nx clearpath-sensors-start[3429]: [microstrain_inertial_driver_node-6] [ERROR] [0000000060.237048608] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Failed to configure node base
Jan 01 00:01:00 echo-jetson-orin-nx clearpath-sensors-start[3429]: [microstrain_inertial_driver_node-6] [FATAL] [0000000060.237064960] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Failed to configure node
Jan 01 00:01:00 echo-jetson-orin-nx clearpath-sensors-start[3429]: [ERROR] [microstrain_inertial_driver_node-6]: process has died [pid 3542, exit code 1, cmd '/home/nvidia/ros2_ws/install/microstrain_inertial_driver/lib/microstrain_inertial_driver/microstrain_inertial_driver_node --ros-args -r __node:=microstrain_inertial_driver -r __ns:=/a200_0524/sensors/imu_0 --params-file /tmp/launch_params_ztka9fm4 --params-file /etc/clearpath/sensors/config/imu_0.yaml --params-file /tmp/launch_params_xd0ycmr3 -r imu/data:=data -r /moving_ang:=moving_ang -r /tf:=tf -r /tf_static:=/a200_0524/tf_static'].
```

**Pass Log**
```
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [INFO] [microstrain_inertial_driver_node-6]: process started with pid [3598]
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6] [INFO] [0000004535.603135936] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Running microstrain_inertial_driver version: unknown
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6] [INFO] [0000004535.603405376] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Using MIP SDK version: v0.0.0
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6] [WARN] [0000004535.603429536] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: No relative position configured. We will not publish relative odometry or transforms.
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6] [WARN] [0000004535.603460992] [a200_0524.sensors.imu_0.microstrain_inertial_driver]:   Please configure relative position to publish relative position data
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6] [INFO] [0000004535.606022080] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Attempting to open serial port </dev/microstrain_main> at <115200>
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6] [INFO] [0000004535.609187232] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Setting device to idle in order to configure
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6] [INFO] [0000004535.625217632] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Main Connection Info:
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6]     #######################
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6]     Model Name:       3DMGX5-AHRS
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6]     Serial Number:    <Hidden on purpose>
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6]     Firmware Version: 1.1.76
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6]     #######################
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6] [INFO] [0000004535.637202720] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Note: The device does not support publishing the topic ekf/dual_antenna_heading
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6] [INFO] [0000004535.637301536] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Note: The device does not support publishing the topic ekf/llh_position
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6] [INFO] [0000004535.637371456] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss_1/llh_position
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6] [INFO] [0000004535.637405024] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss_1/odometry_earth
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6] [INFO] [0000004535.637417152] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss_1/time
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6] [INFO] [0000004535.637426144] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss_1/velocity
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6] [INFO] [0000004535.637434944] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss_1/velocity_ecef
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6] [INFO] [0000004535.637442848] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss_2/llh_position
Jan 01 01:15:35 echo-jetson-orin-nx clearpath-sensors-start[3472]: [microstrain_inertial_driver_node-6] [INFO] [0000004535.637452160] [a200_0524.sensors.imu_0.microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss_2/odometry_earth
[... cut ...]
The driver should work and publish the messages on imu's relevant topics

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Architecture: arm64
 - ROS Version: humble
 - Version: [e.g. 2.0.0]
 - Sensor(s): 3DMGX5-AHRS
**Modifications**
No

**Launch Parameters**
If you have an updated `params.yml` file, please attach it to this issue.

**Additional context**
I have the ROS2 branch of the driver. It runs through clearpath's systemd sensor bringup service. I didn't make any changes to the config files myself between these two runs.

devvaibhav455, 2025-02-09:
For bot: my last comment describes my situation.

aidan-laracy-HBK, 2026-10-01:
We are closing this issue as it has been inactive for a while. If you are still experiencing this problem, please leave a comment and we will gladly reopen it.

closed 2026-10-01 by aidan-laracy-HBK