LORD-MicroStrain/microstrain_inertial #371: GX5 publishes incorrect data on EKF topic

ilansmolySR, 2025-01-01:
**Describe the bug**
Problem in /ekf/imu/data topic with 3DM-GX5-AHRS:
The robot is not moving but the linear accelearion is far from 0.
The roll, pitch are fixed to 0, and the yaw is fixed to 90.

[image]

When we connect 3DM-GV7-AHRS everything seems normal
[image]

But we need compatability with both types of sensors.

**To Reproduce**
1. Launch Node with command `roslaunch microstrain_inertial_driver microstrain.launch`
2. See Error

**Environment (please complete the following information):**
 - OS:  Ubuntu
 - ROS Version: noetic
 - Sensor(s): 3DM-GX5-AHRS
 - docker environment

**Launch Parameters**
If you have an updated `params.yml` file, please attach it to this issue.

ilansmolySR, 2025-01-01:
# Standalone example params file for GX3, GX4, GX/CX5, RQ1 and GQ7 series devices
# Note: Feature support is device-dependent and some of the following settings may have no affect on your device.
# Please consult your device's documentation for supported features

# ****************************************************************** 
# NOTE: This file is formatted to work with ROS and will not work if specified as the params_file argument in ROS2.
#       If you want to override parameters for ROS2, start with https://github.com/LORD-MicroStrain/microstrain_inertial/blob/ros2/microstrain_inertial_driver/config/empty.yml
# ****************************************************************** 

# ******************************************************************
# General Settings
# ******************************************************************

# port is the main port that the device will communicate over. For all devices except the GQ7, this is the only available port.
# aux_port is only available for the GQ7 and is only needed when streaming RTCM corrections to the device from ROS, or if you want to publish NMEA sentences from this node
port         : "/dev/ttyACM0"
baudrate     : 115200
aux_port     : "/dev/ttyACM1"
aux_baudrate : 115200
debug        : False

# If set to true, this will configure the requested baudrate on the device for the main port and aux port if possible.
# Note that this will be set on both USB and serial, but will only actually affect the baudrate of a serial connection.
set_baud : False

# Waits for a configurable amount of time until the device exists
# If poll_max_tries is set to -1 we will poll forever until the device exists
poll_port      : False
poll_rate_hz   : 1.0
poll_max_tries : 60

# Number of times to attempt to reconnect to the device if it disconnects while running
# If configure_after_reconnect is true, we will also reconfigure the device after we reconnect
#
# Note: It is strongly recommended to configure after reconnect unless device_setup is set to false
#       as the device will likely initialize in a different state otherwise
reconnect_attempts : 0
configure_after_reconnect : True

# Controls if the driver-defined setup is sent to the device
#     false - The driver will ignore the settings below and use the device's current settings
#     true  - Overwrite the current device settings with those listed below
device_setup : True

# Controls if the driver-defined settings are saved
#     false - Do not save the settings
#     true  - Save the settings in the device's non-volatile memory
save_settings : False

# Controls if the driver creates a raw binary file
#     false - Do not create the file
#     true  - Create the file
#
# Note: The filename will have the following format -
#       model_number "_" serial_number "_" datetime (year_month_day_hour_minute_sec) ".bin"
#           example: "3DM-GX5-45_6251.00001_20_12_01_01_01_01.bin"
# Note: This file is useful for getting support from the manufacturer
raw_file_enable : False

# (GQ7/CV7 only) Controls if the driver requests additional factory support data to be included in the raw data file
#     false - Do not request the additional data
#     true  - Request the additional channels (please see notes below!)
#
# Note: We recommend only enabling this feature when specifically requested by Microstrain.
#       Including this feature increases communication bandwidth requirements significantly...
#       for serial data connections please ensure the baudrate is sufficient for the added data channels.     
raw_file_include_support_data : True

# The directory to store the raw data file
raw_file_directory : "/home/your_name"
[... 487 lines cut ...]
# Note: filter_nmea_talker_id can be any of the follwing numeric values:
#         1 - Sentences will start with GN
#         2 - Sentences will start with GP
#         3 - Sentences will start with GA
#         4 - Sentences will start with GL
#       The purpose of the talker ID is to differentiate when the same message_id comes from different data sources (descriptor sets)
#       The filter_nmea_talker_id will be applied to all NMEA messages from the filter descriptor set
filter_nmea_talker_id: 3
filter_nmea_gga_data_rate: 0
filter_nmea_gll_data_rate: 0
filter_nmea_rmc_data_rate: 0
filter_nmea_hdt_data_rate: 0
filter_nmea_prka_data_rate: 0  # Note that this message_id will not have any talker ID on it since it is proprietary and can only come from the filter descriptor set

# (CV7-INS only) External aiding measurement configuration
subscribe_ext_time        : False
subscribe_ext_fix         : False
subscribe_ext_vel_ned     : False
subscribe_ext_vel_enu     : False
subscribe_ext_vel_ecef    : False
subscribe_ext_vel_body    : False
subscribe_ext_heading_ned : False
subscribe_ext_heading_enu : False
subscribe_ext_mag         : False
subscribe_ext_pressure    : False

ilansmolySR, 2025-01-01:
[params.txt](https://github.com/user-attachments/files/18285803/params.txt)

robbiefish, 2025-01-07:
Hi @ilansmolySR,

The reason you are seeing different behavior between the sensor has to do with the fact that the heading for the two devices is initializing differently

In your params.txt, you have the following line:
```yaml
# This line sets the heading source for the GX5-AHRS to None
filter_heading_source : 0
```

You also have this line:
```yaml
# This line sets the heading source for the GV7-AHRS to Magnetometer
filter_auto_heading_alignment_selector : 4
```

Since you are looking at the [Linear Accel](https://files.microstrain.com/GQ7+User+Manual/external_content/dcp/Data/0x82/data/0x0d.htm) data quantity which attempts to factor out gravity, the device needs to have a somewhat reasonable estimate for the device's orientation in order to properly factor out gravity.

In order to resolve this issue, I would suggest setting `filter_heading_source` back to 1

closed 2025-02-09 by github-actions[bot]