LORD-MicroStrain/microstrain_inertial #372: /dev/microstrain_main udev not showing up using waveshare FTDI

dustinlim26, 2025-01-11:
my setup:
OS: ubuntu 22.04/ Jammy
ROS: ROS2 Humble
imu:  3DM-CV7-AHRS
usb to ttl : waveshare FTDI (https://www.waveshare.com/usb-to-ttl.htm)

I connect the IMU to the usb-to-ttl (FTDI) and then usb-to-ttl connects to   The device showed up is ttyUSB0 in which I have specified it in param file for the launch file (cause default is ttyACM0). However, when I launched it, it gives error on not able to set the device to idle. I suspect that the FTDI doesn't know how to relay the data to the IMU, so I think its a permission / device discovery problem

I tried to do the udev but I can't seem to get the /dev/microstrain_main result. Does someone knows any solution with a similar setup with mine?

Thank you

limdustin26, 2025-01-11:
I am also unable to connect the imu with FTDI to SensorConnect in windows. It couldn't find the IMU. In the device manager tho, I have confirmed that the FTDI is detected as COM port. I have also set the same baud rate when trying to discover the IMU in SensorConnect

limdustin26, 2025-01-18:
Update: testing the imu on jetson orin nano uart interface immediately -> result: cannot detect imu when ls /dev/tty* 

Voltage of RX and TX of imu is 3V. Imu is powered on by 5V (indicated by green led)

aidan-laracy-HBK, 2026-10-01:
Hello, we are closing this issue as it has been inactive for a while. This appears to be a CV7 question, so please feel free to open a ticket at [support.microstrain.com](support.microstrain.com) if you are still having issues.

closed 2026-10-01 by aidan-laracy-HBK