LORD-MicroStrain/microstrain_inertial #377: Cannot launch microstrain_inertial_driver after connecting 3DM-CV7 to Jetson via UART

henry-buron, 2025-03-07:
**Describe the bug**
After connecting 3DM-CV7 INS to Jetson AGX  via UART and remapping the port, I cannot launch the microstrain_inertial_driver. I see several errors indicating that the INS cannot be configured (see below).

**To Reproduce**
Steps to reproduce the behavior:
1. Connect 3DM-CV7 to Jetson via UART. Remap port to /dev/ttyTHS2
2. Launch the microstrain_inertial_driver

Errors:
[microstrain_inertial_driver]: Unable to set device to idle
[microstrain_inertial_driver_lifecycle_node-20] [ERROR] [1741302784.544948811] [microstrain_inertial_driver]:   Error(-4): Timed out
[microstrain_inertial_driver_lifecycle_node-20] [ERROR] [1741302784.545036587] [microstrain_inertial_driver]: Failed to read configuration for node
[microstrain_inertial_driver_lifecycle_node-20] [ERROR] [1741302784.545114378] [microstrain_inertial_driver]: Failed to configure node base
[robot_system_manager_node-35] [WARN] [1741302784.545737641] [robot_system_manager_node]: Failed to 'change_state' for node microstrain_inertial_driver: microstrain_inertial_driver/change_state service client: async_send_request failed. Performing retry 2 of 10
[microstrain_inertial_driver_lifecycle_node-20] [INFO] [1741302784.546982470] [microstrain_inertial_driver]: Running microstrain_inertial_driver version: ros2-4.5.0-2-g2d69b84
[microstrain_inertial_driver_lifecycle_node-20] [INFO] [1741302784.547109446] [microstrain_inertial_driver]: Using MIP SDK version: v1.0.0-283-g2c4cb6b
[microstrain_inertial_driver_lifecycle_node-20] [INFO] [1741302784.547669700] [microstrain_inertial_driver]: Attempting to open serial port </dev/ttyTHS2> at <921600>
[microstrain_inertial_driver_lifecycle_node-20] [INFO] [1741302784.549879775] [microstrain_inertial_driver]: Setting device to idle in order to configure
[microstrain_inertial_driver_lifecycle_node-20] [ERROR] [1741302790.558009749]

**Expected behavior**
Able to launch microstrain_inertial_driver and see imu/data and imu/mag publishing like normal.

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Architecture: aarch64
 - ROS Version: humble
 - Version: 
 - Sensor(s): 3DM-CV7

**Modifications**
Did you modify the source code in any way while testing? If so, please note those modifications here.

No.

**Launch Parameters**
If you have an updated `params.yml` file, please attach it to this issue.

Attached as params.txt

[params.txt](https://github.com/user-attachments/files/19131479/params.txt)

**Additional context**

A few notes/debugging I've tried:
* Jetpack 6.0 on Jetson AGX Orin.
* The INS worked perfectly when I had it connected via USB.
* Verified that the INS is connected to /dev/ttyTHS2 because I see output on the serial terminal when I connect the INS.
* Used SensorConnect to change the baud on the INS to 921600. Saved settings. Set the baudrate to 921600 in the driver.
* Unchecked the Stream on startup box in SensorConnect
* Tested with device_setup in the driver set to both true and false.
* Set all data rates to 0 in the driver to check if the baud was saturated.

Any ideas on what's going wrong and what I can do to fix it?

henry-buron, 2025-03-08:
Just to follow up on this, I think the issue stems from something a configuration on the Jetson's UART port. Has anyone successfully connected a Microstrain INS (any) to a Jetson via UART?

Alex-Beh, 2025-03-30:
@henry-buron Have you managed to setup the communnication?

You would probably interested on this issue: https://github.com/LORD-MicroStrain/microstrain_inertial/issues/55