LORD-MicroStrain/microstrain_inertial #382: 3DM-GX5-AHRS measurement error

Daegeun-KANG, 2025-04-18:
Hello!

I am using the 3DM-GX5-AHRS model.
It works well in ROS1, but there is a problem in ROS2 humble.

I am attaching the GIF together to explain this issue.
Can you solve this problem?

Thank you.

[image]

robbiefish, 2025-04-25:
Hi @Daegeun-KANG ,

This is incredibly strange behavior. Can you please share how you are running the driver, your params file and any other relevant information?

Daegeun-KANG, 2025-04-26:
Hello!
Thank you for your time.

The driver is in github and is running using "ros2 launch microstrain_inertial_driver microstrain_launch.py".
I am attaching the params file as well.

During execution, the driver is well aware of the IMU type, etc.

Thank you for your help in solving it.

[image]

[params.txt](https://github.com/user-attachments/files/19920473/params.txt)

Daegeun-KANG, 2025-04-30:
HI !

I tested it with the same kind of different IMU and it works fine.
It seems that our IMU is broken, not the driver problem.

I'm sorry

closed 2025-04-30 by Daegeun-KANG