LORD-MicroStrain/microstrain_inertial #383: Calibrating the magnetometer of the 3DM-CV7-AHRS via driver

elevran-vis, 2025-04-23:
Hello,
Is there a magnetometer calibration tool for the 3DM-CV7-AHRS via ROS / the driver?
Since we cannot always use SensorConnect prior to operating the IMU in AHRS mode, we've looked for a solution on the driver, but have not found a tool for that.
Besides writing our own custom calibration node (with MSCL) - is there a simpler way?
Also, is there a way to provide the IMU with prior magnetometer calibration parameters at launch via config files?

Would greatly appreciate any help!
Eldar

robbiefish, 2025-04-25:
Hi @elevran-vis ,

This is something on our radar as a useful tool that should be added to the ROS environment in some way, but unfortunately, it is not currently available.

Currently there is no way to set magnetometer calibration parameters through the config file either. The only supported method right now is to calibrate using SensorConnect save settings as startup and then use the driver.

elevran-vis, 2025-04-27:
> Currently there is no way to set magnetometer calibration parameters through the config file either.

Can you guide me on how to best implement this via the ROS driver?
Alternatively, I've noticed that MSCL has some functions related to magnetometer calibration. Does it have all the required functionalities for me to implement the above? I'm asking because I've noticed that the ROS driver uses MIP SDK.

aidan-laracy-HBK, 2026-10-01:
I am closing this issue due to inactivity. Check out https://github.com/HBK-MicroStrain/microstrain_mag_cal if you are still attempting to do a magnetometer calibration in ROS.

closed 2026-10-01 by aidan-laracy-HBK