LORD-MicroStrain/microstrain_inertial #385: Add ROS 2 rclcpp component support

Aposhian, 2025-05-01:
**Is your feature request related to a problem? Please describe.**
Make the microstrain_inertial_driver nodes registered as rclcpp_components plugins so they can be loaded and used with intraprocess communication.

**Describe the solution you'd like**
We can probably just replace the main function files for the lifecycle and non-lifecycle nodes and just register it as a component, and let the rclcpp_components CMake generate the executables for us.

Alex-Beh, 2025-05-06:
@Aposhian I believe there were some past attempts to introduce ROS 2 component support in the [microstrain_inertial](https://github.com/LORD-MicroStrain/microstrain_inertial) repository. You can refer to this branch:
https://github.com/LORD-MicroStrain/microstrain_inertial/tree/feature/ros2_component

At the time, I wasn’t aware of that branch and ended up creating my own version. Feel free to try it out here:
https://github.com/Alex-Beh/microstrain_inertial/tree/feature/ros2_component

aidan-laracy-HBK, 2026-10-01:
Hello, I am closing this issue due to inactivity. Please feel free to re-open a ticket if your issue persists.

closed 2026-10-01 by aidan-laracy-HBK