LORD-MicroStrain/microstrain_inertial #387: simple 3DM-GV7-INS launch file when only position is used from external GPS

JereKnuutinen, 2025-05-09:
Could some one provide simple launch file when using only lat lon alt from the external GPS. I am mainly interested in global estimation. For some reason I am not able to obtain ekf estimates when using example launch file.

JereKnuutinen, 2025-05-09:
Even if I launch example "cv7_ins_launch.py" I get [microstrain_inertial_driver_node-1] [WARN] [1746791184.603413099] [microstrain_inertial_driver]: Sensor not in full nav mode yet, waiting to publish relative position... What does that mean?

aidan-laracy-HBK, 2026-10-01:
I am closing this ticket due to inactivity. If your issue still persists, we are happy to help if you submit a ticket at [support.microstrain.com](url).

closed 2026-10-01 by aidan-laracy-HBK